Koenkk/zigbee2mqtt #26402: How to install a older Zigbee2Mqtt Version

arthurperezbessa, 2025-02-17:
### What happened?

I'm experiencing stability issues with Zigbee2MQTT (Z2M) and have read that some users had better results with an older version. So, I plan to install a previous version to see if it improves stability.

However, I'm not very familiar with GitHub, and I'm unsure of the best way to do this. So far, I've tried:

Adding the repository, but it returns errors.
Manually adding the add-on locally, but I can't seem to point it correctly.
Does anyone know the easiest way to install an older version of Z2M?

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

No version

### Adapter

Sonoff ZB Dongle - E

### Setup

Home Assistant on Raspberry Pi4

srett, 2025-02-17:
> ### Zigbee2MQTT version
> 
> 1.34.0

If you're really running that version, then maybe *upgrading* is a better thing to try first? That version was released over a year ago.

Otherwise, follow the [official instructions](https://www.zigbee2mqtt.io/guide/installation/01_linux.html) and instead of doing the shallow `git clone --depth 1 https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt`, do a normal clone and check out a version tag afterwards by doing `git checkout 1.39.0` for example. Then follow along with the rest of the instructions.
To try different versions later just do the last couple steps again with a different version tag, e.g.

```sh
cd /opt/zigbee2mqtt
git checkout 1.36.0
pnpm i --frozen-lockfile
pnpm run build
```

arthurperezbessa, 2025-02-17:
This works in Home Assistant OS in Raspberry Pi?

srett, 2025-02-17:
> Does this works in Home Assistant OS in Raspberry Pi? Because you sent me Linux instructions

Ouch sorry, I missed that part from your report :-(

According to [a quick search](https://community.home-assistant.io/t/how-to-downgrade-zigbee2mqtt/231273) it doesn't seem that simple. But can you at least update, or is 1.34 really the latest version the addon repo offers?

arthurperezbessa, 2025-02-17:
I current use the Version 2.1.1 and it's giving me a hard time. Trying to downgrade so I can try a more stable version

Nerivec, 2025-02-18:
Using an older version is not recommended, especially not in HAOS.
It would be better to figure out what is causing your instabilities. Can you describe them? Do you have some `debug` logs to go with that? What's the firmware version on your adapter?

arthurperezbessa, 2025-02-18:
Hi Nerivec!

I've posted about my issue here: https://github.com/Koenkk/zigbee2mqtt/issues/26220

Thanks in advance

Nerivec, 2025-02-19:
Continuing there.

closed 2025-02-19 by Nerivec