Koenkk/zigbee2mqtt #26405: Unsupported attribute - Aurora Lighting AU-A1ZBMPRO1ZX

tmonk, 2025-02-17:
### What happened?

When changing the TWMPROZXBulb50AU 'Color temp startup' startup setting I get the error:

`z2m: Publish 'set' 'color_temp_startup' to 'XXX' failed: 'Error: ZCL command 0xXXX/1 lightingColorCtrl.write({"startUpColorTemperature":455}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

### What did you expect to happen?

The attribute would be set.

### How to reproduce it (minimal and precise)

Pair a TWMPROZXBulb50AU.
Change the 'Color temp startup' attribute.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

7.4.4 [GA]

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Setup

Add-on on Home Assistant, Raspberry Pi 5

### Debug log

error 2025-02-17 15:33:49z2m: Publish 'set' 'color_temp_startup' to 'XXX' failed: 'Error: ZCL command 0xXXX/1 lightingColorCtrl.write({"startUpColorTemperature":65535}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

tmonk, 2025-02-17:
This was also noted in:

> I have installed some of these lights and I'm getting the same error with the "Color temp startup" setting. @Fezziemk4, can you confirm if this has worked in the past or if this setting is working for you? 

 _Originally posted by @Pel1can111 in [#18743](https://github.com/Koenkk/zigbee2mqtt/issues/18743#issuecomment-2321352824)_

Pel1can111, 2025-02-17:
I don't believe its possible with these lights, I have tried the official gateway/hub and there are no options for startup colour or brightness which is a shame as these lights are otherwise really good.

One thing to keep in mind, some of these lights are sold with an older firmware. This firmware has a different default colour temp with the newer firmware having a much more balanced default colour (278). I ended up picking up a official hub just to update a bunch of them.

tmonk, 2025-02-17:
> I don't believe its possible with these lights, I have tried the official gateway/hub and there are no options for startup colour or brightness which is a shame as these lights are otherwise really good.
> 
> One thing to keep in mind, some of these lights are sold with an older firmware. This firmware has a different default colour temp with the newer firmware having a much more balanced default colour (278). I ended up picking up a official hub just to update a bunch of them as it turns out some were the newer firmware and some were older.
> 
> Also transition time is broken on older firmware.

Thanks a lot - I figured this, and am in the process of replacing my dumb switches to smart switches to get around this startup colour issue.

Very good to know about the firmware, shame the official hubs are £100+!

raress96, 2025-03-19:
I am having a similar issue with a Lidl bulb:
https://www.zigbee2mqtt.io/devices/HG06106C.html

The color temperature and RGB color worked fine until recently, but some update maybe to Z2M or Home Assistant broke this, the latest HA release had a not about MQTT:
https://www.home-assistant.io/blog/2025/03/05/release-20253/
```Legacy color_mode support for MQTT JSON lights has been removed. The support was deprecated in HA Core 2024.4.```

Not sure if it is 100% this same issue though

closed 2025-05-27 by github-actions[bot]

gu33mis, 2025-11-05:
> I don't believe its possible with these lights, I have tried the official gateway/hub and there are no options for startup colour or brightness which is a shame as these lights are otherwise really [...]

Thanks for the info — really useful to know about the firmware differences and transition time issue.

Did you ever manage to figure out how to set the startup colour or brightness on those lights? Either through Zigbee2MQTT or the official Aurora AOne hub?

I’m considering getting the original hub just for that feature, but if it still doesn’t work even after a firmware upgrade or when connected to the genuine AOne hub, I might give it a miss. It’s a shame, because the lights themselves are great — that 5000K/100% default every time you power them on is the only real drawback.