Koenkk/zigbee2mqtt #26407: [New device support]: Aqara Display Switch V1 EU

mike-nott, 2025-02-17:
### Link

https://eu.aqara.com/products/aqara-display-switch-v1

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x54ef44100118eacc","nwkAddr":3840,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.switch.aeu001","epList":[1,2,3,4,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[2820,1794,18,5,4,3,0,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.aeu001","manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":37,"stackVersion":27,"hwVersion":1,"dateCode":"20241106"}},"manuSpecificLumi":{"attributes":{"223":{"type":"Buffer","data":[22,33,2,0,23,33,0,0,12,33,0,0,24,33,0,0]},"229":{"type":"Buffer","data":[2,0,2,0,0,36,119,147,93,5,182]},"238":3365,"247":{"type":"Buffer","data":[162,32,0,163,32,0,170,35,1,0,0,0,171,35,2,0,0,0,180,32,0]},"308":{"type":"Buffer","data":[112,197,156,230,83,140]},"322":1,"533":0,"554":2,"614":{"type":"Buffer","data":[255,255,255,255]},"770":{"type":"Buffer","data":[0,1,0,0,0,1,2,0,0,0]},"772":0,"773":{"type":"Buffer","data":[0,0,0,0,1,1,0,0,0,2]},"775":{"type":"Buffer","data":[0,0,0,0,1]},"776":33486596,"777":33486606}},"haElectricalMeasurement":{"attributes":{"acPowerDivisor":10,"acPowerMultiplier":1}},"genOnOff":{"attributes":{"245":34537472,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":258,"inClusterList":[18,5,4,3,6,64704],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"245":34537472,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":258,"inClusterList":[18,5,4,3,64704],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":258,"inClusterList":[18,5,4,3,64704],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":37,"stackVersion":27,"hwVersion":1,"dateCode":"20241106","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1739808813679}

### Zigbee2MQTT version

2.1.1

### Comments

Please can you add support for the Aqara Display V1 switch.

Currently only the 2 relays come through as switches, none of the wireless buttons or display options.

### External definition

```shell
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.switch.aeu001'],
    model: 'lumi.switch.aeu001',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4,"21":21}}), m.electricityMeter(), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Missing wireless switches and display options

amoritan, 2025-03-31:
👍

PayneRae, 2025-04-19:
Hello @mike-nott, I was wondering if there have been any updates on this issue. I am interested in working on it. Have you attempted to extend the generated external definition?

mike-nott, 2025-04-19:
Hi Payne, unfortunately no updates. I'm very keen to help, so please message me exactly what you need.

PayneRae, 2025-04-25:
Good afternoon Mike-nott. I wanted to inform you that I am no longer able to work on this issue in the near future. If I had to recommend something for you or another contributor to try, it would be to _extend the external definition_ as described here: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition
Thank you for the opportunity to contribute.

mike-nott, 2025-06-25:
I'm still hoping that this device will be added to Z2M at some point ??

mojiro, 2026-01-18:
Hello @N0Klu3 ,
There many steps before this list of attributes becomes useful in zigbee2mqtt.

To make them usable, we need to convert them to an external definition file that zigbee2mqtt understands. That will take some time.

mojiro, 2026-01-18:
[image]

We 'got a demo :)
It will take time to compile and debug all attributes, but it is coming :)

mojiro, 2026-01-18:
Take a look to this external converter.
[aqara_display_switch_v1_eu.js](https://github.com/user-attachments/files/24699226/aqara_display_switch_v1_eu.js)

**Instructions**
Assuming that everyone uses Home Assistant..
1. Open the File editor
2. Create the folder *external_converters* under the folder *zigbee2mqtt*
3. Create a new file *aqara_display_switch_v1_eu.js* under the folder *external_converters*
4. Place there the contents of the attached file and save. Then restart Zigbee2MQTT.

**What works**
Pretty much the majority of the operations :)
1. Button assignment
2. Button names and icons
3. Actions (single-click only)
4. Most of the settings

**What does not work**
1. The same color is assigned to all buttons. I think it is possible to have separate colors. The color attributes will be revisited.
2. *double-click* or *hold* actions do not seem to be supported.
3. Weather information on display. That one requires a lot of debugging but it seems possible to make it work.
4. Sensor information on display. Once you use Aqara's hub, the display can show information from Aqara Sensors. I do not have any. Does anyone has made it work? Help me find a sensor that it can be paired (through Aqara hub) and I will be able to figure how this works too.
5. Attribute reading - Right now, all attributes are write-only. I do not know how to "read" yet values.

mojiro, 2026-01-19:
Hello @mike-nott, it seems that:
1. The relays do operate from both Zigbee2MQTT and Home Assistant.
2. The `action` attribute while it is reactive in Zigbee2MQTT, it does not pass to Home Assistant.

[image]

^^ That `action` will help you trigger automations, etc.

mojiro, 2026-01-20:
I implemented everything that I was able to achieve. The switch is now fully operational with Zigbee2MQTT and Home Assistant. I will soon submit a PR so to have it officially supported too.

The only feature that I am unable to implement is the temperature sensor binding, since I need to get an Aqara T1 for that.

Let me know how it works for you :)

N0Klu3, 2026-01-20:
@mojiro thank you so much for your efforts. 

Out of curiosity is there any way to get this into ZHA? 
Or do a PR there too if possible?

mojiro, 2026-01-20:
ZHA? nope... ZHA is extremely limited in options. I highly recommend you to migrate to Zigbee2MQTT.

N0Klu3, 2026-01-20:
> ZHA? nope... ZHA is extremely limited in options. I highly recommend you to migrate to Zigbee2MQTT.

Haha well I guess I know what I’m doing this weekend

mojiro, 2026-04-23:
Hello @mvidarsson87-ui , I am unsure if I can help, but it could help if you post some logs, some more evidence showing what you see.

JuliusZet, 2026-06-06:
Do you also have the 1 second delay between pressing a button and the switch displaying "Executing" and sending the Zigbee signal?

I suspect this happens because the switch firmware is waiting 1 second for the user to press the same button again, because then it would register as a double tap. But I am not sure.

Do you think that there is a way to make the switch send the Zigbee signal instantly when a button is pressed?

lukaskoeni, 2026-09-07:
Hi all,

is anybody still analyzing this 1sec delay issue? It keeps annoying me :). I have no idea how to fix it though. Happy to keep the momentum on this one...

cyraxx, 2026-11-09:
I think this should be kept open, and ideally merged in modified form.

I've been using the patched version from my [latest comment](https://github.com/Koenkk/zigbee2mqtt/issues/26407#issuecomment-4865143484) for the last months and it's been working absolutely great.

The only thing missing is figuring out the button press delay, as noted in the previous comment.

Rodov, 2026-11-09:
Revision 24 (Timezone UI added)

[aqara_display_switch_v1_eu.js](https://github.com/user-attachments/files/34579260/aqara_display_switch_v1_eu.js)

Picture of the switch for Zigbee2MQTT

[image]

Rodov, 2026-11-11:
I’ve done some research and reverse engineering, and I’ve figured out how to get data from the motion sensor. I’ll be posting an updated converter soon.

Rodov, 2026-11-14:
### **Aqara display switch V1 EU converter revision 25**
<br>

‼️ _Proximity sensor added_ ‼️
[aqara_display_switch_v1_eu_rev25.js](https://github.com/user-attachments/files/34834540/aqara_display_switch_v1_eu_rev25.js)

<br>
<details>
  <summary>Full change log: rev23 → rev25</summary>
<br>

**Proximity Sensor (occupancy)**

- Implemented a state machine for attribute 0x0142: the device only sends 1 (the flag gets stuck), and the converter resets it automatically.
- Added a hardware cooldown of 60 seconds with a 2-second margin; repeated 1 signals within this window are ignored as echoes.
- State is now published as a standard occupancy sensor (fully compatible with Home Assistant).
- Added e.occupancy() expose.
- Timers are now correctly cleared on stop / deviceRemoved events and during Z2M startup.

**Clock & Timezone**

- Added a per-device timezone field with UI input support.
- Added support for IANA timezone names (Europe/Madrid, America/New_York, Asia/Kolkata, etc.) with automatic DST handling.
- Added support for fixed offsets (UTC+3, UTC-05:30, +03:00) as a fallback if the Node.js environment lacks ICU timezone data.
- Added an ICU data availability check (ICU_ZONES_SUPPORTED) with a console warning.
- Changed the default timezone to UTC (removed the hardcoded dependency on process.env.TZ).
- Implemented persistent timezone storage in three locations: globalStore, device.meta, and a JSON file (aqara_display_switch_v1_eu.settings.json) to survive restarts.
- Added auto-detection for the first writable directory for the settings file (ZIGBEE2MQTT_DATA, __dirname, os.tmpdir()).
- Set GEN_TIME_MODE = "local" (verified on the device): Time and LocalTime now carry local time, and TimeZone = 0.
- Added diagnostic exposes: timezone_effective, clock_utc_offset, and clock_last_sync.
- Added a CLOCK_DEBUG flag for synchronization logging.
- Ensured the attachGenTimeLocalReadResponse hook is attached to the device exactly once via AEU001_HOOK_KEY, with additional guarantees from fromZigbee and configure.

 **Weather Signage (weather_*)**

- All weather_* attributes from a single message are now sent as a unified block in the correct order: temperature → humidity (not implemented yet) → condition → null packet.
- Added protection against reprocessing the same message using a processedKey in globalStore.
- Separated sensor_* logic, which continues to use the legacy initialization scheme.

**Utilities & Code Quality**

- Added isOn() and clamp255() helper functions for input value normalization.
- Added section divider comments to improve code readability.
- Added start / stop / deviceRemoved event handlers in onEvent for timezone cache warming and proximity timer cleanup.

</details>

<details>
  <summary>Custom icon for the device</summary>

[image]

</details>

<details>
  <summary>Connecting a converter to Zigbee2MQTT and setup custom icon</summary>
<br>

### **1.	Create folders**

You need to create the `external_converters` and `device_icons` folders in the zigbee2mqtt directory.
The path to the folder depends on the installation method:

- If zigbee2mqtt is installed directly on the server, the folder is usually located at (for non-HassOS systems):

`/opt/zigbee2mqtt/data`

(for HassOS):

`/config/zigbee2mqtt/`
[... 38 lines cut ...]
**1. Light Control (Button 2) (just for example)**

When the second button is pressed (`single_button_2`), the flow sends a toggle command to a specified light entity (`light.your_light`).

**2. Screen Saver Toggle (Button 4)**

Pressing the fourth button (`single_button_4`) switches the display mode on the switch between Clock (`digital_clock`) and Weather (`weather_conditions`).
The flow also dynamically updates the text label of the 4th button on the switch's screen: if the weather is currently displayed, the button will be labeled "Clock" (to indicate it will switch to the clock), and vice versa.

**3. The flow fetches weather data from the Met.no integration (weather.forecast_home_assistant).**

Update Triggers: The weather on the switch display updates automatically every 15 minutes or instantly upon any state change of the weather entity in Home Assistant.
Data Conversion: The HA → Aqara function translates standard Home Assistant weather states and Met.no symbol codes (e.g., `lightrain`, `partlycloudy`) into specific codes understood by the Aqara display (e.g., `light_rain`, `partly_cloudy`), and also extracts and passes the current outdoor temperature.

**4. Sending Commands to the Device**

All changes (weather updates, screen saver mode changes, button renaming) are published back to the MQTT topic `zigbee2mqtt/your_aqara_switch/set`, so the physical switch can update its display accordingly.

**⚠️ Important Note Before Running:**

The flow uses placeholder names. Before importing and using it, you need to replace them with your actual entities:
**MQTT topics:** `zigbee2mqtt/your_aqara_display_switch` and `zigbee2mqtt/your_aqara_switch/set` (replace with your real friendly_name, for example, `zigbee2mqtt/Switch_in_the_room` or `zigbee2mqtt/bathroom_switch`).
**Light entity:** `light.your_light` (replace with your real entity)
**Screen saver selector:** `select.your_aqara_switch_screen_saver_style`
</details>