Koenkk/zigbee2mqtt #26410: Zigbee2mqtt container can't find mqtt container unless running in `network_mode: host`

VorpalBlade, 2025-02-17:
### What happened?

I was trying to secure my docker setup (put everything between traefik with authentication). As part of that I tried moving `zigbee2mqtt` from using `network_mode: host`. However this fails as it for some unknown reason makes zigbee2mqtt unable to connect to the mqtt container.

It appears it is simply unable to use Docker name resolution to resolve the IP of the other container perhaps?

All I'm getting are lines like in the log:

```
[2025-02-17 19:05:14] info:     z2m: Connecting to MQTT server at mqtt:///mqtt
[2025-02-17 19:05:14] error:    z2m: MQTT failed to connect, exiting... ()
```

This is the configuration I'm trying to use, if I comment out `ports` and uncomment `network_mode` it works:

```yml
services:
  zigbee2mqtt:
    user: 1003:1003
    image: docker.io/koenkk/zigbee2mqtt
    restart: unless-stopped
    volumes:
      - /srv/hass/zigbee2mqtt:/app/data
    ports:
      # Frontend port
      - 8080:8080
    #network_mode: host
    depends_on:
      - mqtt
    environment:
      - TZ=Europe/Stockholm
    devices:
      - /dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_UUID_REMOVED-if00-port0:/dev/ttySky:rw
    group_add:
      - dialout

  mqtt:
    user: 1003:1003
    image: eclipse-mosquitto:latest
    restart: unless-stopped
    ports:
      - 1883:1883
    volumes:
      - /srv/hass/mosquitto/config:/mosquitto/config
      - /srv/hass/mosquitto/data:/mosquitto/data
      - /srv/hass/mosquitto/log:/mosquitto/log

  # Homeassistant removed, not relevant to this issue
```

In my configuration for Zigbee2mqtt I have the folllowing lines:

```yml
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt:///mqtt
  user: USER_REMOVED
  password: PASSWORD_REMOVED
```

This exact config works in host network mode, but not in the default bridge mode.
On the host itself I cannot resolve mqtt of course (so that this config works must be some docker magic).

Worth noting is that HASS itself has no problem connecting to the MQTT broker, and it is **not** in host network mode.

### What did you expect to happen?

Zigbee2mqtt should (just like Home Assistant) be able to connect to the MQTT container without depending on host network mode.

### How to reproduce it (minimal and precise)

Use the following docker compose:

```yml
services:
  zigbee2mqtt:
    user: 1003:1003
    image: docker.io/koenkk/zigbee2mqtt
    restart: unless-stopped
    volumes:
      - data:/app/data
    ports:
      # Frontend port
      - 8080:8080
    #network_mode: host
    depends_on:
      - mqtt
    environment:
      - TZ=Europe/Stockholm
    devices:
      - /dev/serial/by-id/usb-Nabu_Casa_SkyConnect_v1.0_UUID_REMOVED-if00-port0:/dev/ttySky:rw
    group_add:
      - dialout

  mqtt:
    user: 1003:1003
    image: eclipse-mosquitto:latest
    restart: unless-stopped
    ports:
      - 1883:1883
    volumes:
      - ./mqtt:/mosquitto/config
```

* Adjust adapter device as needed to work for your Zigbee adapter.

Create a configuration files:

```bash
mkdir mqtt
echo "listener 1883" > mqtt/mosquitto.conf
echo "allow_anonymous true" >> mqtt/mosquitto.conf

mkdir data
cat > data/configuration.yaml <<EOF
homeassistant:
  enabled: true
frontend:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt:///mqtt
serial:
  adapter: ember
  port: /dev/ttySky
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
availability:
  enabled: false
version: 4
EOF
```

Start containers with `docker compose up`.

### Zigbee2MQTT version

2.1.1 commit: 1accb8b66b95b99658c50e4c0209edcb946a5fe2

### Adapter firmware version

7.4.0 [GA]

### Adapter

Nabu_Casa_SkyConnect (using the ember driver)

### Setup

Raspberry Pi 5 (8 GB) running Raspbian as a docker container (I have a lot of non HASS stuff on there, I don't let it take over my OS)

### Debug log

[z2m_debuglog_redacted.log](https://github.com/user-attachments/files/18831338/z2m_debuglog_redacted.log)

chris-1243, 2025-02-17:
You have an error in your `configuration.yaml`

> mqtt:
  ...
    server: mqtt:///mqtt
  ...

This can not work. Instead, this should do the trick

```
mqtt:
  ...
  server: mqtt://mqtt
  ...
```

I am on Docker and behind Traefik and I have no issues at all. I would suggest not to use the default bridge network. It is better to create your on network.

VorpalBlade, 2025-02-17:
@chris-1243  Ah you are right. Thanks.

Very strange that it worked in host network mode though (and not a very helpful error).

VorpalBlade, 2025-02-17:
I believe the issue can be closed (though the diagnostics could be improved, if the developers want to improve it, they can reopen I guess).

closed 2025-02-17 by VorpalBlade