Koenkk/zigbee2mqtt #26412: [Wrong device]: Tuya ZTH01 (_TZE284_yjjdcqsq) recognized as TS0601

FREE2WIN2, 2025-02-17:
### Link

https://www.amazon.de/dp/B0BWJGRHFW?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1

### Model

ZTH01

### Description

Temperature and Hygrometer

### Vendor

Tuya

### Picture (link)

https://m.media-amazon.com/images/I/61CxPTQlN6L._SL1500_.jpg

### Database entry

{"id":10,"type":"EndDevice","ieeeAddr":"0xa4c138d9304249a4","nwkAddr":18838,"manufId":4417,"manufName":"_TZE284_yjjdcqsq","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE284_yjjdcqsq","powerSource":3,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1739824702875}

thetodd, 2025-02-20:
I ran in the same problem after purchasing it on Amazon this week. Got it working with an external converter based on an already [existing one](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts) in the zigbee-herdsman-converters repo. I'm currently trying to integrate my configuration into the device configuration from zigbee-herdsman-converters and post a pull request for that.

For now here is my external converter:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE284_yjjdcqsq',
        },
    ],
    model: 'TS0601_temperature_humidity_sensor_2',
    vendor: 'Tuya',
    description: 'Temperature and humidity sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({queryOnDeviceAnnounce: true}),
    configure: tuya.configureMagicPacket,
    exposes: [e.temperature(), e.humidity(), tuya.exposes.batteryState(), tuya.exposes.temperatureUnit()],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'temperature', tuya.valueConverter.divideBy10],
            [2, 'humidity', tuya.valueConverter.raw],
            [3, 'battery_state', tuya.valueConverter.batteryState],
            [9, 'temperature_unit', tuya.valueConverter.temperatureUnitEnum],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

FREE2WIN2, 2025-02-20:
It worked for me but with a tiny adjustment
Instead of setting the model to 
`model: 'TS0601_temperature_humidity_sensor_2',`

to
`model: 'Tuya ZTH01',`

Koenkk, 2025-02-21:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-02-21 by Koenkk