Koenkk/zigbee2mqtt #26413: zh:ember: Delivery of BROADCAST failed for '65535'.

jeroenst, 2025-02-18:
### What happened?

I get the messages zh:ember: Delivery of BROADCAST failed for '65535' multiple times per minute, what is wrong?

### What did you expect to happen?

This error not occuring

### How to reproduce it (minimal and precise)

Just run zigbee2mqtt with a Sonof Zigbee Dongle E with EmberZNet

### Zigbee2MQTT version

2.1.1 commit: 1accb8b66b95b99658c50e4c0209edcb946a5fe2

### Adapter firmware version

 8.0.2 [GA]

### Adapter

EmberZNet (Sonoff Zigbee Dongle E)

### Setup

Running on Synology DS224+ in Docker

### Debug log

[log.log](https://github.com/user-attachments/files/18840956/log.log)

Nerivec, 2025-02-18:
This usually happens when you are getting interference somewhere on the network, or at the adapter's USB.
Make sure to follow the usual steps to lower interference to a minimum: https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html
Start by trying another channel than 11, you might be getting some conflicts with nearby WiFi networks: https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#configurator

_Make sure you use the latest 8.0.2 firmware https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update6 just in case._

Seems device `17426` (which appears to be `Bulb_Livingroom_Dresser`) is a bit spammy with broadcasts. Can you see if the problem subsides after removing it from the network (and powering it off completely)?

jeroenst, 2025-03-02:
Thank you, I don't know why but the message now comes only once in a while, I guess that's not an issue since everything works fine.

Nerivec, 2025-03-02:
Good to know. Might be passing interference of some kind (another device using the 2.4GHz band that is being moved around the house, and _only sometimes_ causes troubles).

closed 2025-03-02 by Nerivec