Koenkk/zigbee2mqtt #26414: Missing refresh_colored action for EGLO 99099

szamozoli, 2025-02-18:
### What happened?

Pressing the colored refresh button on [EGLO 99099 remote](https://www.zigbee2mqtt.io/devices/99099.html) ('ERCU_3groups_Zm') shows null action instead of "refresh_colored". However the fz.awox_refreshColored function is called for inside the zigbee-herdsman-converter.
I am using the latest available firmware for the remote.

### Zigbee2MQTT version

2.1.1-1

### Adapter firmware version

7.4.4

### Adapter

Home Assistant Connect ZBT-1 using the new ember driver

### Setup

Running Zigbee2MQTT as an add-on using an Intel NUC11

ghoz, 2025-03-02:
Hi,
This is most likely the breaking change due to the removal of the legacy actions sensors from defaults in z2m 2.0.0 : https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> All action sensors are now disabled by default (sensor.*_action entities). It's recommended to use the MQTT device trigger instead. In case you really need the action sensors, add the following to your [zigbee2mqtt] configuration.yaml.
> ```
> homeassistant:
>   legacy_action_sensor: true
> ```

coldridj, 2025-03-03:
@ghoz - are these action sensors HA specific, if so I don't believe this is the issue as I have this device an use z2m without HA integration and see no action property emitted in the mqtt payload when this button is pressed.

I will attempt to investigate this when I get time as I am also seeing the issue described in #21664

szamozoli, 2025-03-03:
Hi,
@ghoz thanks for the tip, but I already have this option enabled, which lets me see the action sensor inside HA.
All the buttons are visible in this action sensor, excluding the colored refresh button:

[image]

@coldridj I also observed the sporadic "on" messages using this remote.

closed 2025-05-10 by github-actions[bot]