Koenkk/zigbee2mqtt #26420: Ember - Skyconnect - Z2M stopped with mutilple errors

palmitoto, 2025-02-18:
### What happened?

Z2M stopped after multiple watchdog relaunch.

### Zigbee2MQTT version

2.1.1  commit: unknown

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet

### Setup

Pi 3B

### Debug log

Hello,

I’ve recently noticed increasing instability with Ember coordinators. I can’t identify the root cause, but I see my issues reflected in multiple tickets in this repository. I’ve tried several firmware versions, including 7.4.4.4 and 8.0.2. I’ve also tested the edge version multiple times. I didn’t have this kind of error a few weeks ago. Could it be caused by a specific device?

However, in all cases, after a while, Z2M stops running without displaying any significant errors. Here are the types of errors I encounter:

```
[2025-02-17 16:06:24] error:  zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-02-17 16:06:26] error:  zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-02-17 16:06:28] error:  zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-02-17 16:06:30] error:  zh:ember: Adapter fatal error: HOST_FATAL_ERROR
```
Often i have this error : 
```
error: 	zh:ember:uart:ash: Received ERROR from adapter while connecting, with code=RESET_WATCHDOG.
```

And a complete log trace where, in the end, Z2M is stopped:

🔗 https://pastebin.com/BGHM8MFu

Even when the network is up, I notice latency issues, making it unreliable.

I imagine I’m not the only one experiencing this with a SkyConnect. What version of Z2M are you using, and which firmware? What would you recommend I do? Should I consider switching to another coordinator?

Thanks in advance for your help.

Nerivec, 2025-02-18:
`RESET_WATCHDOG` happens only on startup due to the reconnection to the adapter. It shouldn't affect anything after that.

#26391 is likely unrelated (route errors can happen even during normal operation).

`ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT` could mean interference (or https://github.com/Koenkk/zigbee2mqtt/issues/23861 which should be fixed with most recent 8.0.2 releases). Make sure you use the latest firmware: https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update6 

You could also have interference that isn't affecting the adapter itself, but some routers on the network (which could produce just as much latency).

The pi3 has been known to cause issues when using Z2M on HAOS. It no longer has enough RAM to sustain HA, and this translates into interference-like problems with Z2M.

_Since I believe you switched firmware type recently (NCP<>router), assuming this is the same adapter, you may need to clear the NVM3 to ensure nothing from the previous network is messing with the new one. You can do that with [ember-zli](https://github.com/Nerivec/ember-zli/wiki/Bootloader#clear-nvm3) or by flashing the [appropriate GBL](https://github.com/Nerivec/silabs-firmware-recovery?tab=readme-ov-file#nvm3-clear) (both do the same). Note: this erases the current network on the adapter; you have to re-pair everything after._

palmitoto, 2025-02-20:
Thank you so much @Nerivec for taking the time to respond with such precision. I was indeed on the latest firmware (update6).

I understand the limitations of the Pi3, I ordered a NUC and will start a clean installation.

I did indeed switch to a router firmware by mistake, so I will follow your advice and clear the NVM3!

Thank you very much!

closed 2025-02-20 by palmitoto