Koenkk/zigbee2mqtt #26422: [New device support]: thermostat TOUCHMI (TS0601/_TZE204_atdqo4nj)

sgilyin, 2025-02-18:
### Link

https://ozon.ru/t/Klnd2Q

### Database entry

  const definition = {     zigbeeModel: ['TS0601'],     model: 'TS0601',     vendor: '_TZE204_atdqo4nj',     description: 'Automatically generated definition',     extend: [],     meta: {}, };  module.exports = definition;

### Zigbee2MQTT version

2.1.1-1

### Comments

How can I help to support new thermostat into z2m?

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_atdqo4nj',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Nothing works 😕

sgilyin, 2025-02-24:
Work: 
- On/Off
- Local temperature
- Current heating setpoint

Not work:
- Calibration
- Current Running state
- Preset

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_atdqo4nj',
        },
    ],
    model: 'ZG-WK-DA-Wh-Zigbee',
    vendor: 'Tuya',
    description: 'Fill in a description of the device here',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.humidity(),
        e
            .climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'auto', 'temporary_manual'])
            .withSetpoint('current_heating_setpoint', 15, 35, 0.5, ea.STATE_SET)
            .withRunningState(['idle', 'heat'], ea.STATE)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET),
        e
            .min_temperature_limit() // 11
            .withUnit('°C')
            .withValueMin(0)
            .withValueMax(99)
            .withValueStep(0.5)
            .withPreset('default', 15, 'Default value'),
        e
            .max_temperature_limit() // 12
            .withUnit('°C')
            .withValueMin(0)
            .withValueMax(99)
            .withValueStep(0.5)
            .withPreset('default', 35, 'Default value'),
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [1, 'system_mode', tuya.valueConverterBasic.lookup({heat: true, off: false})],
            [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [3, 'local_temperature', tuya.valueConverter.divideBy10],
            [11, 'min_temperature_limit', tuya.valueConverter.divideBy10],
            [12, 'max_temperature_limit', tuya.valueConverter.divideBy10],
//            [19, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration3],
//            [6, 'running_state', tuya.valueConverterBasic.lookup({heat: 1, idle: 0})],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
`

sgilyin, 2025-02-24:
@Koenkk fyi

Koenkk, 2025-02-25:
For adding the missing functionality you would have to find the [datapoints](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html). If you are not able to do that, it is also fine to create a PR without it, you can make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

sgilyin, 2025-03-24:
> For adding the missing functionality you would have to find the [datapoints](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html).

I haven't Tuya Gateway :'(

> If you are not able to do that, it is also fine to create a PR without it, you can make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9058

closed 2025-05-31 by github-actions[bot]

AcidReptile, 2026-10-07:
https://github.com/Koenkk/zigbee2mqtt/issues/33311