Koenkk/zigbee2mqtt #26429: [Feature request]: Startup without adapter

dirstel, 2025-02-19:
### Is your feature request related to a problem? Please describe

Onboarding for new users is inconvenient: Instead of configuring z2m by using the interface, z2m does not start without a configured adapter.

### Describe the solution you'd like

Instead of exiting without a configured adapter base functionality (using the frontend, answering mqtt-topics related to z2m, etc.) should be given to allow the user to configure z2m without editing .conf-files. 

### Describe alternatives you've considered

Maybe implementing a "dummy" adapter (without any function but answering the right calls) is a lean way to accomplish a startup without too much effort.

### Additional context

More and more distributions tend to be user-friendly and provide an app-like experience (f.e. [umbrel](https://umbrel.com/umbrelos) or [casa os](https://casaos.io/)). z2m should join this to be more accessible to users with lower linux skills.

Nerivec, 2025-03-05:
https://github.com/Koenkk/zigbee2mqtt/pull/26608

If you have any notes on current features... 😉

CC: @bjornfor, @hoppel118, @raducotescu

dirstel, 2025-03-06:
> [#26608](https://github.com/Koenkk/zigbee2mqtt/pull/26608)

Very nice! I think, that should "kill that cat"! :)

closed 2025-04-28 by dirstel