Koenkk/zigbee2mqtt #26430: Device is preventing Zigbee2Mqtt to start

ahattink, 2025-02-19:
### What happened?

When updating to version 2.0 the issue started. Actually, the Zigbee2MQTT add-on did not start anymore.
So, I downgraded and it worked.

Tried again with version 2.1.1.1 and the issue is still there. So started googling and stumbled upon this discussion: https://github.com/Koenkk/zigbee2mqtt/discussions/24198

When deleting a specific device from the database.db ( Centralite 3323-D Micro Door Sensor) the add-on starts. So there is an issue.

### What did you expect to happen?

A completed update and working add-on.

### How to reproduce it (minimal and precise)

I started with a working 2.1.1.1 version.
Paired the Centralite 3323-D Micro Door Sensor. This device is not supported, so I have the converter in place.
The device is paired and works fine.
Restart Zigbee2Mqtt add-on.
Add-on won't start.

Take note I have also other Centralite devices. The 3328-D is a motion sensor. Also needs an external converter. This one works also fine.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

zStack3x0 (ZigStar V4 USB)

### Setup

Add-on on Home Assistant OS on a NUC

### Debug log

Zigbee2MQTT log
[log.log](https://github.com/user-attachments/files/18867219/log.log)

manuelkue, 2025-02-19:
Thanks for creating the issue @ahattink

I came up with the idea of removing devices from the `database.db` as it has already helped all the people in the following issue that had the same problem with thermostats of the company "Danfoss".
They're sold by different names: [Danfoss Ally](https://www.zigbee2mqtt.io/devices/014G2461.html), [Popp 701721](https://www.zigbee2mqtt.io/devices/701721.html), [Hive UK7004240](https://www.zigbee2mqtt.io/devices/UK7004240.html#hive-uk7004240)

Previous issue with another topic but some people that all had the same problem: https://github.com/Koenkk/zigbee2mqtt/issues/25715#issuecomment-2593735141

Koenkk, 2025-02-20:
> Paired the Centralite 3323-D Micro Door Sensor. This device is not supported, so I have the converter in place.
The device is paired and works fine.

Without the external converter in place, does z2m start? Also, could you share your `data/database.db` file?

ahattink, 2025-02-20:
@Koenkk ,
I tried starting z2m when the converter was absent in the data map. (deleted all converter files in all maps). Still having the devices paired. Z2M did not start.
Attached the database.db file. I had to rename it to .txt. *.db files can not be uploaded.
[databas.db.txt](https://github.com/user-attachments/files/18887933/databas.db.txt)
ID's 21, 24 and 25 are the devices.
To be complete, Also the converter file: (also renamed)
[3323-D.js.txt](https://github.com/user-attachments/files/18887950/3323-D.js.txt)

If I can be of further assistance, please let me know!

manuelkue, 2025-02-20:
@ahattink can you try to re-pair the faulty devices, but not directly to the coordinator? So when switching on the pairing mode of z2m, select a router device in the drop down (if you have a router device), that's not the coordinator. 

Does z2m then restart without the issue?

There's currently another issue where the faulty devices (in that case the Popp thermostats) work without problems  when they're not connected to the coordinator directly:
https://github.com/Koenkk/zigbee2mqtt/issues/26450

ahattink, 2025-02-21:
@manuelkue I did as you asked.
Enabled joining on the closest Nous A1Z device.
Paired the Centralite 3323-D door sensor
Checked working of the device => worked
Restarted Z2M add-on
Waited 10 minutes to see if Z2M worked => No. Z2M did not start.

Stopped Z2M, deleted the Centralite 3323-D from database.db, restarted Z2M. Z2M started normally.

So far as expected by me...

I noticed an error during paring of my device, which i didn't notice before.
```
2025-02-21 08:10:32z2m: Failed to configure 'Schuurdeur', attempt 3 (ReferenceError: reporting is not defined at Object.configure 
(/app/data/extension/3323-D.js:18:9) at Configure.configure (/app/lib/extension/configure.ts:124:37) at /app/lib/extension/
configure.ts:82:62 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:23) at EventEmitter.emit (node:events:536:35) at 
EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:74:22) at Controller.<anonymous> (/app/lib/zigbee.ts:79:27) at Controller.emit 
(node:events:524:28) at Controller.selfAndDeviceEmit (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-
herdsman/src/controller/controller.ts:671:14) at Controller.onZclPayload (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/
node_modules/zigbee-herdsman/src/controller/controller.ts:945:18))
``` 
I will try to find out if this can be a cause of my issue.

ahattink, 2025-02-21:
Ok,

Found out I needed to add a line in the external converter.
```
const reporting = require('zigbee-herdsman-converters/lib/reporting');

``` 
So I did.
Restarted Z2M
Enabled joining on the closest Nous A1Z device.
Paired the Centralite 3323-D door sensor => Error message gone.
Checked working of the device => worked
Restarted Z2M add-on
Waited 10 minutes to see if Z2M worked => No. Z2M did not start.

I think this was not the issue. Still, something ruled out!

@Koenkk , to be complete, here the updated external converter
[3323-D.js.txt](https://github.com/user-attachments/files/18903162/3323-D.js.txt)

Koenkk, 2025-02-22:
Fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-02-22 by Koenkk