Koenkk/zigbee2mqtt #26435: _TZE284_dikb3dp6

anrada73, 2025-02-19:
### Link

https://de.aliexpress.com/item/1005006801446920.html?spm=a2g0o.order_list.order_list_main.5.6c5b5c5fjBzxan&gatewayAdapt=glo2deu

### Database entry

 const definition = {     zigbeeModel: ['TS0601'],     model: 'TS0601',     vendor: '_TZE284_dikb3dp6',     description: 'Automatically generated definition',     extend: [],     meta: {}, };  module.exports = definition;

### Zigbee2MQTT version

2.1.1-1 

### Comments

dont support

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_dikb3dp6',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

dont support

Koenkk, 2025-02-20:
`_TZE284_dikb3dp6` will be supported in the Z2M 1 March release.

closed 2025-02-20 by Koenkk

anrada73, 2025-03-01:
i see no release.