Koenkk/zigbee2mqtt #26438: After recent update zigbee network became unstable

cracrama, 2025-02-19:
### What happened?

Hi
Im not sure what happened but since recent update ZigBee network becomes extremely unstable - devices randomly disconnecting from network or not reporting. 
If device happen to work then there is also problem with serious delays in triggered actions.
I have tried to reset Mosquito, z2m but it seems like nothing works.
Thanks in advance for response.

### What did you expect to happen?

Stable and quick network functionality. 

### How to reproduce it (minimal and precise)

Constant issue. 

### Zigbee2MQTT version

 2.1.1 

### Adapter firmware version

20230507

### Adapter

Sonof ZBDongle-P

### Setup

HA Addon

Zortris, 2025-02-19:
i have the same problem!

Soukyuu, 2025-02-19:
Me as well, and of course my 1.40.x backup is corrupted. It's somewhat better on dev branch, but the symptoms are the same: network quality deteriorating over the course of time, with first some routers becoming randomly unresponsive, then followed by what seems to be duplicated events being reported for the hue v1 dimmers - single click code becomes unreachable and the double click code ia always being executed. Restarting the addon seems to fix it most of the times.

mrmaximas, 2025-02-20:
https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20240710 will most likely help you

xammmue, 2025-02-20:
> edit: the duplicate event thing seems to be [#26374](https://github.com/Koenkk/zigbee2mqtt/issues/26374), I'll update to the new version and see report back there if it fixes things.

The fix works for the dimmer switches as far as I could test. Unfortunately it doesn't for hue smart buttons. I have the same issue with them as you have, I can only execute my double click actions. Single click and tiple click are not accessible anymore.

A single click on a dimmer switch also still leads to 6 mqtt messages being sent, but most of them are the last seen state, so it doesn't really matter functionality wise

Koenkk, 2025-02-20:
The `20230507` firmware could give some issues, please try 20240710 as @mrmaximas indicated (https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20240710)

cracrama, 2025-02-23:
Hi. Just to share experience after 2 days of testing. 
Yes indeed coordinator firmware 20240710 fixed the problem. Many thanks for suggestions. 
I know that many people have a problem with firmware upgrade but if you use "certain platform" then there is addon ZigStar TI CC2652P/P7 FW Flasher, (Sonof)  which allows to upgrade firmware in completely painless way. I really appreciate work on Z2M since for me this is the most stable platform in home automation and i think that option to upgrade coordinator from Z2M level would be really welcomed. 
Again thanks for all help and amazing work.

closed 2025-02-23 by cracrama