Koenkk/zigbee2mqtt #26439: Latest Firmware not found - Bticino/Legrand NLL etc

glenharris, 2025-02-19:
### What happened?

Zigbee OTA firmware version detection seems intermittent.

- I have a number of Legrand devices that have been upgraded using Z2M, and upgraded successfully (thanks!).
- However, once they upgraded, when I click on 'Check for Updates' it says 'Firmware is newer than latest available firmware'.
- This would not be a problem, except that I have a couple of devices that have not upgraded, and when I click 'Check for Updates' it says 'No Update Available'
- Examples of this are:
1. [https://www.zigbee2mqtt.io/devices/K4003C_L4003C_N4003C_NT4003C.html] Legrand NLL?  Legrand says latest production version is 65 (0x41), https://github.com/Koenkk/zigbee-OTA/tree/master/images/Legrand has this version.  I have 28 at 0x41, 1 at 0x3d, 1 at 0x1c
2. [https://www.zigbee2mqtt.io/devices/4411C_L4411C_N4411C_NT4411C.html] Legrand NLFN?  Legrand says latest production version is 38 (0x26), https://github.com/Koenkk/zigbee-OTA/tree/master/images/Legrand only has 0x25 (I have attached 0x26 from [https://developer.legrand.com/production-firmware-download/] to this issue).  I have 23 at 0x25, 1 at 0x24
Again, all fully upgraded devices (0x41, 0x25) were upgraded via Z2M, but somehow Z2M seems to have forgotten that those versions were applicable.

[1021-0019-00264203-NLFN.zip](https://github.com/user-attachments/files/18875481/1021-0019-00264203-NLFN.zip)

### What did you expect to happen?

All devices would be upgraded to the latest version from [https://github.com/Koenkk/zigbee-OTA]

### How to reproduce it (minimal and precise)

See 'What Happened' above.  Note that in the attached log, I am upgrading a light switch ('R57 Art') that is at version 0x1c (according to the Z2M UI), and from the log it appears to be finding the correct 0x41 version online, yet is reporting no upgrade available.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20230507

### Adapter

Zigbee 3.0 USB Dongle plus (CC2652P)

### Setup

Docker image, x86064

### Debug log

[z2m.log](https://github.com/user-attachments/files/18875477/z2m.log)

glenharris, 2025-02-28:
@Koenkk sorry to ping you directly, but do you think this issue should be lodged with zigbee-OTA instead of zigbee2mqtt?

The logfile at 2025-02-20 09:41:45 indicates that it is downloading firmware file https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/Legrand/1021-000f-00414203-NLL.zigbee (which is version 0x41).  The light is at version 0x1C, and yet at 2025-02-20 09:41:45 we get the messages 'Upgrade available: NO' and  'Firmware is newer than latest available firmware'.

It seems that the problem must be either with metadata coming from zigbee-OTA which has the wrong version number, or logic in zigbee2mqtt that is comparing versions incorrectly.  Any tips on how to triage this further?

glenharris, 2025-03-04:
Anyone have any idea where the 'UI' version (in this case 0x1C') is coming from?  I can see two possible problems that might be unrelated bugs:
1) The UI is displaying that the device has firmware version 0x1C, whereas when I query the device directly using the dev console (endpoint 1, cluster genBasic, read 'swBuildId') I get version 0041 (0x41).  What could cause this 'sticky' mismatch?
2) When z2m queries the devices to initiate a firmware upgrade, the version coming back from the unit is fileVersion 4277759 (or 0x4145FF).  I am pretty sure (but not certain) that this unit was upgraded using zigbee-OTA file 'https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/Legrand/1021-000f-00414203-NLL.zigbee' which is fileVersion 4276739 (or 0x414203, which you can see in the filename).  So either the internal version inside this file is actually 0x4145FF, or the firmware on the device is somehow mangling the fileVersion from 0x414203 to 0x4145FF.  How can I check this?  Any tools that can look inside a '.zigbee' file?

When I look at devices that were *definitely* upgraded by z2m, I see the same pattern in their 'state' "installed_version": 4277759, "latest_version": 4276739.

Nerivec, 2025-03-05:
https://github.com/Koenkk/zigbee-OTA/blob/master/src/print_ota_image_header.ts
This should print the header of any valid ZigBee OTA file.
_This is done automatically for all OTA files in zigbee-OTA and stored in https://github.com/Koenkk/zigbee-OTA/blob/master/index.json for Z2M to retrieve/compare._

From your logs, your version is higher than current in zigbee-OTA, so `NO` is the proper answer:
current: 4277759
Result for latest image availability, meta: 4276739

The firmware version is not used for OTA check, only `fileVersion` is.

Note: it's not actually downloading the firmware, it's just getting the metadata. It only downloads it if it starts actually updating.

glenharris, 2025-03-06:
Thanks for the pointer @Nerivec, that was very helpful.

Useful information for anyone following these breadcrumbs (ts-node seems to barf with later version of node):
```
git clone https://github.com/Koenkk/zigbee-OTA.git
npm install
node --loader ts-node/esm src/print_ota_image_header.ts  images/Legrand/1021-000f-00414203-NLL.zigbee
```
That gives the result
```
{
  otaUpgradeFileIdentifier: <Buffer 1e f1 ee 0b>,
  otaHeaderVersion: 256,
  otaHeaderLength: 56,
  otaHeaderFieldControl: 0,
  manufacturerCode: 4129,
  imageType: 15,
  fileVersion: 4276739,
  zigbeeStackVersion: 2,
  otaHeaderString: '                                ',
  totalImageSize: 254391
}
```
which has the expected fileVersion 4276739 (or 0x414203).  Looking at the binary zigbee file, I am pretty sure this data is stored in little-endian format at offset 0x0E (0x03 42 41 00).

Next steps:
I agree with @Nerivec that z2m is doing the right think by replying NO for the firmware upgrade - current (4277759) is higher than latest (4276739), so no upgrade is warranted.
However, I am 99% certain that the firmware on this unit (and all the other units) was upgraded by z2m over the past month or so.  I sincerely doubt that z2m managed to find version 4277759 somewhere, but has now forgotten about it.
Is it possible that the device firmware is mangling the fileVersion, and telling z2m that it is version 4277759, even though it was upgraded using the 4276739 image?

The only definitive way I can think of to test this is to force z2m to upgrade (downgrade?) one of the units using the 4276739 image, and then see if it is OK (self-identifies as 4276739) or weird (self-identifies as 4277759).  Is it possible for me to use z2m to force a specific image onto a specific device (e.g. hack the metadata to say that it is fileVersion 4300000).

Thanks, Glen.

glenharris, 2025-03-06:
> Also worth noting a lot of legrand device seem to ship with a 'limited' firmware with a very high version number. I had to force upgrade my DIN Power Meter, since then it got 1 more update and it also enabled a bunch of functionality that was missing out of the box.

How did you force-upgrade your power meter?  I think that is what I need to do in order to work out what is happening with my firmware versions.

sjorge, 2025-03-06:
I provided a local ota json, with the image path and force set to true.

I’m not sure that still works with the bug OTA rewrite though.

ascl00, 2025-03-22:
I have the same issue. Worth noting that updating via z2m is what got it into this state, shouldn't be related to the shipped firmware version. I have one switch I'm yet to update:

[image]

glenharris, 2025-03-22:
Ok, this is good. This should let us get conclusive proof about if/where
there is a bug.  Let me know if you need better descriptions of these steps.

   - First, dont upgrade that device yet
   - Make sure you enable debug logging, and can capture those log entries
   (e.g. via docker logs - debug entries won't show up on the web UI by
   default). Maybe test by trying to upgrade existing upgraded devices and
   ensure you are getting useful information.
   - Save the state (device 'state' tab) for this device.  Check if the
   installed and latest version numbers make sense, and in particular if the
   installed version matches the 0x1C that the UI is displaying
   - Upgrade the firmware on the device, making sure you save the log
   (should give us some interesting information)
   - Save the new 'state' of the upgraded device
   - Attempt to upgrade the device again (saving as a separate log file)
   - ...
   - Profit

Lastly, note that I recently proposed that I could do some development of
an enhanced 'firmware' tab that might help with these types of issues,  but
so far there has been no feedback from anyone that might point me in the
right direction...

https://github.com/Koenkk/zigbee2mqtt/discussions/26641

ascl00, 2025-03-23:
Hmm, well this was anti-climatic. Firstly, state as reported in the z2m UI:
`{
    "last_seen": "2025-03-23T12:20:57+11:00",
    "linkquality": 120,
    "state": "ON",
    "update": {
        "installed_version": 1851907,
        "latest_version": 4276739,
        "state": "available"
    },
    "update_available": true
}`

I don't see where the 0x1c comes from, but the about page also shows that:

[image]

Unfortunately, when I do click update, the update fails (which is, tbh, maybe why this specific switch wasn't updated. It's possible I attempted to update all four and this one failed for some reason. It was a while ago, I don't remember the specifics sorry!).

`Mar 23 12:25:20 zigbee npm[734]: [2025-03-23 12:25:20] debug:         z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"LaundryLight","transaction":"tn6k2-1"}'
Mar 23 12:25:20 zigbee npm[734]: [2025-03-23 12:25:20] info:         z2m: Updating 'LaundryLight' to latest firmware
Mar 23 12:25:20 zigbee npm[734]: [2025-03-23 12:25:20] debug:         z2m: Received Zigbee message from 'LaundryLight', type 'readResponse', cluster 'genBasic', data '{"dateCode":" \u000020210622\u0000\u0000\u0000\u0000\u0000","swBuildId":"001c"}' from endpoint 1 with groupID 0
Mar 23 12:25:20 zigbee npm[734]: [2025-03-23 12:25:20] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T12:25:20+11:00","led_if_on":null,"led_in_dark":null,"linkquality":123,"state":"OFF","update":{"installed_version":1851907,"latest_version":4276739,"state":"available"},"update_available":true}'
Mar 23 12:25:41 zigbee npm[734]: [2025-03-23 12:25:41] debug:         z2m: Received Zigbee message from 'LaundryLight', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Mar 23 12:25:41 zigbee npm[734]: [2025-03-23 12:25:41] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T12:25:41+11:00","led_if_on":null,"led_in_dark":null,"linkquality":123,"state":"OFF","update":{"installed_version":1851907,"latest_version":4276739,"state":"available"},"update_available":true}'
Mar 23 12:26:20 zigbee npm[734]: [2025-03-23 12:26:20] debug:         z2m: Update of 'LaundryLight' failed (Error: Device didn't respond to OTA request)
Mar 23 12:26:20 zigbee npm[734]: [2025-03-23 12:26:20] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T12:25:41+11:00","led_if_on":null,"led_in_dark":null,"linkquality":123,"state":"OFF","update":{"installed_version":1851907,"latest_version":4276739,"state":"available"},"update_available":true}'
Mar 23 12:26:20 zigbee npm[734]: [2025-03-23 12:26:20] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'LaundryLight' failed (Device didn't respond to OTA request)","status":"error","transaction":"tn6k2-1"}'
Mar 23 12:26:20 zigbee npm[734]: [2025-03-23 12:26:20] error:         z2m: Update of 'LaundryLight' failed (Device didn't respond to OTA request)`

Full logs attached (spammy!):
[z2m.logs.txt](https://github.com/user-attachments/files/19406560/z2m.logs.txt)

While we are here, it's worth noting that DiningLight:
`{
    "last_seen": "2025-03-23T12:33:03+11:00",
    "linkquality": 116,
    "state": "OFF",
    "update": {
        "installed_version": 4277759,
        "latest_version": 4276739,
        "state": "idle"
    },
    "update_available": false,
    "identify": null,
    "led_if_on": null,
    "led_in_dark": null
}`

and  KitchenDownlight:
`{
    "last_seen": "2025-03-23T12:32:05+11:00",
    "linkquality": 83,
    "state": "OFF",
    "update": {
        "installed_version": 54594304,
        "latest_version": 4276739,
        "state": "idle"
    },
    "update_available": false,
    "identify": null,
    "led_if_on": null,
    "led_in_dark": null
}`

Should be the same as LaundryLight. They should be the same hardware, purchased at the same time, and originally all had the same firmware, and I think likely I would have updated them all about the same time via z2m.

glenharris, 2025-03-23:
Maybe try the update again a few times.  I definitely have had LeGrand
devices take multiple attempts before they upgrade successfully. Sometimes
they don't even start, other times they time out part way through, but all
70 odd of my devices eventually got there.

Worst case, maybe try removing and re-adding to the network?

Glen.

ascl00, 2025-03-23:
Okay some success (kinda). I forced the device to re-join the network, and the update completed:

[image]

This matches 1 of the other 3 devices, but I do still get the same error if I click check for updates (Firmware is newer than latest available firmware.).

Logs look like this:
`Mar 23 16:28:01 zigbee npm[2824481]: [2025-03-23 16:28:01] info:         z2m: Configuring 'LaundryLight'
Mar 23 16:28:01 zigbee npm[2824481]: [2025-03-23 16:28:01] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T16:28:01+11:00","led_if_on":"ON","led_in_dark":"OFF","linkquality":127,"state":"OFF","update":{"installed_version":1851907,"latest_version":4276739,"state":"available"},"update_available":true}'
Mar 23 16:28:01 zigbee npm[2824481]: [2025-03-23 16:28:01] info:         z2m: Successfully configured 'LaundryLight'
Mar 23 16:28:01 zigbee npm[2824481]: [2025-03-23 16:28:01] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"LaundryLight"},"status":"ok","transaction":"orbb4-2"}'
Mar 23 16:28:25 zigbee npm[2824481]: [2025-03-23 16:28:25] debug:         z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"LaundryLight","transaction":"41e7a-4"}'
Mar 23 16:28:25 zigbee npm[2824481]: [2025-03-23 16:28:25] info:         z2m: Updating 'LaundryLight' to latest firmware
Mar 23 16:28:25 zigbee npm[2824481]: [2025-03-23 16:28:25] debug:         z2m: Received Zigbee message from 'LaundryLight', type 'readResponse', cluster 'genBasic', data '{"dateCode":" \u000020210622\u0000\u0000\u0000\u0000\u0000","swBuildId":"001c"}' from endpoint 1 with groupID 0
Mar 23 16:28:25 zigbee npm[2824481]: [2025-03-23 16:28:25] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T16:28:25+11:00","led_if_on":"ON","led_in_dark":"OFF","linkquality":127,"state":"OFF","update":{"installed_version":1851907,"latest_version":4276739,"state":"available"},"update_available":true}'
Mar 23 16:28:25 zigbee npm[2824481]: [2025-03-23 16:28:25] debug:         z2m: Received Zigbee message from 'LaundryLight', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1851907,"imageType":15,"manufacturerCode":4129}' from endpoint 1 with groupID 0
Mar 23 16:28:25 zigbee npm[2824481]: [2025-03-23 16:28:25] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T16:28:25+11:00","led_if_on":"ON","led_in_dark":"OFF","linkquality":127,"state":"OFF","update":{"installed_version":1851907,"latest_version":4276739,"state":"available"},"update_available":true}'
Mar 23 16:28:32 zigbee npm[2824481]: [2025-03-23 16:28:32] debug:         z2m: Received Zigbee message from 'LaundryLight', type 'commandImageBlockRequest', cluster 'genOta', data '{"fieldControl":0,"fileOffset":62,"fileVersion":4276739,"imageType":15,"manufacturerCode":4129,"maximumDataSize":8}' from endpoint 1 with groupID 0
Mar 23 16:28:32 zigbee npm[2824481]: [2025-03-23 16:28:32] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T16:28:32+11:00","led_if_on":"ON","led_in_dark":"OFF","linkquality":127,"state":"OFF","update":{"installed_version":1851907,"latest_version":4276739,"state":"available"},"update_available":true}'
Mar 23 16:28:32 zigbee npm[2824481]: [2025-03-23 16:28:32] info:         z2m: Update of 'LaundryLight' at 0.02%, ≈ 455 minutes remaining
Mar 23 16:28:32 zigbee npm[2824481]: [2025-03-23 16:28:32] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T16:28:32+11:00","led_if_on":"ON","led_in_dark":"OFF","linkquality":127,"state":"OFF","update":{"installed_version":1851907,"latest_version":4276739,"progress":0.02,"remaining":27283,"state":"updating"},"update_available":true}'
Mar 23 16:28:32 zigbee npm[2824481]: [2025-03-23 16:28:32] debug:         z2m: Received Zigbee message from 'LaundryLight', type 'commandImagePageRequest', cluster 'genOta', data '{"fieldControl":0,"fileOffset":70,"fileVersion":4276739,"imageType":15,"manufacturerCode":4129,"maximumDataSize":64,"pageSize":256,"responseSpacing":200}' from endpoint 1 with groupID 0
Mar 23 16:28:32 zigbee npm[2824481]: [2025-03-23 16:28:32] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T16:28:32+11:00","led_if_on":"ON","led_in_dark":"OFF","linkquality":123,"state":"OFF","update":{"installed_version":1851907,"latest_version":4276739,"progress":0.02,"remaining":27283,"state":"updating"},"update_available":true}'
Mar 23 16:28:33 zigbee npm[2824481]: [2025-03-23 16:28:33] debug:         z2m: Received Zigbee message from 'LaundryLight', type 'commandImagePageRequest', cluster 'genOta', data '{"fieldControl":0,"fileOffset":326,"fileVersion":4276739,"imageType":15,"manufacturerCode":4129,"maximumDataSize":64,"pageSize":256,"responseSpacing":200}' from endpoint 1 with groupID 0
Mar 23 16:28:33 zigbee npm[2824481]: [2025-03-23 16:28:33] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T16:28:33+11:00","led_if_on":"ON","led_in_dark":"OFF","linkquality":123,"state":"OFF","update":{"installed_version":1851907,"latest_version" [... cut ...]
Mar 23 16:46:34 zigbee npm[2824481]: [2025-03-23 16:46:34] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LaundryLight', payload '{"identify":null,"last_seen":"2025-03-23T16:46:34+11:00","led_if_on":"ON","led_in_dark":"OFF","linkquality":123,"state":"OFF","update":{"installed_version":4276739,"latest_version":4276739,"state":"idle"},"update_available":true}'
Mar 23 16:46:34 zigbee npm[2824481]: [2025-03-23 16:46:34] info:         z2m: Successfully configured 'LaundryLight'`

If that isn't enough of the logs let me know, the whole log file is 32 megs, so github doesn't want it.

closed 2025-05-30 by github-actions[bot]