Koenkk/zigbee2mqtt #26446: IKEA Trafri LED Dimmers don't expose brightness in HomeAssistant

victorhooi, 2025-02-20:
### What happened?

I have several IKEA Trafri LED dimmers that are connected to Zigbee2MQTT:

https://www.zigbee2mqtt.io/devices/ICPSHC24-10EU-IL-1_ICPSHC24-10EU-IL-2.html

**Zigbee2mqtt version**: 2.1.1-dev commit: [382c89c49eae395da3a452c8851281829967fec7](https://github.com/Koenkk/zigbee2mqtt/commit/382c89c49eae395da3a452c8851281829967fec7)
**HomeAssistant version (Core)**: 2025.3.0.dev202502200229

Previously (e.g. 1-2 weeks ago), the dimmers were working fine in HomeAssistant, and I was able to successfully adjust the brightness.

However, as of a recent zigbee2mqtt update, these are now simply appearing as binary on/off lights in HomeAssistant:

[image]

In the zigbee2mqtt webUI - I can confirm that brightness is still exposed, and seems to work when you directly control it there:

[image]

### What did you expect to happen?

The dimmable LED drivers should exposed brighness controls in HomeAssistant integration.

### How to reproduce it (minimal and precise)

Setup HomeAssistant and latest zigbee2mqtt (dev).

Pair a IKEA Tradfri LED dimmer ([device](https://www.zigbee2mqtt.io/devices/ICPSHC24-10EU-IL-1_ICPSHC24-10EU-IL-2.html))

### Zigbee2MQTT version

2.1.1-dev commit: 382c89c49eae395da3a452c8851281829967fec7

### Adapter firmware version

20240710

### Adapter

UZG-01

### Setup

Docker

### Debug log

[log.log](https://github.com/user-attachments/files/18886793/log.log)

AlexanderLeporiday, 2025-02-20:
If you go to state do you see the brightness value reported there?
If you check the Zigbee2MQTT logs, after making a change to the brightness level, do you see the message reported as a state change that is sent to Home Assistant?
Have you restarted Zigbee2MQTT since updating the dev version?
Have you tried reconfiguring the device since the issue started?
Have you tried pairing the device again since the issue started?

It's possible there's been an underlying change to the converter that requires the configuration to be reloaded for the device.

AlexanderLeporiday, 2025-02-20:
I can see you attached a debug log and from what I can see the following:

> [2025-01-20 11:00:14] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Downstairs Bunk Bed Light', payload '{"brightness":254,"effect":null,"identify":null,"linkquality":7,"power_on_behavior":null,"state":"OFF","update":{"installed_version":587753009,"latest_version":587753009,"state":"idle"}}'

So the MQTT payload has a brightness value, I'm not sure why it wouldn't be appear in home assistant unless something under the hood is causing the topic to no longer include brightness.

Have you tried reloading the MQTT integration or restarting the MQTT service?

itkama, 2025-02-28:
Having the same issue on 3 hue bulbs. Interestingly I have no trouble with Ikea bulbs (yet).

I also seem to have this in my HA logs:
```supported_color_modes list (Optional) A list of color modes supported by the list. Possible color modes are onoff, brightness, color_temp, hs, xy, rgb, rgbw, rgbww, white.  Note that if onoff or brightness are used, that must be the only value in the list.```

closed 2025-05-09 by github-actions[bot]