Koenkk/zigbee2mqtt #26450: Danfoss Ally /Popp TRV problems

Mikkooooo, 2025-02-20:
### What happened?

Hello, I still have the problem that sporadically after some time the Popp/Danfoss Ally TRV's can no longer be controlled, even though they update the status after manual temperature setting.
These are still only those TRVs that are directly connected to the coordinator.
Is there a solution for this now?
Regards, Mikkooooo

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20221226

### Adapter

SLZB-06

### Setup

HomeSeer Hs4

lux73, 2025-02-21:
Update your z2m & Coordinator

Mikkooooo, 2025-02-21:
@lux73,
do you have a stable version for both ?

lux73, 2025-02-21:
Sure, z2m 1.42.0 & Coordinator FW 20240710

no dropouts with my Allys

Mikkooooo, 2025-02-21:
DANKE, lux73 !!!

closed 2025-05-01 by github-actions[bot]

andrzejzukowski, 2026-01-22:
@Mikkooooo have you solved your problem?
I experience it as well.