Koenkk/zigbee2mqtt #26451: Multiple Devices Disconnect After Irregular Intervals  "failed with status '(0x11: BUFFER_FULL)'

chrineuh, 2025-02-20:
### What happened?

It has happened multiple times that some devices in my Zigbee network randomly drop out and do not automatically reconnect.
Even restarting the Zigbee2MQTT adapter in Home Assistant usually does not solve the issue. Instead, I have to unplug the SLZB-06 antenna from power and then restart the network manually to restore normal operation.

[image]

### What did you expect to happen?

I’m not experiencing regular failures, but when it does happen, I would like the system to restart automatically.

### How to reproduce it (minimal and precise)

I can’t identify a clear pattern, as it happens at irregular intervals—sometimes after a day, then not again for several days.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

Radio 20240710 / Core v2.7.1

### Adapter

SLZB-06 zStack3x0

### Setup

Homeassistant Addon in Proxmox CT

### Debug log

Zigbee2Mqtt : 

warning 2025-02-20 17:34:56z2m: Failed to ping 'KG_Klima_PLG' (attempt 1/1, ZCL command 0x84ba20fffe6e75fa/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":33489,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":46,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,119,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
warning 2025-02-20 17:35:01z2m: Failed to ping 'OG_Buero_ROL1' (attempt 1/1, ZCL command 0x54ef441000d6e3e2/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":6998,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":58,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,131,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
warning 2025-02-20 17:35:06z2m: Failed to ping 'EG_Wohnzimmer_ROL1' (attempt 1/1, ZCL command 0x54ef441000b8f30a/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":4224,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":59,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,132,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
warning 2025-02-20 17:35:11z2m: Failed to ping 'KG_Tiefkühler_2_PLG' (attempt 1/1, ZCL command 0xb0c7defffedd5459/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":50785,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":71,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,144,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
warning 2025-02-20 17:35:16z2m: Failed to ping 'KG_Tiefkühler_1_PLG' (attempt 1/1, ZCL command 0x60a423fffeff48e3/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":54363,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":72,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,145,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
warning 2025-02-20 17:35:21z2m: Failed to ping 'OG_Geschirrspüler' (attempt 1/1, ZCL command 0x60a423fffeae69be/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":59097,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":84,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,157,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
warning 2025-02-20 17:35:26z2m: Failed to ping 'EG_Wohnzimmer_RL2' (attempt 1/1, ZCL command 0x54ef441000b37d6d/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) 

SLZ06B

[00:00:05] stats | Statistics sent
[00:00:05] stats | Heap: 2536
[12:32:18] time | timezone: CET-1CEST,M3.5.0,M10.5.0/3
[12:32:18] time | Wednesday, February 19 2025 12:32:18
[12:32:18] zbSelfOta | start task
[12:32:18] taskTimeSync | Heap: 1376
[12:32:18] time | stop task
[12:32:30] taskZB | New client: 192.168.0.10 id: 0
[13:32:18] zbSelfOta | Heap: 4120
[14:32:18] zbSelfOta | Heap: 4072
[15:32:18] zbSelfOta | Heap: 4072
[16:32:18] zbSelfOta | Heap: 4072
[17:32:18] zbSelfOta | Heap: 4072
[18:32:18] zbSelfOta | Heap: 4072
[19:32:18] zbSelfOta | Heap: 4072
[20:32:18] zbSelfOta | Heap: 4072
[21:32:18] zbSelfOta | Heap: 4072
[22:32:18] zbSelfOta | Heap: 4072
[23:32:18] zbSelfOta | Heap: 4072
[00:32:18] zbSelfOta | Heap: 4072
[01:32:18] zbSelfOta | Heap: 4072
[02:32:18] zbSelfOta | Heap: 4072
[03:32:18] zbSelfOta | Heap: 4072
[04:32:19] zbSelfOta | Heap: 4072
[05:32:19] zbSelfOta | Heap: 4072
[06:32:19] zbSelfOta | Heap: 4072
[07:32:19] zbSelfOta | Heap: 4072
[08:32:19] zbSelfOta | Heap: 4072
[09:32:19] zbSelfOta | Heap: 4072
[10:32:19] zbSelfOta | Heap: 4072
[11:32:19] zbSelfOta | Heap: 4072
[11:32:19] zbSelfOta | Checking ZB firmware on server
[11:32:19] zbSelfOta | zbChipNum: 0
[11:32:19] zbSelfOta | Connected to OTA server
[11:32:19] ConfigHelper | write config
[11:32:19] ConfigHelper | config saved
[11:32:20] zbSelfOta | Parsing firmware info...
[11:32:20] zbSelfOta | Firmware is up to date
[12:32:20] zbSelfOta | Heap: 2760
[13:32:20] zbSelfOta | Heap: 2760
[14:32:20] zbSelfOta | Heap: 2760
[15:32:20] zbSelfOta | Heap: 2760
[16:32:20] zbSelfOta | Heap: 2760
[17:27:00] EventSender | [_handleNewClient] new client: 192.168.0.2
[17:32:20] zbSelfOta | Heap: 2760

Koenkk, 2025-02-21:
Could you try with this fw: https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20221226 ?

Koenkk, 2025-02-23:
The buffer sizes changed, let me know if this fw is stable

Koenkk, 2025-03-01:
Let's continue in https://github.com/Koenkk/Z-Stack-firmware/issues/540, can you try with the fw from https://github.com/Koenkk/Z-Stack-firmware/issues/540#issuecomment-2692420392?

closed 2025-03-01 by Koenkk