Koenkk/zigbee2mqtt #26452: Missing Exposes for Backlight, Power-On Behavior, and Indicator on TS0003 Tuya 3-Gang Switch

arthurperezbessa, 2025-02-20:
### What happened?

 I have multiple TS0003 Tuya 3-gang switches integrated into my Zigbee2MQTT network. While switches 1, 2, and 3 are detected and exposed correctly, certain attributes like the backlight, power-on behavior, and indicator status are not showing up for this specific model.

Interestingly, I also have TS0003 devices with 2-gang configurations from the same brand and model, and these 2-gang switches are properly exposing all relevant attributes, including backlight, power-on behavior, and indicator.

Is there any way to fix this issue for the 3-gang version, so all relevant exposes (backlight, power-on behavior, indicator) are properly detected?

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

8.0.2 [GA]

### Adapter

Sonoff Dongle E

### Setup

Home Assistant OS on RPI4

roger1176, 2025-03-13:
I'm facing the same issue with a black Zemismart switch, model ZM-L03E-Z TS0003. In Zigbee2MQTT, two backlight settings appear: "Backlight mode" and "Indicator mode," but neither seems to work. If I turn it off and on, nothing changes on the switch.

Can anyone help me? I couldn't find anything in the forums.

DelanoLima, 2025-03-25:
I'm facing the same issue with a white Zemismart switch, model KES-606US-L3.

In the z2m, only the expose on/off appear for each gang. It is a 3 gang switch.

It is recognized in z2m like this:
**Manufaturer**: Tuya
**Zigbee Model**: TS0003
**Zigbee Manufatucturer**: _TZ3000_uilitwsy
**Description**: 3 gang switch
**Model**: [TS003](https://www.zigbee2mqtt.io/devices/TS0003.html)

[image]

devkaiwang, 2025-04-17:
I fixed the `None/Relay/Pos` indicator issue, a I wonder if it is related - https://github.com/Koenkk/zigbee-herdsman-converters/pull/9158

PhilipStubbs, 2025-07-10:
@devkaiwang are you willing to fix this for `_TZ3000_hf6cjpor`? 😁
I think its the same issue you fixed for the others.

PhilipStubbs, 2025-07-10:
Hey @devkaiwang 
I've updated `_TZ3000_hf6cjpor` to `_TZ3002_rbnycsav` in the db (`/homeassistant/zigbee2mqtt/database.db`). 

The extra settings aren't exposed though. 
Looking at the db it looks like the exposed settings are hardcoded in there, probably decided on interview?

[image]

[image]

PhilipStubbs, 2025-07-11:
I was able to manually change the `tuyaBacklightMode` though the Dev console.

Koenkk, 2025-07-12:
@PhilipStubbs could you provide the data/database.db entry of your device?

PhilipStubbs, 2025-07-12:
@Koenkk sure.

```
{
  "id": 18,
  "type": "Router",
  "ieeeAddr": "0xa4c1383b9ac16306",
  "nwkAddr": 26672,
  "manufId": 4417,
  "manufName": "_TZ3002_rbnycsav",
  "powerSource": "Mains (single phase)",
  "modelId": "TS0003",
  "epList": [
    1,
    2,
    3,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 256,
      "inClusterList": [
        3,
        4,
        5,
        6,
        57344,
        57345,
        0
      ],
      "outClusterList": [
        25,
        10
      ],
      "clusters": {
        "57344": {
          "attributes": {
            "53251": "AAAAAgAABAAA"
          }
        },
        "genBasic": {
          "attributes": {
            "65503": "��\u00020\u0012",
            "65506": 56,
            "65508": 0,
            "65534": 0,
            "stackVersion": 0,
            "dateCode": "",
            "manufacturerName": "_TZ3002_rbnycsav",
            "zclVersion": 3,
            "appVersion": 83,
            "modelId": "TS0003",
            "powerSource": 1,
            "hwVersion": 1
          }
        },
        "genOnOff": {
          "attributes": {
            "onOff": 1,
            "onTime": 0,
            "offWaitTime": 0,
            "tuyaBacklightMode": 3,
            "moesStartUpOnOff": 2,
            "tuyaBacklightSwitch": 1
          }
        },
        "manuSpecificTuya_3": {
          "attributes": {
            "powerOnBehavior": 2,
            "switchType": 1
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x90395efffe3fac93",
          "endpointID": 1
        },
        {
          "cluster": 57345,
          "type": "endpoint",
          "deviceIeeeAddress": "0x90395efffe3fac93",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 6,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 65000,
          "repChange": 1
        },
        {
          "cluster": 6,
          "attrId": 32769,
          "minRepIntval": 0,
          "maxRepIntval": 3600,
          "repChange": 0
        }
      ],
      "meta": {}
    },
    "2": {
      "profId": 260,
      "epId": 2,
      "devId": 256,
      "inClusterList": [
        4,
        5,
        6,
        57345
      ],
      "outClusterList": [],
      "clusters": {
        "genOnOff": {
          "attributes": {
            "onOff": 1,
            "onTime": 0,
            "offWaitTime": 0
          }
        },
        "manuSpecificTuya_3": {
          "attributes": {
            "powerOnBehavior": 2
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x90395efffe3fac93",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 6,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 65000,
          "repChange": 1
        }
      ],
      "meta": {}
    },
    "3": {
      "profId": 260,
      "epId": 3,
      "devId": 256,
      "inClusterList": [
        4,
        5,
        6,
        57345
      ],
      "outClusterList": [],
      "clusters": {
        "genOnOff": {
          "attributes": {
            "onOff": 1,
            "onTime": 0,
            "offWaitTime": 0
          }
        },
        "manuSpecificTuya_3": {
          "attributes": {
            "powerOnBehavior": 2
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x90395efffe3fac93",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 6,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 65000,
          "repChange": 1
        }
      ],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 83,
  "stackVersion": 0,
  "hwVersion": 1,
  "dateCode": "",
  "zclVersion": 3,
  "interviewCompleted": true,
  "interviewState": "SUCCESSFUL",
  "meta": {
    "configured": 332242049
  },
  "lastSeen": 1752320107634
}
```

PhilipStubbs, 2025-07-12:
maybe its the modelId? I'll try change that quick