Koenkk/zigbee2mqtt #26458: Container start stuck in loop

potatostuff13, 2025-02-21:
### What happened?

Upgraded to 2.1.1 and now docker container is stuck in a loop on start and never fully starts. Container does not work.

### What did you expect to happen?

I expected that the container would complete startup.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

zstack: Sonoff Zigbee Dongle P

### Setup

Docker container on Unraid

### Debug log

`
Starting Zigbee2MQTT without watchdog.
[2025-02-21 08:34:53] info:     z2m: Logging to console, file (filename: log.log)
[2025-02-21 08:34:53] info:     z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-21 08:34:53] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-21 08:34:53] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
Starting Zigbee2MQTT without watchdog.
[2025-02-21 08:34:54] info:     z2m: Logging to console, file (filename: log.log)
[2025-02-21 08:34:54] info:     z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-21 08:34:54] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-21 08:34:54] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
Starting Zigbee2MQTT without watchdog.
[2025-02-21 08:34:56] info:     z2m: Logging to console, file (filename: log.log)
[2025-02-21 08:34:56] info:     z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-21 08:34:56] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-21 08:34:56] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
`

Koenkk, 2025-02-22:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

potatostuff13, 2025-02-22:
`
Starting Zigbee2MQTT without watchdog.
[2025-02-22 09:31:13] debug:    z2m: Removing old log directory '/config/log/2025-02-22.09-30-56'
[2025-02-22 09:31:13] info:     z2m: Logging to console, file (filename: log.log)
[2025-02-22 09:31:13] debug:    z2m: Loaded state from file /config/state.json
[2025-02-22 09:31:13] info:     z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-02-22 09:31:13] debug:    z2m: sd-notify loaded
[2025-02-22 09:31:13] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-22 09:31:13] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/database.db\",\"databaseBackupPath\":\"/config/database.db.backup\",\"backupPath\":\"/config/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-02-22 09:31:13] debug:    zh:adapter:discovery: Error while trying to match USB adapter (spawn udevadm ENOENT).
[2025-02-22 09:31:13] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/database.db","databaseBackupPath":"/config/database.db.backup","backupPath":"/config/coordinator_backup.json"}'
[2025-02-22 09:31:13] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
`

Koenkk, 2025-02-22:
Do you see more in the STDOUT? (`docker logs -f CONTAINER_ID`). Also on what hardware are you running z2m? (I'm especially interested in the CPU architecture)

potatostuff13, 2025-02-22:
A few cycles of the STDOUT attached.

Hardware:
CPU: Intel Core i5-13500 (Raptor Lake)
Motherboard: Asrock Z690 Extreme

OS: Unraid

[STDOUT.txt](https://github.com/user-attachments/files/18925975/STDOUT.txt)

potatostuff13, 2025-02-22:
Not sure if this is helpful, but if I designate the port by-id vs ttyUSB0, I get a different set of errors:

[by_id_logs.txt](https://github.com/user-attachments/files/18926032/by_id_logs.txt)

Koenkk, 2025-02-26:
@potatostuff13 with what exit code foes the container stop? "docker ps -a" should show you

potatostuff13, 2025-02-26:
Exit Code = 1

Koenkk, 2025-03-01:
Could you try with z2m 2.1.2?

potatostuff13, 2025-03-02:
Unfortunately no change. Same error and exit code

Nerivec, 2025-03-02:
@Koenkk wasn't `spawn udevadm ENOENT` solved with the [udev changes in the Dockerfile](https://github.com/Koenkk/zigbee2mqtt/blob/ba337bd329aeb4ca17735c0cf09b31293c8cff06/docker/Dockerfile#L15) a while back? `STDOUT.txt` shows it.

Koenkk, 2025-03-03:
Could you try running the container with `--privileged` (https://docs.docker.com/engine/containers/run/#runtime-privilege-and-linux-capabilities) and provide the debug log again?

potatostuff13, 2025-03-03:
```
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2025-03-03 10:48:15] debug:    z2m: Removing old log directory '/app/data/log/2025-03-01.19-52-42'
[2025-03-03 10:48:15] info:     z2m: Logging to console, file (filename: log.txt)
[2025-03-03 10:48:15] debug:    z2m: Loaded state from file /app/data/state.json
[2025-03-03 10:48:15] info:     z2m: Starting Zigbee2MQTT version 2.1.3 (commit #ba337bd329aeb4ca17735c0cf09b31293c8cff06)
[2025-03-03 10:48:15] debug:    z2m: sd-notify loaded
[2025-03-03 10:48:15] info:     z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-03 10:48:15] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8cd948d2a9dbed119e69e92d62c613ac-if00-port0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-03-03 10:48:15] debug:    zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyUSB0"}]
[2025-03-03 10:48:15] debug:    zh:adapter:discovery: Unable to match USB adapter: zstack | /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8cd948d2a9dbed119e69e92d62c613ac-if00-port0
[2025-03-03 10:48:15] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8cd948d2a9dbed119e69e92d62c613ac-if00-port0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
[2025-03-03 10:48:15] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8cd948d2a9dbed119e69e92d62c613ac-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-03-03 10:48:15] error:    z2m: Error while starting zigbee-herdsman
[2025-03-03 10:48:15] error:    z2m: Failed to start zigbee-herdsman
[2025-03-03 10:48:15] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-03 10:48:15] error:    z2m: Exiting...
[2025-03-03 10:48:15] error:    z2m: Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8cd948d2a9dbed119e69e92d62c613ac-if00-port0
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2025-03-03 10:48:20] debug:    z2m: Removing old log directory '/app/data/log/2025-03-01.19-53-01'
[2025-03-03 10:48:20] info:     z2m: Logging to console, file (filename: log.txt)
[2025-03-03 10:48:20] debug:    z2m: Loaded state from file /app/data/state.json
[2025-03-03 10:48:20] info:     z2m: Starting Zigbee2MQTT version 2.1.3 (commit #ba337bd329aeb4ca17735c0cf09b31293c8cff06)
[2025-03-03 10:48:20] debug:    z2m: sd-notify loaded
[2025-03-03 10:48:20] info:     z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-03 10:48:20] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8cd948d2a9dbed119e69e92d62c613ac-if00-port0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-03-03 10:48:20] debug:    zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyUSB0"}]
[2025-03-03 10:48:20] debug:    zh:adapter:discovery: Unable to match USB adapter: zstack | /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8cd948d2a9dbed119e69e92d62c613ac-if00-port0
[2025-03-03 10:48:20] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8cd948d2a9dbed119e69e92d62c613ac-if00-port0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
[2025-03-03 10:48:20] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8cd948d2a9dbed119e69e92d62c613ac-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-03-03 10:48:20] error:    z2m: Error while starting zigbee-herdsman
[2025-03-03 10:48:20] error:    z2m: Failed to start zigbee-herdsman
[2025-03-03 10:48:20] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-03 10:48:20] error:    z2m: Exiting...
[2025-03-03 10:48:20] error:    z2m: Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8cd948d2a9dbed119e69e92d62c613ac-if00-port0
```

Nerivec, 2025-03-03:
You have to use whatever serial port path you set in the container.
```
Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyUSB0"}]
```
Should be `/dev/ttyUSB0`

potatostuff13, 2025-03-05:
Yes. I have tried both /de/ttyUSB0 and by-id (as the error above). Neither are working.

Nerivec, 2025-03-05:
We need the logs from the failure with the correct port, the error from the [incorrect one](https://github.com/Koenkk/zigbee2mqtt/issues/26458#issuecomment-2695268215) doesn't help since it's expected to not work when the port is wrong.
Can you follow the instructions from Koenkk but set the port to `/dev/ttyUSB0` this time?

potatostuff13, 2025-03-06:
```
Starting Zigbee2MQTT without watchdog.
[2025-03-05 16:00:07] debug:    z2m: Removing old log directory '/config/log/2025-03-05.15-59-50'
[2025-03-05 16:00:07] info:     z2m: Logging to console, file (filename: log.txt)
[2025-03-05 16:00:07] debug:    z2m: Loaded state from file /config/state.json
[2025-03-05 16:00:07] info:     z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-03-05 16:00:07] debug:    z2m: sd-notify loaded
[2025-03-05 16:00:07] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-03-05 16:00:07] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/database.db\",\"databaseBackupPath\":\"/config/database.db.backup\",\"backupPath\":\"/config/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-03-05 16:00:07] debug:    zh:adapter:discovery: Error while trying to match USB adapter (spawn udevadm ENOENT).
[2025-03-05 16:00:07] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/database.db","databaseBackupPath":"/config/database.db.backup","backupPath":"/config/coordinator_backup.json"}'
[2025-03-05 16:00:07] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
Starting Zigbee2MQTT without watchdog.
[2025-03-05 16:00:09] debug:    z2m: Removing old log directory '/config/log/2025-03-05.15-59-52'
[2025-03-05 16:00:09] info:     z2m: Logging to console, file (filename: log.txt)
[2025-03-05 16:00:09] debug:    z2m: Loaded state from file /config/state.json
[2025-03-05 16:00:09] info:     z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-03-05 16:00:09] debug:    z2m: sd-notify loaded
[2025-03-05 16:00:09] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-03-05 16:00:09] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/database.db\",\"databaseBackupPath\":\"/config/database.db.backup\",\"backupPath\":\"/config/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2025-03-05 16:00:09] debug:    zh:adapter:discovery: Error while trying to match USB adapter (spawn udevadm ENOENT).
[2025-03-05 16:00:09] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/database.db","databaseBackupPath":"/config/database.db.backup","backupPath":"/config/coordinator_backup.json"}'
[2025-03-05 16:00:09] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
```

potatostuff13, 2025-03-06:
Sorry. Updated logs:

```
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2025-03-05 16:20:00] debug:    z2m: Removing old log directory '/app/data/log/2025-03-05.16-12-48'
[2025-03-05 16:20:00] info:     z2m: Logging to console, file (filename: log.txt)
[2025-03-05 16:20:00] debug:    z2m: Loaded state from file /app/data/state.json
[2025-03-05 16:20:00] info:     z2m: Starting Zigbee2MQTT version 2.1.3 (commit #ba337bd329aeb4ca17735c0cf09b31293c8cff06)
[2025-03-05 16:20:00] debug:    z2m: sd-notify loaded
[2025-03-05 16:20:00] info:     z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-05 16:20:00] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\"},\"adapter\":{\"disableLED\":false}}"'
[2025-03-05 16:20:00] debug:    zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyUSB0"}]
[2025-03-05 16:20:00] error:    z2m: Error while starting zigbee-herdsman
[2025-03-05 16:20:00] error:    z2m: Failed to start zigbee-herdsman
[2025-03-05 16:20:00] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-05 16:20:00] error:    z2m: Exiting...
[2025-03-05 16:20:00] error:    z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:523:15)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:33)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[2025-03-05 16:20:00] error:    z2m: If this happens after updating to Zigbee2MQTT 2.0.0, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2025-03-05 16:20:10] debug:    z2m: Removing old log directory '/app/data/log/2025-03-05.16-12-50'
[2025-03-05 16:20:10] info:     z2m: Logging to console, file (filename: log.txt)
[2025-03-05 16:20:10] debug:    z2m: Loaded state from file /app/data/state.json
[2025-03-05 16:20:10] info:     z2m: Starting Zigbee2MQTT version 2.1.3 (commit #ba337bd329aeb4ca17735c0cf09b31293c8cff06)
[2025-03-05 16:20:10] debug:    z2m: sd-notify loaded
[2025-03-05 16:20:10] info:     z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-05 16:20:10] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\"},\"adapter\":{\"disableLED\":false}}"'
[2025-03-05 16:20:10] debug:    zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"},{"path":"/dev/ttyUSB0"}]
[2025-03-05 16:20:10] error:    z2m: Error while starting zigbee-herdsman
[2025-03-05 16:20:10] error:    z2m: Failed to start zigbee-herdsman
[2025-03-05 16:20:10] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-05 16:20:10] error:    z2m: Exiting...
[2025-03-05 16:20:10] error:    z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:523:15)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:33)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[2025-03-05 16:20:10] error:    z2m: If this happens after updating to Zigbee2MQTT 2.0.0, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364
```

potatostuff13, 2025-03-06:
I'm going to be closing this issue. I introduced a problem to my compose after rebuilding with the most recent update. Just resolved that and everything appears to be functioning again. Appreciate the help from both of you. Apologies that I didn't catch it over the weekend

closed 2025-03-06 by potatostuff13