Koenkk/zigbee2mqtt #26459: E1743 infinite update loop

CyberSparkNL, 2025-02-21:
### What happened?

I'm an issue wit the [E1743](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743) switches. They are on version 20210722 / [2.3.079](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) and an OTA update is available for them. They seem to update fine, but when the update is done the version number does not update, and when you check for new OTA updates it finds the update again.

### What did you expect to happen?

I have one [E1743](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743) switch which I already updated somewhere last year, which is on version 20230308 / [24.4.6](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html). But that doesn't seem to happen to the other ones at this moment.

### How to reproduce it (minimal and precise)

Go to the OTA tab and click Update Device Firmware for the switch.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

Slaesh

### Setup

Home Asisstant Add-on on x86-64 in Proxmox

### Debug log

Log of the update with info logging, I just switched to debug logging but it will still take a few hours for another update to fail.

```
[2025-02-21 17:10:42] info: 	z2m: Updating 'RC wk' to latest firmware
[2025-02-21 17:10:56] info: 	z2m: Update of 'RC wk' at 0.00%
[2025-02-21 17:10:56] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"battery":100,"identify":null,"linkquality":105,"update":{"installed_version":587699761,"latest_version":604241926,"progress":0,"state":"updating"},"update_available":false}'
[2025-02-21 17:11:26] info: 	z2m: Update of 'RC wk' at 0.46%, ≈ 140 minutes remaining
[ removed all the update status updates ]
[2025-02-21 18:51:26] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"battery":100,"identify":null,"linkquality":127,"update":{"installed_version":587699761,"latest_version":604241926,"progress":98.21,"remaining":110,"state":"updating"},"update_available":false}'
[2025-02-21 18:51:57] info: 	z2m: Update of 'RC wk' at 98.74%, ≈ 1 minutes remaining
[2025-02-21 18:51:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"battery":100,"identify":null,"linkquality":127,"update":{"installed_version":587699761,"latest_version":604241926,"progress":98.74,"remaining":77,"state":"updating"},"update_available":false}'
[2025-02-21 18:52:28] info: 	z2m: Update of 'RC wk' at 99.25%, ≈ 1 minutes remaining
[2025-02-21 18:52:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"battery":100,"identify":null,"linkquality":127,"update":{"installed_version":587699761,"latest_version":604241926,"progress":99.25,"remaining":46,"state":"updating"},"update_available":false}'
[2025-02-21 18:52:58] info: 	z2m: Update of 'RC wk' at 99.69%, ≈ 0 minutes remaining
[2025-02-21 18:52:58] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"battery":100,"identify":null,"linkquality":127,"update":{"installed_version":587699761,"latest_version":604241926,"progress":99.69,"remaining":19,"state":"updating"},"update_available":false}'
[2025-02-21 18:53:21] info: 	z2m: Update of 'RC wk' at 100.00%
[2025-02-21 18:53:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"battery":100,"identify":null,"linkquality":127,"update":{"installed_version":587699761,"latest_version":604241926,"progress":100,"remaining":19,"state":"updating"},"update_available":false}'
[2025-02-21 18:53:30] info: 	z2m: Finished update of 'RC wk'
[2025-02-21 18:53:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"battery":100,"identify":null,"linkquality":127,"update":{"installed_version":604241926,"latest_version":604241926,"state":"idle"},"update_available":false}'
[2025-02-21 18:53:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"RC wk","ieee_address":"0xf4b3b1fffeaa1121"},"type":"device_announce"}'
[2025-02-21 18:53:31] info: 	z2m: Device 'RC wk' was updated from '{"dateCode":"20210722","softwareBuildID":"2.3.079"}' to '{"dateCode":"20210722","softwareBuildID":"2.3.079"}'
[2025-02-21 18:53:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20210722","software_build_id":"2.3.079"},"id":"RC wk","to":{"date_code":"20210722","software_build_id":"2.3.079"}},"status":"ok","transaction":"j7wnn-1"}'
[2025-02-21 18:57:49] info: 	z2m: Configuring 'RC wk'
[2025-02-21 18:58:00] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"battery":100,"identify":null,"linkquality":127,"update":{"installed_version":604241926,"latest_version":604241926,"state":"idle"},"update_available":false}'
[2025-02-21 18:58:00] info: 	z2m: Successfully configured 'RC wk'
[2025-02-21 20:25:50] info: 	z2m: Checking if update available for 'RC wk'
[2025-02-21 20:25:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"action":"off","battery":100,"identify":null,"linkquality":112,"update":{"installed_version":604241926,"latest_version":604241926,"state":"idle"},"update_available":false}'
[2025-02-21 20:25:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk/action', payload 'off'
[2025-02-21 20:25:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"action":"off","battery":100,"identify":null,"linkquality":116,"update":{"installed_version":604241926,"latest_version":604241926,"state":"idle"},"update_available":false}'
[2025-02-21 20:25:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk/action', payload 'off'
[2025-02-21 20:25:54] info: 	zhc:ota: [0xf4b3b1fffeaa1121 | TRADFRI on/off switch] Firmware release notes: https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html
[2025-02-21 20:25:54] info: 	z2m: Update available for 'RC wk'
[2025-02-21 20:25:54] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC wk', payload '{"battery":100,"identify":null,"linkquality":109,"update":{"installed_version":587699761,"latest_version":604241926,"state":"available"},"update_available":false}'
```

Osjuh, 2025-03-30:
I have the same issue with a E1743 on version 2.3.079 as mentioned above, however I experience the same with the E1812 shortcut button on version 2.3.080 and the E1766 Open/Close buttons on version 2.3.079. All buttons are 3 years old.

The E1743 I got with the latest drivers 6 months ago as well the E1766 I got 4 weeks ago with my new blinds where updating without issues and those where on 2.3.079 as well.

LagaV, 2025-06-02:
Same problem here. One of my E1743 works through the update, somehow doesn't finish the update and shows up again for the same update to be supplied.

derzwen, 2025-06-19:
Same problem here with one of my E1743…

whisdol, 2025-09-07:
I have the same issue two E1743. Starting version is 2.3.079. Update appears to go to 100%, but then says it updated to the version that was already installed.
I'm using a Sonoff Zigbee 3.0 USB Dongle Plus V2, but have also tried re-pairing the Button via an IKEA router (Tradfri Dimmer 30W) instead of the coordinator with the same result.

```
[2025-09-05 10:04:10] [32minfo[39m: 	z2m: Updating 'Bettbeleuchtung Button 1' to latest firmware
[2025-09-05 10:04:12] [32minfo[39m: 	zhc:ota: [0xccccccfffe56cfdf | TRADFRI on/off switch] Getting latest image, meta: {"fileName":"tradfri-onoff-controller_release_prod_v604241926_3c2e5569-667c-49ed-a286-78a0e031935d.ota","fileVersion":604241926,"fileSize":205488,"url":"https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/IKEA/tradfri-onoff-controller_release_prod_v604241926_3c2e5569-667c-49ed-a286-78a0e031935d.ota","imageType":4549,"manufacturerCode":4476,"sha512":"41adf236facf5176dd7cd7aad4e8fcdfc9294e07ac5db0b0aad9f79d914817a04fb5c647d08d61188c10e52819ffc6058cfe93611d2ccf35234b9167a96e839a","otaHeaderString":"GBL GBL_tradfri_onoff_controller","originalUrl":"https://fw.ota.homesmart.ikea.com/files/tradfri-onoff-controller_release_prod_v604241926_3c2e5569-667c-49ed-a286-78a0e031935d.ota","releaseNotes":"https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html"}
[2025-09-05 10:04:23] [32minfo[39m: 	z2m: Update of 'Bettbeleuchtung Button 1' at 0.00%
...
[2025-09-05 11:46:39] [32minfo[39m: 	z2m: Update of 'Bettbeleuchtung Button 1' at 100.00%
[2025-09-05 11:46:48] [32minfo[39m: 	z2m: Finished update of 'Bettbeleuchtung Button 1'
[2025-09-05 11:46:48] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bettbeleuchtung Button 1', payload '{"battery":60,"identify":null,"linkquality":180,"update":{"installed_version":604241926,"latest_version":604241926,"state":"idle"}}'
[2025-09-05 11:46:48] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Bettbeleuchtung Button 1","ieee_address":"0xccccccfffe56cfdf"},"type":"device_announce"}'
[2025-09-05 11:46:49] [32minfo[39m: 	z2m: Device 'Bettbeleuchtung Button 1' was updated from '{"dateCode":"20210722","softwareBuildID":"2.3.079"}' to '{"dateCode":"20210722","softwareBuildID":"2.3.079"}'
[2025-09-05 11:46:49] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20210722","software_build_id":"2.3.079"},"id":"0xccccccfffe56cfdf","to":{"date_code":"20210722","software_build_id":"2.3.079"}},"status":"ok","transaction":"z8uv9-7"}'
```

whisdol, 2025-11-07:
This issue still exists with Zigbee2MQTT version
[2.6.3](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.6.3).

andrei-lazarov, 2026-01-07:
If the device still shows the old version, even after successful interview, it means it didn't update successfully.
Did you notice a change in behavior? It should be like this:
- Old firmware: bind to individual devices only
- New firmware: bind to groups only

New batteries, better signal and try again?