Koenkk/zigbee2mqtt #26460: Sonoff TRVZB with zigbeeMQTT traces error in automation- seems mix up calibration temperature with set temperature.

PavelTFX, 2025-02-21:
### What happened?

Hello, I tried automation on Sonoff TRVZB to set temperature based on time schedule. I got erorr and traces shows Error: Value 23.0 for number.0x08ddebfffef6329e_local_temperature_calibration is outside valid range -12.8 - 12.7  but in config I have:
choose:
  - conditions:
      - condition: state
        entity_id: schedule.teplota_mistnost
        state: 'on'
    sequence:
      - device_id: 4d761e27f3f224ef17f80dea9772578f
        domain: number
        entity_id: 490b7db0c3a1efa6dfc57f3603ebd7c8
        type: set_value
        value: 23

it seems, it somehow mixed entity_id for calibration temperature with set temperature.
happened to enyone? any solution? 

### What did you expect to happen?

it supposed to set temperature to 23°C

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

7.4.4

### Adapter

EmberZNet

### Setup

Home Assistant Add-on: Zigbee2MQTT  on HA green

closed 2025-04-30 by github-actions[bot]

Ziron, 2025-05-15:
I believe this is caused by these lines in the code:
https://github.com/Koenkk/zigbee2mqtt/blob/9d1427c1041c74a412d48a3f3be23b4c71397ec3/lib/extension/homeassistant.ts#L1210-L1213

This forces Home Assistant to assign default names to the entities. When a device such as the Sonoff TRVZB have multiple entities of the same type/device_class then they all get the same default name.... In this case both the setpoint and calibration offset get named just "Temperature" (but at least translated to the local language set in Home Assistant!)

I would like to retain the ability to override any name from the Z2M-config without having it discarded like this... But at the very least it shouldn't mash up multiple entities into the same default name when they have the same device_class!

I think this issue should be reopened and fixed.

PavelTFX, 2025-05-16:
Hello, Thanks for comment. This looks like possible cause. any idea, what we could do about it? I am not experienced IT/geek/Home assitant user. Could i change something in my system maybe?
thanks again,
Pavel