Koenkk/zigbee2mqtt #26462: Error: Failed to connect to the adapter - AGAIN

Westcott1, 2025-02-21:
### What happened?

Z2M fails to start with ` z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)`
It was working OK until I restarted it to use an external converter.
Now it fails to start.
The log shows that the stick was auto-discovered correctly.

### What did you expect to happen?

Z2M to start correctly

### How to reproduce it (minimal and precise)

npm start

### Zigbee2MQTT version

Latest

### Adapter firmware version

Latest

### Adapter

See log below

### Setup

Plain Raspberry Pi4b

### Debug log

Starting Zigbee2MQTT without watchdog.
[2025-02-21 22:06:34] debug:    z2m: Removing old log directory '/opt/zigbee2mqtt/data/log/2025-02-21.20-06-54'
[2025-02-21 22:06:34] info:     z2m: Logging to console, file (filename: log.log)
[2025-02-21 22:06:34] debug:    z2m: Loaded state from file /opt/zigbee2mqtt/data/state.json
[2025-02-21 22:06:34] info:     z2m: Starting Zigbee2MQTT version 2.1.1 (commit #1accb8b6)
[2025-02-21 22:06:34] debug:    z2m: sd-notify loaded
[2025-02-21 22:06:34] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-02-21 22:06:34] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[20],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/opt/zigbee2mqtt/data/database.db\",\"databaseBackupPath\":\"/opt/zigbee2mqtt/data/database.db.backup\",\"backupPath\":\"/opt/zigbee2mqtt/data/coordinator_backup.json\",\"serialPort\":{\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false,\"transmitPower\":5}}"'
[2025-02-21 22:06:34] debug:    zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"0658","pnpId":"usb-0658_0200-if00","vendorId":"0658","productId":"0200"},{"path":"/dev/ttyAMA0"},{"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"0001","pnpId":"usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0","vendorId":"10c4","productId":"ea60"},{"path":"/dev/ttyUSB1","manufacturer":"1a86","pnpId":"usb-1a86_USB2.0-Serial-if00-port0","vendorId":"1a86","productId":"7523"},{"path":"/dev/ttyUSB2","manufacturer":"1a86","pnpId":"usb-1a86_USB_Serial-if00-port0","vendorId":"1a86","productId":"7523"}]
[2025-02-21 22:06:34] info:     zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"0001","pnpId":"usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: path=/dev/ttyUSB0, score=2
[2025-02-21 22:06:34] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[20]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false,"transmitPower":5},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}'
[2025-02-21 22:06:34] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-02-21 22:06:34] info:     zh:zstack:znp: Serialport opened
[2025-02-21 22:06:34] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-02-21 22:06:34] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-02-21 22:06:35] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-02-21 22:06:35] debug:    zh:zstack:unpi:writer: --> buffer [239]
[2025-02-21 22:06:36] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-02-21 22:06:36] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-02-21 22:06:36] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-02-21 22:06:36] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-02-21 22:06:36] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-02-21 22:06:42] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-02-21 22:06:42] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-02-21 22:06:48] debug:    zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-02-21 22:06:48] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-02-21 22:06:54] error:    z2m: Error while starting zigbee-herdsman
[2025-02-21 22:06:54] error:    z2m: Failed to start zigbee-herdsman
[2025-02-21 22:06:54] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-21 22:06:54] error:    z2m: Exiting...
[2025-02-21 22:06:54] error:    z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:142:13)
    at start (/opt/zigbee2mqtt/index.js:161:5)

McGiverGim, 2025-02-21:
I think I have just the same problem. It started without touching anything that I remember. Some idea?

Westcott1, 2025-02-22:
--- SOLVED ---

The adapter auto-discovery was choosing the wrong port!
It had worked correctly ever since my update of Z2M to v2.
I had to add a port: to serial: in configuration.yaml

closed 2025-02-22 by Westcott1