Koenkk/zigbee2mqtt #26467: _TZ3000_0ghwhypc not reporting energy, power, current & voltage

K00Kai18, 2025-02-22:
### What happened?

Device paired, but i can only switch it on and off.
Device is not reporting power, tension and current while ZHA is.
Picture shows a 1 gang switch while the one I'm having is the following one:
https://a.aliexpress.com/_EQV2kxO

Device found as "TS0001" instead of "TS0001_power".

Zigbee manufacturer: _TZ3000_0ghwhypc

### What did you expect to happen?

Expecting power, current and tension to be reported

### How to reproduce it (minimal and precise)

Pairing

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

SMLIGHT SLZB-06

### Setup

Add-on on Home Assitant (Rasbery Pi)

Koenkk, 2025-02-22:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZ3000_0ghwhypc` to `_TZ3000_jsfzkftc`
- Start Z2M and see if the device now works

K00Kai18, 2025-02-22:
> Could you try to:
> 
> * Stop Z2M
> * Open `data/database.db` with a text editor and change all occurrences of `_TZ3000_0ghwhypc` to `_TZ3000_jsfzkftc`
> * Start Z2M and see if the device now works

Just tried, power appeared but shown as "Unknown"

K00Kai18, 2025-02-23:
Sorry... just checked again and now it works 😁.
Thank you

closed 2025-02-23 by K00Kai18