Koenkk/zigbee2mqtt #26470: Buttons 3 and 4 on OSRAM Smart+ switch do not fire any events

jabbot2, 2025-02-22:
### What happened?

Similar to this issue from a few years ago: https://github.com/Koenkk/zigbee2mqtt/issues/8335
I have several of these switches (OSRAM Smart+ switch model 4058075816459) and the left buttons register as "On" and "Off" but the right buttons do not send anything.  There is a comment in there that the US versions have the right buttons disabled, but this works in ZHA, but when I moved them to z2m I lost that functionality.

### What did you expect to happen?

I expected that I would have the same features as the device in ZHA.

### How to reproduce it (minimal and precise)

To reproduce this, I simply paired the switch with z2m.  I then observed the log for what events were firing and when.  The left buttons generate an on and off, the right buttons generate nothing

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

7.4.4 [GA]

### Adapter

ITEAD Sonoff Zigbee 3.0 Plus v2

### Setup

HAOS on Intel NUC

### Debug log

There is nothing in the logs as there is no message from the buttons 3 and 4.

This is the ZHA device:
`{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4364,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 0,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0020",
        "0x1000",
        "0xfd00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000",
        "0xfd00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000",
        "0xfd00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000",
        "0xfd00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "5": {
      "profile_id": "0x0104",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000",
        "0xfd00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    },
    "6": {
      "profile_id": "0x0104",
      "device_type": "0x0810",
      "input_clusters": [
        "0x0000",
        "0x1000",
        "0xfd00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000"
      ]
    }
  },
  "manufacturer": "OSRAM",
  "model": "Switch 4x-LIGHTIFY",
  "class": "zhaquirks.osram.lightifyx4.LightifyX4"
}`

tsaro, 2025-02-25:
Same issue here, all 4 buttons worked in zha but only getting press and hold/dim events from the left 2 buttons under z2m

klarf-pixel, 2025-03-09:
With this same switch+ I have similar problem :
One click work on the 4 buttons
Hold click work on the 2 left buttons
Hold click dont work on the right buttons, these 2 buttons are on ON at the same time.

With the older version of zigbee2mqtt the switch worked well 😕 with the 2.x version I have 19 devices discovered in domoticz. 
I use zigbee2mqtt with domoticz 2025.1 bêta on a rpi3b+ and a sonoff usb Key.

klarf-pixel, 2025-03-10:
To compare the information I have under domoticz (mqtt):

[image]

What is expected to be discovered under zigbee2mqtt for comparison:

[image]

Thank you for your help

klarf-pixel, 2025-05-10:
Hi,
Are you some news about this problem ?
Thank

jabbot2, 2025-06-16:
When I do debug log, I get this:
`info 2025-06-16 14:37:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"SW-1","ieee_address":"0x000d6f000ec8ff44"},"type":"device_announce"}'
info 2025-06-16 14:37:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":104,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-16 14:37:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":104,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-16 14:37:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-16 14:37:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"off","battery":80,"linkquality":112,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-16 14:37:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":112,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-16 14:37:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'off'`

Databases.db:

[image]

I have 2 of these switches, here are both entries.  They exhibit the same symptoms.

Koenkk, 2025-06-17:
This log only shows the info level, after debug logging is activated you can grab the log from the `data/log` folder.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

jabbot2, 2025-06-17:
[45df7312_zigbee2mqtt_2025-06-17T22-26-43.564Z.log](https://github.com/user-attachments/files/20785044/45df7312_zigbee2mqtt_2025-06-17T22-26-43.564Z.log)

Here's the debug log, I believe!  Sorry for missing the debug.

jabbot2, 2025-06-20:
Yes, some of those bindings are missing, however when I try to add them, I get the following:

info 2025-06-20 06:11:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-20 06:11:49z2m: Successfully bound cluster 'genPollCtrl' from 'SW-1' to 'Coordinator'
info 2025-06-20 06:11:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"SW-1","ieee_address":"0x000d6f000ec8ff44"},"type":"device_announce"}'
info 2025-06-20 06:11:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":84,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-20 06:11:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":84,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-20 06:11:50z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-20 06:11:51zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "43925".
info 2025-06-20 06:11:51z2m: Successfully bound cluster 'genOnOff' from 'SW-1' to 'Coordinator'
info 2025-06-20 06:11:52z2m: Successfully bound cluster 'genLevelCtrl' from 'SW-1' to 'Coordinator'
info 2025-06-20 06:11:53z2m: Successfully bound cluster 'genPowerCfg' from 'SW-1' to 'Coordinator'
info 2025-06-20 06:11:54z2m: Successfully bound cluster 'lightingColorCtrl' from 'SW-1' to 'Coordinator'
info 2025-06-20 06:12:00zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "43925".
warning 2025-06-20 06:12:04z2m: Failed to setup reporting for 'Coordinator/1' cluster 'genOnOff' (ZCL command 0xe406bffffe40c095/1 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":0,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":189},"zclSequence":240,"commandIdentifier":7} timed out after 10000ms))
warning 2025-06-20 06:12:14z2m: Failed to setup reporting for 'Coordinator/1' cluster 'genLevelCtrl' (ZCL command 0xe406bffffe40c095/1 genLevelCtrl.configReport([{"attribute":"currentLevel","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":0,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":192},"zclSequence":241,"commandIdentifier":7} timed out after 10000ms))
warning 2025-06-20 06:12:24z2m: Failed to setup reporting for 'Coordinator/1' cluster 'lightingColorCtrl' (ZCL command 0xe406bffffe40c095/1 lightingColorCtrl.read(["colorCapabilities"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":0,"apsFrame":{"profileId":260,"clusterId":768,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":195},"zclSequence":242,"commandIdentifier":1} timed out after 10000ms))
info 2025-06-20 06:12:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genPollCtrl","genOnOff","genLevelCtrl","genPowerCfg","lightingColorCtrl"],"failed":[],"from":"SW-1","from_endpoint":"1","to":"Coordinator","to_endpoint":null},"status":"ok","transaction":"3qg5v-1"}'

Koenkk, 2025-06-21:
@jabbot2 try repeatedly pressing buttons on the switch while binding, likely the device falls asleep.

jabbot2, 2025-06-22:
I continuously used button 1 while binding and these are the logs:
`info 2025-06-22 09:09:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":112,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":112,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-22 09:09:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":120,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":120,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-22 09:09:09z2m: Successfully bound cluster 'genPollCtrl' from 'SW-1' to 'Coordinator'
info 2025-06-22 09:09:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-22 09:09:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":128,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":128,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-22 09:09:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":128,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":128,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-22 09:09:10z2m: Successfully bound cluster 'genOnOff' from 'SW-1' to 'Coordinator'
info 2025-06-22 09:09:10z2m: Successfully bound cluster 'genLevelCtrl' from 'SW-1' to 'Coordinator'
info 2025-06-22 09:09:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":136,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":136,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-22 09:09:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-22 09:09:11z2m: Successfully bound cluster 'genPowerCfg' from 'SW-1' to 'Coordinator'
info 2025-06-22 09:09:11z2m: Successfully bound cluster 'lightingColorCtrl' from 'SW-1' to 'Coordinator'
info 2025-06-22 09:09:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":132,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":132,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-22 09:09:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":128,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":128,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-22 09:09:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-22 09:09:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":128,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
[... cut ...] 260,"clusterId":768,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":203},"zclSequence":239,"commandIdentifier":1} timed out after 10000ms))
info 2025-06-22 09:09:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genPollCtrl","genOnOff","genLevelCtrl","genPowerCfg","lightingColorCtrl"],"failed":[],"from":"SW-1","from_endpoint":"1","to":"Coordinator","to_endpoint":null},"status":"ok","transaction":"ucdc7-1"}'`

Koenkk, 2025-06-22:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/08281b02b0d91d6fce74a43f249ca5a5): 

- save this as file next to `configuration.yaml` in the `external_converters` directory (create if it doesn't exist yet) as `ext_converter.js` (so `external_converters/ext_converter.js`)
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- re-configure the device by pressing the yellow refresh button on the device page (make sure to wake it up again)
- see if the issue is fixed now.

jabbot2, 2025-06-28:
Hello!  Ok, so I added the ext_converter and I can see the device ending with CUSTOM:

[image]

I then reconfigured the device and see that it reconfigured successfully, however buttons 2 and 4 still don't work.
Button 1 - Top left
Button 2 - Top right
Button 3 - Bottom left
Button 4 - Bottom right

I was able to bind the commands as suggested above with the custom extension, but it still doesn't log any button presses:

[image]

Koenkk, 2025-06-29:
Are you sure you woke up the device by pressing button on it while binding?

jabbot2, 2025-06-29:
Yes, I was constantly pressing the button 1.  I will do it again and log that data and return it here:
`info 2025-06-29 15:43:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":92,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-29 15:43:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":100,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":100,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:22z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-29 15:43:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":88,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":88,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-29 15:43:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":120,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":120,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
warning 2025-06-29 15:43:25z2m: Failed to setup reporting for 'Coordinator/1' cluster 'genLevelCtrl' (ZCL command 0xe406bffffe40c095/1 genLevelCtrl.configReport([{"attribute":"currentLevel","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":0,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":171},"zclSequence":123,"commandIdentifier":7} timed out after 10000ms))
info 2025-06-29 15:43:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":92,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":92,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-29 15:43:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":100,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":100,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-29 15:43:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":96,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":96,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:28z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-29 15:43:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":120,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":120,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
info 2025-06-29 15:43:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"on","battery":80,"linkquality":96,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1', payload '{"action":"","battery":80,"linkquality":96,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2900}'
info 2025-06-29 15:43:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SW-1/action', payload 'on'
warning 2025-06-29 15:43:35z2m: Failed to setup reporting for 'Coordinator/1' cluster 'lightingColorCtrl' (ZCL command 0xe406bffffe40c095/1 lightingColorCtrl.read(["colorCapabilities"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv": [... cut ...] et":0,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":178},"zclSequence":125,"commandIdentifier":7} timed out after 10000ms))
info 2025-06-29 15:43:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":["genLevelCtrl","lightingColorCtrl","genOnOff"],"failed":[],"from":"SW-1","from_endpoint":"4","to":"Coordinator","to_endpoint":null},"status":"ok","transaction":"4hlm1-1"}'`

klarf-pixel, 2025-11-05:
Hi,

I try to create a log file with my switch.

The configuration of the switch with the external converter :

[image]

The log file with annotation :

`#top left

[05/11/2025 08:05:49] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"brightness_move_up","action_rate":38,"battery":60,"brightness":215,"linkquality":51,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:49] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":215,"linkquality":51,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:49] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'brightness_move_up'
[05/11/2025 08:05:49] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":null,"action_brightness_delta":20,"battery":60,"brightness":255,"linkquality":51,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:49] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"brightness_stop","battery":60,"brightness":255,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:49] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":255,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:49] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'brightness_stop'

#top right

[05/11/2025 08:05:53] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"move_to_saturation","action_saturation":254,"action_transition_time":2,"battery":60,"brightness":255,"linkquality":58,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:53] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":255,"linkquality":58,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:54] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'move_to_saturation'
[05/11/2025 08:05:54] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"hue_stop","action_rate":0,"battery":60,"brightness":255,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:54] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":255,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:54] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'hue_stop'

#bottom left

[05/11/2025 08:05:59] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"brightness_move_down","action_rate":38,"battery":60,"brightness":255,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:59] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":255,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:59] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'brightness_move_down'
[05/11/2025 08:05:59] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":null,"action_brightness_delta":-20,"battery":60,"brightness":195,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:59] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"brightness_stop","battery":60,"brightness":195,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:59] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":195,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:05:59] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'brightness_stop'

#bottom right

[05/11/2025 08:06:04] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"move_to_saturation","action_saturation":254,"action_transition_time":2,"battery":60,"brightness":195,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:06:04] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":195,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:06:04] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'move_to_saturation'
[05/11/2025 08:06:05] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"hue_stop","action_rate":0,"battery":60,"brightness":195,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:06:05] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":195,"linkquality":47,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[05/11/2025 08:06:05] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'hue_stop'`

Thank you for your help.

Koenkk, 2025-11-05:
@klarf-pixel for all buttons in your log I see a published `action`, can you elaborate more on what the problem is?

klarf-pixel, 2025-11-06:
I forgot to mention, but previously the log showed the actions of long presses.
The two right buttons (up and down) are combined; the bottom right button doesn't produce a single action that can be used (within Domoticz).

For comparison, here is the log of short presses:

```
#top left
[06/11/2025 08:26:33] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"on","battery":60,"brightness":195,"linkquality":87,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[06/11/2025 08:26:33] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":195,"linkquality":87,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[06/11/2025 08:26:33] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'on'

#top right

[06/11/2025 08:26:36] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"color_temperature_step_up","action_step_size":25,"action_transition_time":0.1,"battery":60,"brightness":195,"linkquality":83,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[06/11/2025 08:26:36] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":195,"linkquality":83,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[06/11/2025 08:26:36] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'color_temperature_step_up'

#bottom left

[06/11/2025 08:26:39] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"off","battery":60,"brightness":195,"linkquality":83,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[06/11/2025 08:26:39] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":195,"linkquality":83,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[06/11/2025 08:26:39] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'off'

#bottom right

[06/11/2025 08:26:42] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"color_temperature_step_down","action_step_size":25,"action_transition_time":0.1,"battery":60,"brightness":195,"linkquality":83,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[06/11/2025 08:26:42] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau', payload '{"action":"","battery":60,"brightness":195,"linkquality":83,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2800}'
[06/11/2025 08:26:42] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Ledvance Switch bureau/action', payload 'color_temperature_step_down'
```

closed 2025-11-12 by Koenkk

klarf-pixel, 2025-11-13:
@Koenkk
One quick question: in the next update, will I be able to remove the external converter without changing the names of the entries in MQTT?

Koenkk, 2025-11-14:
yes!