Koenkk/zigbee2mqtt #26473: Failed to configure zbminir2: UNSUPPORTED_ATTRIBUTE

jreiffers, 2025-02-22:
### What happened?

Hey! Until recently, z2m was working absolutely perfectly. However, all of a sudden, I'm getting a lot of these errors for all my Sonoff zbminir2 devices:

```
Failed to configure 'XYZ', attempt 1 (Error: ZCL command 0xe406bffffe21893a/1 customClusterEwelink.read(["radioPower",1,20,21,22,23], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
 at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28)
 at Endpoint.zclCommand (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26)
 at Endpoint.read (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29)
 at configure (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/devices/sonoff.ts:1480:13)
 at Object.configure (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:21)
 at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:124:13)
 at Immediate.<anonymous> (/opt/zigbee2mqtt/lib/extension/configure.ts:69:17))
```

This seems to also somewhat affect the stability of the whole system. I tried enabling debug logging, but I didn't find any additional info there.

Thanks in advance for taking a look (and thanks for building this wonderful software)!

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

plain, x86-64, bare metal

jreiffers, 2025-02-22:
Stability isn't really the right word, what's actually happening is:

- Devices can still be controlled without any issues
- Devices still report being turned on via the switch
- Devices no longer report being turned off via the switch

TheNexter, 2025-02-22:
https://github.com/Koenkk/zigbee2mqtt/issues/25532#issuecomment-2580856937

Do you have update the firmware ? ;)

jreiffers, 2025-02-25:
Just as suddenly as it started it stopped again. Everything is working fine now, despite me not doing anything. /shrug

closed 2025-02-25 by jreiffers