Koenkk/zigbee2mqtt #26477: IR zigbee tuya blasters are exposing the wrong entity to home assistant: cant send repeated IR commands and cant send big IR commands with more than 255 characters

lattice0, 2025-02-22:
### What happened?

infrared zigbee tuya basters, in my case, these models:
 
`_TZ3290_ot6ewjvmejq5ekhl` by `TS1201`
`_TZ3290_u9xac5rv` by `TS1201`
`_TZ3290_7v1k4vufotpowp9z` by `TS1201`

which are these:

[image]

work by exposing 

`Ir code to send`

which is a text entity on the home assistant side. This has 2 problems:

- text entities on home assistant only get published as MQTT messages if they change. Therefore if I send the same command twice it won't send. So if, for example, you send volume increase twice, it only sends the first time. The workaround was to do an automation that clear the value once its set, which is not nice
- text entities on home assistant are at most 255 characters, but many IR codes can go over that. An example is my Samsung Air conditioner which has a turn off code with 350 characters

So we need to change those to not be a text entity but instead an action that sends arbitraty text. I dont know how to do that though.

Since MQTT can publish anything, I think this is a problem related to the protocol agreed between home assistant and zigbee2mqtt which I dont know how it works.

### Zigbee2MQTT version

2.1.1 commit: 1accb8b66b95b99658c50e4c0209edcb946a5fe2

### Adapter firmware version

20210708

### Adapter

sonoff P

### Setup

raspberry pi 3b+, docker container

lux73, 2025-02-23:
this has to be fixed (workaround) on your HA side...

OpenHAB with z2m and Tuya UFO-R11 are works as expected, even with the same repeated IR Commands

PLEASE don't break generally working Solutions 😉

lattice0, 2025-02-23:
@lux73 on the community it was of their opinion that it should be fixed on zigbee2mqtt side. 

My idea was not to break what is working but to add a second way of controlling these devices, one that is not a simple text input

closed 2025-05-02 by github-actions[bot]

Ruud-cb, 2026-01-12:
Any updates or workarounds on this?