Koenkk/zigbee2mqtt #26483: The lights will turn off immediately after they are turned on.

yuanxin32323, 2025-02-23:
### What happened?

I found that for the spotlights and the strip light driver of TS0502B, after a certain period of time, approximately a few hours, when the operation of turning on the lights is executed, the lights will turn off immediately after being turned on.

I discovered in the issues on the GitHub of zigbee2mqtt that several other people are also experiencing the same situation.

However, in the end, none of them were able to find a good solution.

This problem occurs quite frequently, yet it is very difficult to reproduce. We can only wait for a few hours and then turn on the lights again to have a chance of encountering this problem. This undoubtedly increases the difficulty of debugging.

I would like to share my observations. By using debugging to check the data packets sent by the zigbee gateway to the device, first, the zigbee gateway will send a command to turn on the lights. Then the device will turn on and send back a data packet indicating that the lights have been turned on. But within a few seconds, another data packet indicating that the lights have been turned off will be sent back, and at this moment, the lights will go out.  

**The following are the issues I have collected where people have encountered the same problem.** 
[https://github.com/Koenkk/zigbee2mqtt/issues/21625](url)
[https://github.com/Koenkk/zigbee2mqtt/issues/18902](url)
[https://github.com/Koenkk/zigbee2mqtt/issues/22197](url)

### What did you expect to happen?

I hope the lights can be turned on and off normally.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

x86

olili, 2025-02-26:
confrmed, i perceive the same issue and behaviour.

yuanxin32323, 2025-03-04:
@Koenkk

soorur, 2025-04-08:
Same problem.

neiromaster, 2025-04-25:
I have this problem too

soorur, 2025-05-28:
Has anyone had any success? I recently tried "timer" setup and it still turns off after turning on. 
@Koenkk

closed 2025-08-04 by github-actions[bot]

vomov, 2025-10-15:
I thought `lastseen` was passive, simply making a note of when a device was seen. But, if I understand correctly, this setting results in zigbee2mqtt sending _two_ commands?

Koenkk, 2025-10-16:
@vomov yes, when last seen is activated a state update will be send whenever the device checks-in with z2m.

olili, 2025-10-17:
I have "lastseen" disabled in my config and never the less I'm facing the issue in every switch-on attempt. 
Is there any thing else what I can do by configuration to make the device working?

Koenkk, 2025-10-17:
could you provide the debug log again?

vomov, 2025-10-18:
@Koenkk; I've changed the option `last_seen` to `disabled`, and noticed the field disappearing from the UI. However, the two devices that are misbehaving are still misbehaving.
The unexpected messages are gone, but one light still turns itself off just after I told it to go on, while another is still 'finnicky'. Here's another debug log:

```
1. [2025-10-18 21:07:43] info: 	z2m: Successfully changed options
2. [2025-10-18 21:07:43] info: 	z2m:mqtt: MQTT publish: topic 'house/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok"}'
3. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,160,46,1,61,170,156,254,28,68,129,0,0,5,4,160,46,1,1,0,87,0,55,219,230,0,0,8,8,110,10,0,0,33,236,19,61,170,28,58]
4. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,160,46,1,61,170,156,254,28,68,129,0,0,5,4,160,46,1,1,0,87,0,55,219,230,0,0,8,8,110,10,0,0,33,236,19,61,170,28,58]
5. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [160,46,1,61,170] - 156
6. [2025-10-18 21:07:44] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
7. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,5,4,160,46,1,1,0,87,0,55,219,230,0,0,8,8,110,10,0,0,33,236,19,61,170,28,58]
8. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,5,4,160,46,1,1,0,87,0,55,219,230,0,0,8,8,110,10,0,0,33,236,19,61,170,28] - 58
9. [2025-10-18 21:07:44] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":1029,"srcaddr":11936,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":87,"securityuse":0,"timestamp":15129399,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[8,110,10,0,0,33,236,19]}}
10. [2025-10-18 21:07:44] debug: 	zh:controller: Received payload: clusterID=1029, address=11936, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=87, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":110,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":33,"attrData":5100}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
11. [2025-10-18 21:07:44] debug: 	zh:controller:endpoint: ZCL command 0x287681fffeaafc73/1 msRelativeHumidity.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":110,"writeUndiv":false})
12. [2025-10-18 21:07:44] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x287681fffeaafc73:11936/1 (0,0,1)
13. [2025-10-18 21:07:44] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":11936,"destendpoint":1,"srcendpoint":1,"clusterid":1029,"transid":33,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,110,11,10,0]}}
14. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,160,46,1,1,5,4,33,0,30,5,24,110,11,10,0,232]
15. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --- parseNext []
16. [2025-10-18 21:07:44] debug: 	z2m: Received Zigbee message from 'study/vindstryka', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5100}' from endpoint 1 with groupID 0
17. [2025-10-18 21:07:44] info: 	z2m:mqtt: MQTT publish: topic 'house/study/vindstryka', payload '{"humidity":51,"last_seen":"2025-10-17T10:46:22+02:00","linkquality":87,"pm25":2,"temperature":22,"update":{"installed_version":16777233,"latest_version":16777233,"state":"idle"},"voc_index":473}'
18. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
19. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
20. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
21. [2025-10-18 21:07:44] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
22. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --- parseNext []
23. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,33,231]
24. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,33,231]
25. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,33] - 231
26. [2025-10-18 21:07:44] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":33}
27. [2025-10-18 21:07:44] debug: 	zh:zstack:unpi:parser: --- parseNext []
28. [2025-10-18 21:07:44] debug: 	z2m:mqtt: Received MQTT message on 'house/study/humidity' with data '51'
29. [2025-10-18 21:07:44] debug: 	z2m:mqtt: Received MQTT message on 'house/study/pm25' with data '2'
30. [2025-10-18 21:07:44] debug: 	z2m:mqtt: Received MQTT message on 'house/study/voc' with data '473'
31. [2025-10-18 21:07:44] debug: 	z2m:mqtt: Received MQTT message on 'house/study/temperature' with data '22'
32. [2025-10-18 21:07:52] debug: 	z2m:mqtt: Received MQTT message on 'house/bathroom/ledstrip/set' with data '{"state":"ON","transition":10,"brightness":254,"topic":" house/bathroom/ledstrip/set"}'
33. [2025-10-18 21:07:52] debug: 	z2m: Publishing 'set' 'brightness' to 'bathroom/ledstrip'
[... cut ...]
Which I would expect; no more duplicate messages. Incidentally, it seems to retain the `last_seen` datetime that was there when I disabled `last_seen`.

Here's a tiny tidbit that might be relevant: the ledstrip has a button, which allows me to turn it toggle and dim it. If I press this button, it turns on as normal, and cheerfully reports its state correctly. Then, for a short while, the behaviour is as expected: an `ON` turns it ON, `OFF` turns it off. After a while, it starts flashing on and then turning off again.

Any ideas?

Koenkk, 2025-10-21:
So it seems the device does not correctly implement the Zigbee spec which causes this issue when using a `transition`. Nice that it is solved.

Larsvb0, 2026-02-03:
Did you try the workaround I provided? Second one is rock solid for me