Koenkk/zigbee2mqtt #26485: A lot of Failed to read state and publish state

JC38, 2025-02-23:
### What happened?

My network is made with a Smlight SLZB-06M controller and 77  Nous - A1Z sockets.
All A1Z are connected with **availability ok** and all last seens below 3 minutes. But some are not sending their state (Failed to read state) or are raising a Publish state error.
I have tested all the 2024mmdd Zigbee available firmwares from Smlight, opened a ticket with their support but without solution from their side, except saying that the issue could be Z2M.
Regarding Z2M, I built the network with the 2.x and upgraded it with all the official versions. Today, I switch to the dev branch before opening this issue, and the issue is still there.

### What did you expect to happen?

I have no idea where the problem is coming from, the controller , the software, the endpoints (all the sames).

### Zigbee2MQTT version

2.1.1-dev commit: 0d193eb

### Adapter firmware version

7.4.5 [GA]

### Adapter

EmberZNet  firmware 20241127

### Setup

Proxmox LXC

### Debug log

[log.log](https://github.com/user-attachments/files/18931583/log.log)

hexonMD, 2025-02-23:
Same. Seemed to coincide with setting up DIYhue and updating HA core to 2025.2.5

JC38, 2025-02-24:
It can't be HA as we can see the issue into Z2M without having to get HA running.
For the interference, I choose the 25 which is away from the 2.4 Ghz.
Here is a screenshot of the map, and the info that are displayed on the right of the Z2M UI
I will also add a full log in debug because the first one was only set to record errors.

[log2.log](https://github.com/user-attachments/files/18937547/log2.log)

[image]

AlexanderLeporiday, 2025-02-24:
It might help to look at #22516 as this issue has been reported multiple times.
The solutions vary from setup to setup.

Nerivec, 2025-03-03:
Can you check channels usage and existing networks with ember-zli?
https://github.com/Nerivec/ember-zli/wiki#installation
https://github.com/Nerivec/ember-zli/wiki/Stack#scan-network

Can you also check if specific devices are causing the most errors? From glancing at the network map, I'd say those `Nous` plugs are probably not liking the depth of the network.

JC38, 2025-03-05:
@Nerivec: Thanks for your suggestion. I am not at the office these days (the Zigbee network is in an office building). However, I don’t believe any other Zigbee networks are in use. We are the building owners, and no other company is in it that could have installed something else.