Koenkk/zigbee2mqtt #26493: [Wrong device]: Tuya Zigbee Smart USB 3-Gang Adapter

mrmaximas, 2025-02-24:
### Link

https://aliexpress.com/item/1005007279741443.html?sku_id=12000040782257899

### Model

TS0003

### Description

Tuya Zigbee Smart USB 3-Gang Adapter

### Vendor

Tuya

### Picture (link)

https://m.media-amazon.com/images/I/41pn0t0x4eL.jpg

### Database entry

{"id":123,"type":"Router","ieeeAddr":"0xa4c138c8421d41d4","nwkAddr":20228,"manufId":4417,"manufName":"_TZ3000_mw1pqqqt","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAAAgAABAAA"}},"genBasic":{"attributes":{"65503":"U\u0012O/\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0003","manufacturerName":"_TZ3000_mw1pqqqt","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1740395306651}

### Notes

Deivice is recognized as https://www.zigbee2mqtt.io/devices/TS0003.html and works but It's strange that the converter has no power-on behaviour option. 

[image]

The classic TUYA / SL app has this option.

AlexanderLeporiday, 2025-02-24:
There does not appear to be a specific entry for _TZ3000_mw1pqqqt so it's defaulting to the closest matching TS0003.

The device capabilities likely need to be investigated along the lines of [[Support new Tuya devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#support-new-tuya-devices)](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html) as the TS0003 only supports switching be default. Other behaviour needs to be added into. There are many versions of TS0003.

I can't see a USB smart switch already supported, but the closest device appears to be either [TS0003_switch_module_1](https://www.zigbee2mqtt.io/devices/TS0003_switch_module_1.html) or [TS0003_switch_module_2](https://www.zigbee2mqtt.io/devices/TS0003_switch_module_2.html)

For testing you'd want to try making an external converter copying the behaviour of either of these devices, test if it works and then you can include the device finger print to whichever device best matches.

mrmaximas, 2025-02-24:
@AlexanderLeporiday thanks! I don't have much experience in writing external converters, but I did get a DP schema of this device.

```
{
        "name": "3CH Zigbee USB Switch Module ZTU",
        "id": "xxx",
        "key": "xxx>",
        "mac": "a4:c1:38:c8:42:1d:41:d4",
        "uuid": "a4c138c8421d41d4",
        "sn": "HPDQ-03800419",
        "category": "tdq",
        "product_name": " 3CH Zigbee USB Switch Module ZTU",
        "product_id": "mw1pqqqt",
        "biz_type": 18,
        "model": "",
        "sub": true,
        "icon": "https://images.tuyaeu.com/smart/icon/ay1533257659316XGZOd/bd658ce454d954c5142304e044292a92.png",
        "node_id": "a4c138c8421d41d4",
        "mapping": {
            "1": {
                "code": "switch_1",
                "type": "Boolean",
                "values": {}
            },
            "2": {
                "code": "switch_2",
                "type": "Boolean",
                "values": {}
            },
            "3": {
                "code": "switch_3",
                "type": "Boolean",
                "values": {}
            },
            "7": {
                "code": "countdown_1",
                "type": "Integer",
                "values": {
                    "unit": "s",
                    "min": 0,
                    "max": 43200,
                    "scale": 0,
                    "step": 1
                }
            },
            "8": {
                "code": "countdown_2",
                "type": "Integer",
                "values": {
                    "unit": "s",
                    "min": 0,
                    "max": 43200,
                    "scale": 0,
                    "step": 1
                }
            },
            "9": {
                "code": "countdown_3",
                "type": "Integer",
                "values": {
                    "unit": "s",
                    "min": 0,
                    "max": 43200,
                    "scale": 0,
                    "step": 1
                }
            },
            "14": {
                "code": "relay_status",
                "type": "Enum",
                "values": {
                    "range": [
                        "0",
                        "1",
                        "2"
                    ]
                }
            },
            "19": {
                "code": "switch_inching",
                "type": "String",
                "values": "{\"maxlen\":255}"
            }
        },
        "parent": "bf2601a3acbbd6fdxxxxxx"
    },
```

mrmaximas, 2025-03-09:
here is working converter for 3-gang device, supporter power on behavior and timers.
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0003', manufacturerName: '_TZ3000_mw1pqqqt'}],
    model: 'TS0003_USB_switch_3_gang',
    description: 'Tuya 3ch USB Switch',
    vendor: 'Tuya',
    extend: [tuya.modernExtend.tuyaOnOff({switchType: false, indicatorMode: false, onOffCountdown: true, endpoints: ['l1', 'l2', 'l3']})],
    endpoint: (device) => {
        return {l1: 1, l2: 2, l3: 3};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
    },  
};

module.exports = definition;
```

AlexanderLeporiday, 2025-03-09:
Wow that must have taken a lot of trial and error to get working but this looks really good!
Well done!

Have you tested it thoroughly and you're happy that it works as expected?
If so you could add this to https://github.com/Koenkk/zigbee-herdsman-converters/blob/8d21b389364565c8766538b03c02cb0397a5942c/src/devices/tuya.ts as a pull request. :)

mrmaximas, 2025-03-09:
@AlexanderLeporiday thanks! Also i have added timers for _TZ3000_ljhbw1c9 
If it posssible make PR please.
https://github.com/Koenkk/zigbee2mqtt/issues/20162#issuecomment-2709097613

AlexanderLeporiday, 2025-03-09:
I'd recommend you create a fork of https://github.com/Koenkk/zigbee-herdsman-converters and then go to src/devices/tuya.ts and make the required changes there, think of the tuya.ts file as one large converter file where everything already been defined for you.
So you just need to add or change the definition file.

I'd recommend you make a copy of the device under _TZ3000_ljhbw1c9 and move the white label entry to the new entry with extended functionality. Or add logic to the existing converter to add the countdown functionality only to models _TZ3000_ljhbw1c9.

As for the device in this thread, your converter is good and you should just be able to add it.

mrmaximas, 2025-03-09:
@AlexanderLeporiday thanks, i'll try it tomorow ;)

mrmaximas, 2025-03-10:
strange. i have six gang exposed -  1 / 2 / 3 and left / center / right 🤦🏻‍♂️
i don’t know how to fix it.
{
    "countdown_l1": 0,
    "countdown_l2": 0,
    "countdown_l3": 0,
    "last_seen": "2025-03-10T03:36:28+03:00",
    "linkquality": 65,
    "power_on_behavior": "previous",
    "power_on_behavior_l1": "previous",
    "state_center": "ON",
    "state_l1": "ON",
    "state_l2": "ON",
    "state_l3": "OFF",
    "state_left": "ON",
    "state_right": "OFF"
}

mrmaximas, 2025-03-10:
state left / center / right disappeared after deleting and re-adding the device.
i don't know how to fix power on behavior that are displayed twice

```
{
    "countdown_l1": 0,
    "countdown_l2": 0,
    "countdown_l3": 0,
    "last_seen": "2025-03-10T11:50:19+03:00",
    "linkquality": 98,
    "power_on_behavior": "previous",
    "power_on_behavior_l1": "previous",
    "state_l1": "ON",
    "state_l2": "ON",
    "state_l3": "OFF"
}
```