Koenkk/zigbee2mqtt #26502: Hue motion outdoor sensor (9290030674) | Motion sensitivity issue

eekamouze, 2025-02-25:
### What happened?

Hue motion outdoor sensor (9290030674) firmware 2.77.35
Z2M: 2.1.1

Motion sensitivity set on medium, low: never detects any motion (even very close to the device)
Motion sensitivity set on high, very high, max: detects motion with very high sensititvity (don't see any difference between the 3 values)

motion sensitivity behaves like a switch: ON(high, very high,max) / OFF(medium, low)

### What did you expect to happen?

I expect a graduated motion sensitivity from low to max and not only on/off

### How to reproduce it (minimal and precise)

change motion sensitivity using z2m or HA

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

docker container

tzahari, 2025-03-03:
It seems I have the same issue.

bossloper, 2025-03-14:
Same issue after 2.77.35 firmware update.
Motion sensitivity of 'low' or 'medium' never detects. But thanks for pointing out higher sensitivities work, so at least I have a workaround for now.

Hypfer, 2025-05-24:
No new firmwares seem to have been released since then, meaning that this is still relevant.

closed 2025-08-01 by github-actions[bot]