Koenkk/zigbee2mqtt #26505: Zigbee2MQTT crashes and restarts every time I check for OTA updates

ik3rp3r3, 2025-02-25:
### What happened?

I've been using zigbee2MQTT for some weeks now, i's been working fine and I love it. However I recently added a couple of new devices to the network, an one of them, a SONOFF ZBMINIR2 switch, will periodically have an error when using the physical switch connected to it. So, I found that it can be fixed with a firmware update. But I go and try checking for OTA updates, and either the whole Zigbee2MQTT system crashes and restarts, or the adapter does, I cannot know the difference.

The case is that I lose control of all my zigbee devices for some seconds and I cannot access the Zigbee2MQTT frontend for a moment, then, it comes back. I've tried checking for updates of other devices because I thought the new switch could be the problem, but it still does the same for the other devices. I also have realised that if I look at the firmware attribute in HA, for all the devices that support OTA updates, both "installed_version" and "latest_version" have -1 values. 

Could anyone help me figure out what I can do to fix my OTA updates?

Thanks!!

### What did you expect to happen?

Check for updates succesfully and show if there's any new version

### How to reproduce it (minimal and precise)

1. OTA tab
2. Click "Check for updates" in any device

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

8.0.2 [GA]

### Adapter

ZB-GW04 (v1.2)

### Setup

 docker container in RPi3b+

### Debug log

No log is created when it happens, because it instantly crashes after "z2m: Checking if update available for ..."

Nerivec, 2025-03-01:
I saw you mentioned HA, does that mean HAOS with Z2M as addon?
Can you check the system logs when that happens, see if anything is logged there?

_Also note that pi 3 has been known to create troubles with recent versions of HAOS (limited RAM), which translates into Z2M instability._

Glideh, 2025-03-02:
I seem to have the same issue since Z2M 2.1.1 I believe.
Still happening with the 2.1.2-1.
I initialy thought it was just one buggy device (the LiXee TIC v2), but I have now 3 that I can't update.
A friend of mine also has this issue (also having the LixSee device BTW)

I begins the update until 4% max then

```
[2025-03-02 07:10:39] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-03-02 07:10:39] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-03-02 07:10:39] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-03-02 07:10:39] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Total frames: RX=2628, TX=3414
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   DATA frames : RX=2621, TX=783
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   DATA bytes  : RX=65429, TX=31050
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Retry frames: RX=5, TX=3
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=2627
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Retry dupes     : RX=5
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-03-02 07:10:39] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=3
[2025-03-02 07:10:48] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=13 Len=7].
[2025-03-02 07:10:48] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_NCP_FATAL_ERROR. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=13 Len=7].
[2025-03-02 07:10:48] warning: 	zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006206023/1 seMetering.read(["currentTier2SummDelivered","currentTier2SummDelivered","currentTier2SummDelivered","currentTier3SummDelivered"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (~x~> [ZCL to=0x00158d0006206023:60559 apsFrame={"profileId":260,"clusterId":1794,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=FAIL.)
[2025-03-02 07:10:51] warning: 	zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006206023/1 seMetering.read(["currentTier4SummDelivered","currentTier5SummDelivered","currentTier6SummDelivered","meterSerialNumber"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (~x~> [ZCL to=0x00158d0006206023:60559 apsFrame={"profileId":260,"clusterId":1794,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.)
[2025-03-02 07:10:54] warning: 	zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006206023/1 haMeterIdentification.read(["availablePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (~x~> [ZCL to=0x00158d0006206023:60559 apsFrame={"profileId":260,"clusterId":2817,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.)
[2025-03-02 07:11:07] warning: 	zhc:lixee: Failed to read zigbee attributes: Error: ZCL command 0x00158d0006206023/1 liXeePrivate.read(["currentTarif","motDEtat","scheduleHPHC","startNoticeEJP"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":null,"writeUndiv":false}) failed (~x~> [ZCL to=0x00158d0006206023:60559 apsFrame={"profileId":260,"clusterId":65382,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.)
[2025-03-02 07:11:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Chiottes - Lave linge', payload '{"child_lock":"LOCK","countdown":0,"current":0,"energy":190.91,"indicator_mode":"off/on","linkquality":255,"power":0,"power_outage_memory":"restore","state":"ON","update":{"installed_version":192,"latest_version":192,"state":"idle"},"update_available":false,"voltage":234}'
[2025-03-02 07:11:10] info: 	zh:ember:uart:ash: Port closed.
[2025-03-02 07:11:10] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[... cut ...]
[2025-03-02 07:11:10] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-03-02 07:11:10] error: 	z2m: Adapter disconnected, stopping
[2025-03-02 07:11:10] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-03-02 07:11:10] info: 	z2m: Disconnecting from MQTT server
[2025-03-02 07:11:10] info: 	z2m: Stopping zigbee-herdsman...
[2025-03-02 07:11:10] info: 	z2m: Stopped zigbee-herdsman
[2025-03-02 07:11:10] info: 	z2m: Stopped Zigbee2MQTT
```

Nerivec, 2025-03-02:
@Glideh are you running the latest 8.0.2 firmware? It should help with the LiXee (that device is very taxing on the network).
https://github.com/Nerivec/silabs-firmware-builder/releases

ik3rp3r3, 2025-03-02:
> I saw you mentioned HA, does that mean HAOS with Z2M as addon? Can you check the system logs when that happens, see if anything is logged there?
> 
> _Also note that pi 3 has been known to create troubles with recent versions of HAOS (limited RAM), which translates into Z2M instability._

No, I have both implemented as docker container. I don't think it's a RAM problem, because when I check for updates, only z2m crashes, and not HA. But also, I've tried stopping all the rest of the containers and checking for updates just with z2m running, and it crashed anyway.

Thanks!

Nerivec, 2025-03-02:
> I don't think it's a RAM problem

Indeed that's limited to HAOS, which isn't your case.

The [logic for OTA checking](https://github.com/Koenkk/zigbee-herdsman-converters/blob/b18f23eb70499f7cf18c359ceec758569bffd5ca/src/lib/ota.ts#L688-L729) involves several mechanisms (and it could also just be a side-effect of triggering the check that makes it crash...).
Make sure to set `log_level` to `debug`, so Z2M tries to log as much as possible (before crashing), which would give us a clue at which step exactly it is crashing.

Glideh, 2025-03-03:
Ok indeed I missread the topic, mine is about actually doing the OTA update. But it crashes with any device (not only with the LiXee)
> Are you running the latest 8.0.2 firmware?

Are you talking about the coordinator firmware ?

[image]

ik3rp3r3, 2025-03-03:
> Indeed that's limited to HAOS, which isn't your case.
>
> The [logic for OTA [...]

I've tried with debug logging mode. This is the log before the crash:

```
[2025-03-03 11:54:18] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Interruptor despacho","transaction":"sfauv-1"}'
[2025-03-03 11:54:18] info:     z2m: Checking if update available for 'Interruptor despacho'
[2025-03-03 11:54:18] debug:    zhc:ota: [0x1cc089fffe5569fc | ZBMINIR2] Checking if an update is available
[2025-03-03 11:54:18] debug:    zhc:ota: [0x1cc089fffe5569fc | ZBMINIR2] Using endpoint '1'
[2025-03-03 11:54:18] debug:    zh:controller:endpoint: CommandResponse 0x1cc089fffe5569fc/1 genOta.imageNotify({"payloadType":0,"queryJitter":100}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"})
[2025-03-03 11:54:18] debug:    zh:ember: ~~~> [ZCL to=0x1cc089fffe5569fc:34698 apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":1,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":19,"commandIdentifier":0}]
[2025-03-03 11:54:18] debug:    zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=145 Len=27]
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=6](ackRx=1)
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=1 frmTx=2)
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=6](frmRx=6) Added to rxQueue
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=2)
[2025-03-03 11:54:18] debug:    zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=145 Len=10]
[2025-03-03 11:54:18] debug:    zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=75 messageTag=85 status=OK]
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=2 frmTx=2)
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=7](frmRx=7) Added to rxQueue
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=2)
[2025-03-03 11:54:18] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=145 Len=26]
[2025-03-03 11:54:18] debug:    zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=DIRECT], [indexOrDestination=34698], [apsFrame={"profileId":260,"clusterId":25,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":75}], [messageTag=85]
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=2 frmTx=2)
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=0](frmRx=0) Added to rxQueue
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=2)
[2025-03-03 11:54:18] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=145 Len=18]
[2025-03-03 11:54:18] debug:    zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=34698], [sourceEui=0x1cc089fffe5569fc], [lastHopLqi=255], [lastHopRssi=-39], [relayCount=0], [relayList=]
[2025-03-03 11:54:18] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
```

And these are some lines of the log when it restarts:

```
[2025-03-03 11:54:49] debug:    z2m: Removing old log directory '/app/data/log/2025-02-28.10-48-10'
[2025-03-03 11:54:49] info:     z2m: Logging to console, file (filename: log.log)
[2025-03-03 11:54:49] debug:    z2m: Loaded state from file /app/data/state.json
[2025-03-03 11:54:49] info:     z2m: Starting Zigbee2MQTT version 2.1.1 (commit #1accb8b66b95b99658c50e4c0209edcb946a5fe2)
[2025-03-03 11:54:49] debug:    z2m: sd-notify loaded
[2025-03-03 11:54:49] info:     z2m: Starting zigbee-herdsman (3.2.5)
[2025-03-03 11:54:49] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":31996,\"extendedPanID\":[127,244,87,201,17,139,136,48],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2025-03-03 11:54:49] debug:    zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"0658","pnpId":"usb-0658_0200-if00","vendorId":"0658","productId":"0200"},{"path":"/dev/ttyAMA0"},{"path":"/dev/ttyUSB0","manufacturer":"1a86","pnpId":"usb-1a86_USB_Serial-if00-port0","vendorId":"1a86","productId":"7523"},{"path":"/dev/ttyUSB1","manufacturer":"1a86","pnpId":"usb-1a86_USB_Serial-if00-port0","vendorId":"1a86","productId":"7523"}]
[2025-03-03 11:54:49] debug:    zh:adapter:discovery: Unable to match USB adapter: ember | /dev/ttyACM0
[2025-03-03 11:54:50] info:     zh:ember: Using default stack config.
[... cut ...]
[2025-03-03 11:54:51] debug:    zh:ember:uart:ash: Waiting for RSTACK... 700/2500
[2025-03-03 11:54:51] debug:    zh:ember:uart:ash: Waiting for RSTACK... 800/2500
[2025-03-03 11:54:51] debug:    zh:ember:uart:ash: Waiting for RSTACK... 900/2500
[2025-03-03 11:54:51] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1000/2500
[2025-03-03 11:54:51] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1100/2500
[2025-03-03 11:54:51] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1200/2500
[2025-03-03 11:54:51] debug:    zh:ember:uart:ash: <--- [FRAME type=RSTACK]
```

Nerivec, 2025-03-03:
@Glideh to clarify, you have two firmware on the 06m, one for the "radio" (EFR32) which is the zigbee coordinator firmware, and one smlight calls "core" (ESP32) which is the firmware used by the adapter to communicate between the ethernet network and zigbee. Your second screenshot corresponds to the left column in your first screenshot.
You have version 7.4.1, which doesn't have the fixes for busy networks. See if you can update to 8.0.2 https://github.com/darkxst/silabs-firmware-builder/releases/tag/20250220 (note: darkxst builds firmware for smlight, so these are the same you would get from the smlight flasher).
_The LiXee device creates a very busy environment on the network (tons of reported data, lots of messages), it doesn't need to be about flashing, just its presence on the network is enough to make things quite messy. So, when you start updating any other device, or itself, you end up with an extremely busy network._

---

@ik3rp3r3 this is strange, it appears to be crashing after a simple zigbee message, not during any OTA-specific operation. Do you get more info from containers logs maybe? `docker logs -f CONTAINER_ID`

Glideh, 2025-03-03:
Thank you for these details @Nerivec it's much appreciated. I'll try to update the radio firmware then.

Glideh, 2025-03-03:
@Nerivec everything is working fine now with the 8.0.2, thanks again !

ik3rp3r3, 2025-03-03:
> [@ik3rp3r3](https://github.com/ik3rp3r3) this is strange, it appears to be crashing after a simple zigbee message, not during any OTA-specific operation. Do you get more info from containers logs maybe? `docker logs -f CONTAINER_ID`

Well something different reported, at least, following your instructions:

```
[2025-03-03 18:52:07] debug:    z2m: Received Zigbee message from 'Interruptor despacho', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":4098,"imageType":8,"manufacturerCode":4742}' from endpoint 1 with groupID 0
[2025-03-03 18:52:07] debug:    zhc:ota: [0x1cc089fffe5569fc | ZBMINIR2] Got request '{"fieldControl":0,"manufacturerCode":4742,"imageType":8,"fileVersion":4098}'
[2025-03-03 18:52:07] debug:    zhc:ota: [0x1cc089fffe5569fc | ZBMINIR2] Checking latest image availability, current: {"fieldControl":0,"manufacturerCode":4742,"imageType":8,"fileVersion":4098}
[2025-03-03 18:52:07] debug:    zhc:ota: [0x1cc089fffe5569fc | ZBMINIR2] Getting image metadata...

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0x73fd0744
```

Nerivec, 2025-03-03:
What's the OS running on your pi?
Can you post the outputs of these commands?
`uname -a`
`uname -m`
`cat /etc/issue.net`
`docker --version`
and for good measure:
`free -h`

ik3rp3r3, 2025-03-03:
> What's the OS running on your pi? Can you post the outputs of these commands? `uname -a` `uname -m` `cat /etc/issue.net` `docker --version` and for good measure: `free -h`

It is Raspbian 10, but more specifically Octopi 0.17.0, based on Raspbian. The rpi started as my octoprint server and it's been acquiring further functionality as years passed by. It's now my octoprint server + cups server + HA + z2m...

It's true that it might be running too many things at once for everything to run well, but it hasn't been a problem until now, and also as I said, I've tried checking for updates with everything else stopped and it doesn't work either, so even if there is a ram problem, it's not the only one.

Could it be that I'm missing any system library needed for some function in particular called when OTA updating?

This is the output of what you asked:

```
pi@ender3:~ $ uname -a
Linux ender3 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
pi@ender3:~ $ uname -m
armv7l
pi@ender3:~ $ cat /etc/issue.net
Raspbian GNU/Linux 10
pi@ender3:~ $ docker --version
Docker version 26.1.4, build 5650f9b
pi@ender3:~ $ free -h
              total        used        free      shared  buff/cache   available
Mem:          871Mi       592Mi        29Mi        25Mi       249Mi       200Mi
Swap:          99Mi        99Mi          0B
```

Nerivec, 2025-03-05:
Can you check for OS updates (v10 is from 2019, you should have v11 with octopi 1.0.0)?
Do the same for docker too.

ik3rp3r3, 2025-03-10:
Tried `apt upgrade`, tried `apt dist-upgrade` and tried upgrading docker too, and nothing worked...

However! I found this post [https://github.com/Koenkk/zigbee2mqtt/issues/12459](url) with similar error (though in different situation), and its instructions solved my problem too. Following this step solved it [https://www.zigbee2mqtt.io/guide/installation/02_docker.html#notes-for-raspberry-pi-users](url). I feel it's weird because according ti the documentation, Z2M was just supposed to not work in the first place if I didn't do that step, and mine worked perfectly until I would try to OTA update, but whatever... It works now

closed 2025-03-10 by ik3rp3r3