Koenkk/zigbee2mqtt #26511: [New device support]: _TZE204_zhiqbr7l

Grigorij858947, 2025-02-26:
### Link

https://kojima.ru/catalog/osveshchenie/kojima-rgbic-cct-sdc-umnyy-zigbee-kontroller-rgbic-cct-dlya-adresnoy-lenty-/

### Database entry

Device '0xa4c138e147b633e6' with Zigbee model 'TS0601' and manufacturer name '_TZE204_zhiqbr7l'

### Zigbee2MQTT version

1.42.0 

### Comments

https://kojima.ru/catalog/osveshchenie/kojima-rgbic-cct-sdc-umnyy-zigbee-kontroller-rgbic-cct-dlya-adresnoy-lenty-/

NOT supported

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_zhiqbr7l',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

NOT supported

Grigorij858947, 2025-02-27:
{"id":2,"type":"Router","ieeeAddr":"0xa4c138e147b633e6","nwkAddr":18336,"manufId":4417,"manufName":"_TZE204_zhiqbr7l","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000bvQ/f\fvQ/\u0012\u000evQ/e","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_zhiqbr7l","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1740554402567}

closed 2025-05-06 by github-actions[bot]

cinsler78, 2025-06-14:
Do I have a device too? Will there be support?