Koenkk/zigbee2mqtt #26515: TZ3000_pl5v1yyy is not defined correctly

Dryvise, 2025-02-26:
### What happened?

Hello! I bought a zigbee surge protector (4 sockets and 1 USB). In z2m, the device is defined as TS011F_plug_1, so only one socket works.
I found that my device was added to the supported list 3 weeks ago (https://github.com/Koenkk/zigbee2mqtt/issues/26100), but my device is not working properly. Please help.

zigbee-herdsman-converters 23.1.1
zigbee-herdsman 3.2.7

`{"id":35,"type":"Router","ieeeAddr":"0xa4c138933133f65a","nwkAddr":63933,"manufId":4417,"manufName":"_TZ3000_pl5v1yyy","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,3,4,5,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":",X .\u0017.X .\u0013.X .\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_pl5v1yyy","powerSource":1,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffec6db90","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffec6db90","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffec6db90","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":257}],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1740564144294}`

`
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS011F'],
    model: 'TS011F',
    vendor: '_TZ3000_pl5v1yyy',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"4":4,"5":5}}), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2","3","4","5"]}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
`

URL: https://market.yandex.ru/product--umnyi-energosberegaiushchii-setevoi-filtr-zigbee-tyua-4-rozetki-i-4-usb-slota-nuzhen-shliuz/670648128?sku=103526529488&uniqueId=83910149&do-waremd5=7bUxuj9IBivOKBXwG2NMnw

Thanks!

### What did you expect to happen?

Not all functions of the device work - there is no control of 4 sockets and USB

### Zigbee2MQTT version

2.1.1-dev commit: 102857a

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

Addon

234256, 2025-05-05:
actual

Dryvise, 2025-06-29:
> Hi! I ran into this problem too. I created a basic external converter for this device. Feel free to try it out—it might help: https://github.com/antesl/hass_example/blob/main/homeassistant/zigbee2mqt/external_converters/RH-EU411-4U-Z.js

Hi! Thanks!
Yes, all inputs are displayed, but when trying to turn any of them on or off, they all turn on or off together.

Andrew-Rayand, 2025-08-12:
To solve this problem, I wrote my own converter 
https://github.com/Andrew-Rayand/RH-EU411-4U-Z-zigbee2mqtt-converter/tree/main

Andrew-Rayand, 2025-08-12:
> Hi! Thanks! Yes, all inputs are displayed, but when trying to turn any of them on or off, they all turn on or off together.

To solve this problem, I wrote my own converter 
https://github.com/Andrew-Rayand/RH-EU411-4U-Z-zigbee2mqtt-converter/tree/main

closed 2025-10-19 by github-actions[bot]