Koenkk/zigbee2mqtt #26520: Groups need to remove `color_mode` and add `supported_color_modes` to be compatible with Home Assistant 2025.3

rccoleman, 2025-02-26:
### What happened?

`color_mode` was removed in https://github.com/Koenkk/zigbee2mqtt/pull/22142 for `light` devices exposed to Home Assistant, but wasn't removed from Zigbee groups (defined in `lib/extension/groups.ts`).  When I remove that manually, groups are also missing `supported_color_modes` that are now required and are defined for regular devices [here](https://github.com/Koenkk/zigbee2mqtt/blob/1accb8b66b95b99658c50e4c0209edcb946a5fe2/lib/extension/homeassistant.ts#L546).  A similar change needs to be made for groups.

### What did you expect to happen?

I expect for groups discovered by Home Assistant to not expose `color_mode` and to expose `supported_color_modes`, as individual devices do.

### How to reproduce it (minimal and precise)

Start HA 2025.3.0b0 when Z2M advertises groups containing lights and observe warnings for the unsupported `color_mode` attribute and when that's removed, the missing `supported_color_modes` attribute.

### Zigbee2MQTT version

2.1.1-dev commit: 102857a7fec60e3f65887d92b6fe7a2d81add110

### Adapter firmware version

20240710

### Adapter

zStack Tubeszb PoE

### Setup

Home Assistant container and Z2M container

### Debug log

Example of HA log running 2025.3.0b0 after removing `color_mode` in the discovered schema:

```
2025-02-26 11:02:38.272 WARNING (MainThread) [homeassistant.components.light] light.uph_light_group (<class 'homeassistant.components.mqtt.light.schema_json.MqttLightJson'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

I can't the `color_mode` message to show up now, but it's similar.

hmmbob, 2025-02-27:
It is not only groups, I have more dimmable lights that all throw the same warning:

```
2025-02-27 08:19:10.768 WARNING (MainThread) [homeassistant.components.light] light.wc_boven (<class 'homeassistant.components.mqtt.light.schema_json.MqttLightJson'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-02-27 08:19:10.773 WARNING (MainThread) [homeassistant.components.light] light.eetkamer_tafel (<class 'homeassistant.components.mqtt.light.schema_json.MqttLightJson'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-02-27 08:19:10.782 WARNING (MainThread) [homeassistant.components.light] light.spots_vaas (<class 'homeassistant.components.mqtt.light.schema_json.MqttLightJson'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-02-27 08:19:10.786 WARNING (MainThread) [homeassistant.components.light] light.badkamer (<class 'homeassistant.components.mqtt.light.schema_json.MqttLightJson'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-02-27 08:19:10.792 WARNING (MainThread) [homeassistant.components.light] light.spots_keuken (<class 'homeassistant.components.mqtt.light.schema_json.MqttLightJson'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-02-27 08:19:10.798 WARNING (MainThread) [homeassistant.components.light] light.spots_eetkamer (<class 'homeassistant.components.mqtt.light.schema_json.MqttLightJson'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-02-27 08:19:10.805 WARNING (MainThread) [homeassistant.components.light] light.spots_woonkamer (<class 'homeassistant.components.mqtt.light.schema_json.MqttLightJson'>) does not set supported color modes, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

Also: they now only show as on/off and not as dimmable lights.

[image]

@Koenkk Tagging for awareness as HA 2025.3 will create a lot of issues for users if z2m isn't updated before that.... (it's a breaking change in HA)

hmmbob, 2025-02-27:
Adding to this: it's not all groups that have this issue. I have exactly 1 group in my z2m config, and it is working fine and not throwing errors.

Discovery message:

```json
{
   "availability":[
      {
         "topic":"zigbee2mqtt/bridge/state",
         "value_template":"{{ value_json.state }}"
      }
   ],
   "brightness":true,
   "brightness_scale":254,
   "command_topic":"zigbee2mqtt/inloopkast_t/set",
   "device":{
      "configuration_url":"https://--snip--//#/group/1",
      "identifiers":[
         "zigbee2mqtt_1221051039810110150109113116116_1"
      ],
      "manufacturer":"Zigbee2MQTT",
      "model":"Group",
      "name":"inloopkast_t",
      "sw_version":"Zigbee2MQTT 2.1.1",
      "via_device":"zigbee2mqtt_bridge_0x00124b0024c94c43"
   },
   "effect":true,
   "effect_list":[
      "blink",
      "breathe",
      "okay",
      "channel_change",
      "finish_effect",
      "stop_effect"
   ],
   "max_mireds":454,
   "min_mireds":250,
   "name":null,
   "object_id":"inloopkast_t",
   "origin":{
      "name":"Zigbee2MQTT",
      "sw":"2.1.1",
      "url":"https://www.zigbee2mqtt.io"
   },
   "schema":"json",
   "state_topic":"zigbee2mqtt/inloopkast_t",
   "supported_color_modes":[
      "color_temp"
   ],
   "unique_id":"1_light_zigbee2mqtt"
}

```

As a matter of fact, 6 out of 7 devices that DO throw the error in my case are of the same type: EcoDim | Eco-Dim.07/Eco-Dim.10. 
The 7th device throwing the error is a ROBB | ROB_200-004-0

I have other devices with color_mode that just work fine.

hmmbob, 2025-02-27:
Yeah, it is because core is checking for a color_mode that supports brightness, and z2m isn't sending a color_mode for these lights that.... dont't support color mode:

https://github.com/home-assistant/core/blob/3ebad8870c2018c3f29245632319fa57305fd23b/homeassistant/components/mqtt/light/schema_json.py#L461-L470

```py
        if ATTR_BRIGHTNESS in kwargs and brightness_supported(
            self.supported_color_modes
        ):
            device_brightness = color_util.brightness_to_value(
                (1, self._config[CONF_BRIGHTNESS_SCALE]),
                kwargs[ATTR_BRIGHTNESS],
            )
            # Make sure the brightness is not rounded down to 0
            device_brightness = max(round(device_brightness), 1)
            message["brightness"] = device_brightness
```

Koenkk, 2025-02-27:
`color_mode` has been removed a long time ago: https://github.com/Koenkk/zigbee2mqtt/pull/22142, I think this only happens for bulbs that only support brightness?

hmmbob, 2025-02-27:
> `color_mode` has been removed a long time ago: [#22142](https://github.com/Koenkk/zigbee2mqtt/pull/22142), I think this only happens for bulbs that only support brightness?

Correct

hmmbob, 2025-02-27:
Confirmed that setting the `ColorMode.BRIGHTNESS` fixes the issue.

I've just sent this discovery message for the same device as mentioned here https://github.com/Koenkk/zigbee2mqtt/issues/26520#issuecomment-2687142504 with `   "supported_color_modes":[
      "brightness"
   ],` added to it:

```json
{
   "availability":[
      {
         "topic":"zigbee2mqtt/bridge/state",
         "value_template":"{{ value_json.state }}"
      }
   ],
   "brightness":true,
   "brightness_scale":254,
   "command_topic":"zigbee2mqtt/Spots vaas/set",
   "device":{
      "configuration_url":"https://--snip--//#/device/0x8cf681fffe893b5f/info",
      "hw_version":1,
      "identifiers":[
         "zigbee2mqtt_0x8cf681fffe893b5f"
      ],
      "manufacturer":"EcoDim",
      "model":"Zigbee & Z-wave dimmer",
      "model_id":"Eco-Dim.07/Eco-Dim.10",
      "name":"Spots vaas",
      "sw_version":"1.01",
      "via_device":"zigbee2mqtt_bridge_0x00124b0024c94c43"
   },
   "effect":true,
   "effect_list":[
      "blink",
      "breathe",
      "okay",
      "channel_change",
      "finish_effect",
      "stop_effect"
   ],
   "name":null,
   "object_id":"spots_vaas",
   "origin":{
      "name":"Zigbee2MQTT",
      "sw":"2.1.1",
      "url":"https://www.zigbee2mqtt.io"
   },
   "schema":"json",
   "state_topic":"zigbee2mqtt/Spots vaas",
   "supported_color_modes":[
      "brightness"
   ],
   "unique_id":"0x8cf681fffe893b5f_light_zigbee2mqtt"
}
```
and immediately it's a dimmable light again:

[image]

rccoleman, 2025-02-27:
Whatever happens in homeassistant.ts needs to replicated in groups.ts.

Bluhme1, 2025-02-28:
@hmmbob Where do you the "supported_color_modes: In what file and where can I find it?

hmmbob, 2025-02-28:
> [@hmmbob](https://github.com/hmmbob) Where do you the "supported_color_modes: In what file and where can I find it?

Not sure what you are asking.... I've linked all info/code I found above.

Adding the `color_mode: brightness` probably needs to happen somewhere in this Zigbee2Mqtt code:

https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.ts#L524-L557

Bluhme1, 2025-02-28:
Let me rephrase. You wrote above that you succeeded in making the device work again by adding "supported_color_modes":[ "brightness" ],My question is: where did you add it?

hmmbob, 2025-02-28:
In the MQTT discovery message for the device.

The original discovery message is in this comment, including the topic (which is specific to this light, yours will vary) https://github.com/Koenkk/zigbee2mqtt/issues/26520#issuecomment-2687142504

The updated discovery message I sent manually to the same topic is in this comment: https://github.com/Koenkk/zigbee2mqtt/issues/26520#issuecomment-2687342812

So it is not a code change, I manually adjusted the already sent discovery message that z2m currently sends to include the `color_mode: brightness` fields (look towards the end of the new message). Every HA or Z2M reboot will overwrite this message though, so it is hardly a good workaround for now -  it it was simply for testing the theory that this was the missing link.

Bluhme1, 2025-02-28:
OK got it. Thank you very much

Koenkk, 2025-03-01:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-03-01 by Koenkk

MSL-DA, 2025-03-01:
https://github.com/Koenkk/zigbee2mqtt/discussions/26551#discussioncomment-12362423

hmmbob, 2025-03-01:
@Koenkk color_mode brightness is exclusive (cannot be combined with the other color modes)

See https://github.com/home-assistant/core/blob/dev/homeassistant/components/light/__init__.py#L133

Koenkk, 2025-03-02:
@hmmbob thanks, fixed in 2.1.3