Koenkk/zigbee2mqtt #26521: OTA Update fail on TS011F_plug_3

CrazYoshi, 2025-02-26:
### What happened?

When I try to check for firmware updates using Zigbee2Mqtt UI I'm getting errors and I'm not able to see if a new firmware is available

### What did you expect to happen?

Check firmware and update if a new one is found

### How to reproduce it (minimal and precise)

Click on Check for new updates on OTA tab for a Tuya device (TS011F_plug_3 in this case) 

### Zigbee2MQTT version

2.1.1 commit: 1accb8b66b95b99658c50e4c0209edcb946a5fe2

### Adapter firmware version

20240710

### Adapter

 SONOFF ZB dongle-p zStack3x0

### Setup

x86-64 on Docker compose

### Debug log

info 2025-02-26 21:39:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{},"error":"Failed to check if update available for 'forza_upsserver' ([0x70b3d52b600bb7b3 | TS011F] Failed to find an endpoint which supports the OTA cluster)","status":"error","transaction":"0i0he-4"}'
error 2025-02-26 21:39:26z2m: Failed to check if update available for 'forza_upsserver' ([0x70b3d52b600bb7b3 | TS011F] Failed to find an endpoint which supports the OTA cluster)

Nerivec, 2025-03-02:
Looks like a Tuya white label model that doesn't have support for OTA.
Can you post a screenshot of the `Clusters` tab of the device in frontend?

CrazYoshi, 2025-03-03:
This is my Clusters
[image]

santia70, 2025-04-02:
I have the same problem, any suggestions?

CrazYoshi, 2025-04-07:
not at the moment.. I hope somebody will solve this

CrazYoshi, 2025-06-07:
Please, somebody can tell me know can i help to solve this issue?

der-seemann, 2025-08-26:
same here...
genOTA is shown but does not react on any command.

CrazYoshi, 2025-10-26:
How can i investigate by myself? How can i do that? 
I wanna fix this issue

closed 2026-02-01 by github-actions[bot]