Koenkk/zigbee2mqtt #26525: Removing Serial: section solved startup error in version 2.1.1

peterhow11, 2025-02-26:
### What happened?

After upgrading to 2.1.1, I was constantly getting this error on start-up:

[2025-02-27 06:28:12] error: 	z2m: Error while starting zigbee-herdsman
[2025-02-27 06:28:12] error: 	z2m: Failed to start zigbee-herdsman
[2025-02-27 06:28:12] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-02-27 06:28:12] error: 	z2m: Exiting...
[2025-02-27 06:28:12] error: 	z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.

In the previous version (1.42.0-2), the configuration.yaml contained the following:

serial:
  adaptor: zstack
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_004e3cc63239ec11a592e0680aac08d5-if00-port0

After noticing that the new version has "auto-device discovery", I removed the entire "serial:" section above and everything started working!  

I can only conclude that there is an error in the serial section but why did it work with the previous version? I checked the port in the hardware section and it is correct, as is the adaptor type.  

Just a little concerned that something might break in the future....

I have not updated the version of the firmware in the Sonoff adaptor since it appears to be working.

### Zigbee2MQTT version

2.1.1

### Adapter firmware version

20211217

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Intel NUC, HaOS in Docker

Nerivec, 2025-03-01:
This is a little strange.
`No valid USB adapter found` means it didn't find any USB currently connected, so, that usually means something's off with the discovery, hence, requires explicitly setting `adapter` and `port` settings (that is, assuming the adapter is actually connected and working).
But the removal of the section forces the use of the discovery, which means at that point, it definitely worked.
The two scenario report conflicting states.

This would point to the USB maybe getting disconnected sometimes? You might want to check the system logs to confirm.

Note: guessing it's a typo from writing the post, but `adaptor` should be `adapter` in the config.

peterhow11, 2025-03-02:
Thanks for the feedback. In fact, I **had** the misspelled "adapter" in the file. I added back the serial section with this correction and it now works fine.  So the problem is...  if you include the correct port address but have a missing or misspelled "adapter" entry then it fails as described above.  Possibly the default adapter setting is not "zstack"?  Since it worked with the misspelled word in 1.42.0-2, I assume this setting is ignored in that version or it did default to "zstack" then?

closed 2025-03-02 by Nerivec