Koenkk/zigbee2mqtt #26526: [New device support]: Tuya _TZE200_qzaing2g TS0601 1 Gang Smart Dimmer

NoahColly, 2025-02-27:
### Link

https://smarthomesmatter.com.au/products/zigbee-touch-dimmer-switch

### Database entry

{"id":47,"type":"Router","ieeeAddr":"0xa4c138346c44e3ee","nwkAddr":47295,"manufId":4417,"manufName":"_TZE200_qzaing2g","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u0019S/i","65506":53,"65508":0,"stackVersion":0,"dateCode":"","appVersion":65,"modelId":"TS0601","manufacturerName":"_TZE200_qzaing2g","powerSource":1,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1740634643440}

### Zigbee2MQTT version

2.1.1

### Comments

Trying to add a new smart dimmer and it looks like its just not supported.
I had a very similar dimmer previously that worked however this new one (leading edge vs trailing edge dimming) doesn't seem to be supported.
I have tried adding my own external definitions but I just cant get it to work. As can see below, I have just copied out the definition that already exists and added the new one to the fingerprint but I just cant seem to make it work.

[image]

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const definition = {
{
        fingerprint: tuya.fingerprint("TS0601", [
            "_TZE200_ip2akl4w",
            "_TZE200_1agwnems",
            "_TZE200_la2c2uo9",
            "_TZE200_579lguh2",
            "_TZE200_vucankjx",
            "_TZE200_4mh6tyyo",
            "_TZE204_hlx9tnzb",
            "_TZE204_n9ctkb6j",
            "_TZE204_9qhuzgo0",
            "_TZE200_9cxuhakf",
            "_TZE200_a0syesf5",
            "_TZE200_3p5ydos3",
            "_TZE200_swaamsoy",
            "_TZE200_ojzhk75b",
            "_TZE200_w4cryh2i",
            "_TZE200_dfxkcots",
            "_TZE200_9i9dt8is",
            "_TZE200_ctq0k47x",
            "_TZE200_ebwgzdqq",
            "_TZE204_vevc4c6g",
            "_TZE200_0nauxa0p",
            "_TZE200_ykgar0ow",
            "_TZE200_qzaing2g",
        ]),
        model: "TS0601_dimmer_1_gang_1",
        vendor: "Tuya",
        description: "1 gang smart dimmer",
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: (device, options) => {
            const exps: Expose[] = [
                tuya.exposes.lightBrightnessWithMinMax(),
                tuya.exposes.countdown(),
                e.power_on_behavior().withAccess(ea.STATE_SET),
            ];

            if (!device || !["_TZE200_ykgar0ow"].includes(device.manufacturerName)) {
                exps.push(tuya.exposes.lightType(), tuya.exposes.backlightModeOffNormalInverted().withAccess(ea.STATE_SET));
            }

            return exps;
        },
        meta: {
            tuyaDatapoints: [
                [1, "state", tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
                [2, "brightness", tuya.valueConverter.scale0_254to0_1000],
                [3, "min_brightness", tuya.valueConverter.scale0_254to0_1000],
                [4, "light_type", tuya.valueConverter.lightType],
                [5, "max_brightness", tuya.valueConverter.scale0_254to0_1000],
                [6, "countdown", tuya.valueConverter.countdown],
                [14, "power_on_behavior", tuya.valueConverter.powerOnBehavior],
                [21, "backlight_mode", tuya.valueConverter.backlightModeOffNormalInverted],
            ],
        },
        whiteLabel: [
            {vendor: "Lerlink", model: "X706U"},
            {vendor: "Moes", model: "ZS-EUD_1gang"},
            {vendor: "Larkkey", model: "ZSTY-SM-1DMZG-EU"},
            {vendor: "Earda", model: "EDM-1ZAA-EU"},
            {vendor: "Earda", model: "EDM-1ZAB-EU"},
            {vendor: "Earda", model: "EDM-1ZBA-EU"},
            {vendor: "Mercator Ikuü", model: "SSWD01"},
            {vendor: "Moes", model: "ZS-USD"},
            {vendor: "Moes", model: "EDM-1ZBB-EU"},
            tuya.whitelabel("Moes", "ZS-SR-EUD-1", "Star ring smart dimmer switch 1 gang", ["_TZE204_hlx9tnzb"]),
            tuya.whitelabel("Moes", "MS-105Z", "Smart Dimmer module", ["_TZE200_la2c2uo9"]),
            tuya.whitelabel("Mercator Ikuü", "SSWM-DIMZ", "Switch Mechanism", ["_TZE200_9cxuhakf"]),
            tuya.whitelabel("Mercator Ikuü", "SSWRM-ZB", "Rotary dimmer mechanism", ["_TZE200_a0syesf5"]),
            tuya.whitelabel("Lonsonho", "EDM-1ZBB-EU", "Smart Dimmer Switch", ["_TZE200_0nauxa0p"]),
            tuya.whitelabel("ION Industries", "ID200W-ZIGB", "LED Zigbee Dimmer", ["_TZE200_ykgar0ow"]),
        ],
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

As mentioned I just cannot seem to get the external definition to work.

I have it located in /homeassistant/zigbee2mqtt/external_converters/tuya_dimmer.js
I cannot see it in the logs and is not reflecting in the front end.

Koenkk, 2025-03-01:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE200_qzaing2g` to `_TZE200_a0syesf5`
- Start Z2M and see if the device now works

NoahColly, 2025-03-05:
Hey there,
I gave that a shot with the one you mentioned and the below - none of which seemed to work.
HA would show it however when I control the light (on/off or brightness) nothing would happen.
`_TZE200_a0syesf5
_TZE200_ip2akl4w
_TZE200_ykgar0ow
_TZE200_1agwnems
_TZE200_la2c2uo9
_TZE200_579lguh2`

NoahColly, 2025-03-11:
Hey there. Just touching base to see if you had any luck looking into this one?
Thank you in advance!

Koenkk, 2025-03-12:
Could you try with `_TZE200_whpb9yts`?

NoahColly, 2025-03-12:
Hey hey, I gave that one a shot too and still doesn't seem to work. Its showing all the exposes and what not but setting brightness or on/off does nothing.
Is there anything I can provide to help?

Koenkk, 2025-03-13:
Seems that the device than really works different than the others, you will need the Tuya gateway to figure out the datapoints: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

NoahColly, 2025-03-13:
So I have had a bit of a play with it again this morning...Now the Tuya app is being stupid (wont let me set the brightness with the slider)
I also tried adding it to HA with the Tuya integration and it doesn't give me the option to even turn it on or off...

[image]

I have however gotten the DP dump but I cant seem to see its state or brightness value here.
When looking at the logs, DP ID 'Bright value' is what seems to set the brightness and DP ID 'Dimmer' is what turns the switch on/off.

Its all a lil confusing as to why its not working the way it did a few weeks ago...

`{
  "result": {
    "properties": [
      {
        "code": "backlight_switch",
        "custom_name": "",
        "dp_id": 16,
        "name": "Backlight",
        "time": 1741906057489,
        "type": "bool",
        "value": true
      },
      {
        "code": "add_ele",
        "custom_name": "",
        "dp_id": 20,
        "time": 1741905566504,
        "type": "value",
        "value": 5
      },
      {
        "code": "cur_current",
        "custom_name": "",
        "dp_id": 21,
        "time": 1741905991821,
        "type": "value",
        "value": 65
      },
      {
        "code": "cur_power",
        "custom_name": "",
        "dp_id": 22,
        "time": 1741905991942,
        "type": "value",
        "value": 13
      },
      {
        "code": "cur_voltage",
        "custom_name": "",
        "dp_id": 23,
        "time": 1741905991889,
        "type": "value",
        "value": 2426
      },
      {
        "code": "child_lock",
        "custom_name": "",
        "dp_id": 101,
        "time": 1741905566729,
        "type": "bool",
        "value": false
      },
      {
        "code": "backlight_color",
        "custom_name": "",
        "dp_id": 107,
        "time": 1741905566824,
        "type": "raw",
        "value": "BmQA5gPo"
      },
      {
        "code": "switch_led_1",
        "custom_name": "",
        "dp_id": 141,
        "time": 1741905990323,
        "type": "bool",
        "value": false
      },
      {
        "code": "bright_value_1",
        "custom_name": "",
        "dp_id": 142,
        "time": 1741905956025,
        "type": "value",
        "value": 440
      },
      {
        "code": "brightness_min_1",
        "custom_name": "",
        "dp_id": 143,
        "time": 1741905566200,
        "type": "value",
        "value": 10
      },
      {
        "code": "brightness_max_1",
        "custom_name": "",
        "dp_id": 144,
        "time": 1741905566253,
        "type": "value",
        "value": 1000
      },
      {
        "code": "countdown_led_1",
        "custom_name": "",
        "dp_id": 145,
        "time": 1741905566112,
        "type": "value",
        "value": 0
      },
      {
        "code": "relay_status_led_1",
        "custom_name": "",
        "dp_id": 146,
        "time": 1741905566362,
        "type": "enum",
        "value": "memory"
      }
    ]
  },
  "success": true,
  "t": 1741906102390,
  "tid": "441dc63d005d11f08d3cfab3a7102e50"
}`

Koenkk, 2025-03-16:
> and DP ID 'Dimmer' is what turns the switch

I don't see this DP in your post?

NoahColly, 2025-03-19:
Sorry my bad, I was copying details from the logs and that's what it had noted.
There was an issue with the Tuya app which now seems to be fixed but there are no more DP IDs showing than what I shared above.

Below is a screenshot of the logs when I turn on, adjust the brightness, then turn off.

[image]

[image]

NoahColly, 2025-03-21:
Let me know what I am doing wrong here...I have done the below:
1. Created new js file with the ext converter content you have referenced (File location /homeassistant/zigbee2mqtt/external_converters/tuya_smart_dimmer.js)
3. Restarted Z2M

From here the device in Z2M still shows as unsupported and in the logs I see the below:
```
error: 	z2m: Failed to call 'ExternalConverters' 'start' (node:internal/modules/cjs/loader:645
      throw e;
      ^
Error: Cannot find module '/app/node_modules/zigbee-herdsman-converters/dist/lib/extend.js'
    at createEsmNotFoundErr (node:internal/modules/cjs/loader:1281:15)
    at finalizeEsmResolution (node:internal/modules/cjs/loader:1269:15)
    at resolveExports (node:internal/modules/cjs/loader:638:14)
    at Function._findPath (node:internal/modules/cjs/loader:743:31)
    at Function._resolveFilename (node:internal/modules/cjs/loader:1230:27)
    at Function._load (node:internal/modules/cjs/loader:1070:27)
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:217:24)
    at Module.require (node:internal/modules/cjs/loader:1335:12)
    at require (node:internal/modules/helpers:136:16))
```

For reference I am using Zigbee2MQTT version 2.1.3-1

With this being the full log on startup:
```
[14:28:03] INFO: Preparing to start...
[14:28:03] INFO: Socat not enabled
[14:28:03] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-21 14:28:04] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-21 14:28:04] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
[2025-03-21 14:28:04] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-21 14:28:04] info: 	zh:ember: Using default stack config.
[2025-03-21 14:28:04] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-03-21 14:28:04] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-03-21 14:28:04] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-03-21 14:28:04] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-03-21 14:28:04] info: 	zh:ember:uart:ash: Serial port opened
[2025-03-21 14:28:04] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-03-21 14:28:05] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-03-21 14:28:05] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-03-21 14:28:05] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-03-21 14:28:05] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2025-03-21 14:28:05] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2025-03-21 14:28:05] info: 	zh:ember: [STACK STATUS] Network up.
[2025-03-21 14:28:05] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2025-03-21 14:28:05] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2025-03-21 14:28:05] info: 	z2m: zigbee-herdsman started (resumed)
[2025-03-21 14:28:05] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":4,"revision":"7.4.4 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2025-03-21 14:28:05] info: 	z2m: Hallway 3 Gang (0xa4c138cca15aaeaf): TS0003 - Tuya 3 gang switch (Router)
[2025-03-21 14:28:05] info: 	z2m: Butlers 1 Gang (0xa4c138dd67218e26): TS0001 - Tuya 1 gang switch (Router)
[2025-03-21 14:28:05] info: 	z2m: Garage 2 Gang (0xa4c1386779d255be): TS0002 - Tuya 2-Gang switch with backlight, countdown and inching (Router)
[2025-03-21 14:28:05] info: 	z2m: Master 4 Gang (0xa4c1384f3743af70): TS0004 - Tuya Smart light switch - 4 gang with neutral wire (Router)
[2025-03-21 14:28:05] info: 	z2m: Library 2 Gang (0xa4c138df4cd2f00f): TS0002 - Tuya 2-Gang switch with backlight, countdown and inching (Router)
[2025-03-21 14:28:05] info: 	z2m: Living 2 Gang (0xa4c1384c78598f58): TS0002 - Tuya 2-Gang switch with backlight, countdown and inching (Router)
[2025-03-21 14:28:05] info: 	z2m: Ensuite 4 Gang (0xa4c138f651788808): TS0004 - Tuya Smart light switch - 4 gang with neutral wire (Router)
[2025-03-21 14:28:05] info: 	z2m: WIR 1 Gang (0xa4c138328d174d9d): TS0001 - Tuya 1 gang switch (Router)
[2025-03-21 14:28:05] info: 	z2m: Office 2 Gang (0xa4c13819087efd6c): TS0002 - Tuya 2-Gang switch with backlight, countdown and inching (Router)
[2025-03-21 14:28:05] info: 	z2m: Alfresco 2 Gang (0xa4c13898a2423a22): TS0002 - Tuya 2-Gang switch with backlight, countdown and inching (Router)
[2025-03-21 14:28:05] info: 	z2m: Entry 3 Gang (0xa4c1388c3abe6f59): TS0003 - Tuya 3 gang switch (Router)
[2025-03-21 14:28:05] info: 	z2m: Laundry 2 Gang (0xa4c13883c3e514b8): TS0002 - Tuya 2-Gang switch with backlight, countdown and inching (Router)
[2025-03-21 14:28:05] info: 	z2m: Den 1 Gang (0xa4c13890acaf9758): TS0001 - Tuya 1 gang switch (Router)
[2025-03-21 14:28:05] info: 	z2m: Spare 1 Gang (0xa4c13864a397c952): TS0001 - Tuya 1 gang switch (Router)
[2025-03-21 14:28:05] info: 	z2m: Hallway 1 Gang (0xa4c1387096497b0e): TS0001 - Tuya 1 gang switch (Router)
[2025-03-21 14:28:05] info: 	z2m: Main Bath 4 Gang (0xa4c138675e87e8f1): TS0004 - Tuya Smart light switch - 4 gang with neutral wire (Router)
[2025-03-21 14:28:05] info: 	z2m: Powder 1 Gang (0xa4c138cb5c8be896): TS0001 - Tuya 1 gang switch (Router)
[... cut ...]
    at TracingChannel.traceSync (node:diagnostics_channel:322:14)
    at wrapModuleLoad (node:internal/modules/cjs/loader:217:24)
    at Module.require (node:internal/modules/cjs/loader:1335:12)
    at require (node:internal/modules/helpers:136:16))
[2025-03-21 14:28:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/1001/availability', payload '{"state":"online"}'
[2025-03-21 14:28:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/101/availability', payload '{"state":"online"}'
[2025-03-21 14:28:06] info: 	z2m: Started frontend on port 8099
```

Koenkk, 2025-03-21:
Updated https://gist.github.com/Koenkk/c4bf7723166640c38b13291a874b9b9b, can you try again?

NoahColly, 2025-03-22:
Okay perfect, so that one worked...kinda!
State doesn't work (as you mentioned), but neither does brightness (I turned the light on manually then tried setting the brightness).
Below are the logs of the issue.

```
[2025-03-22 15:06:13] info: 	z2m: Successfully configured 'Kitchen Pendant'
[2025-03-22 15:06:27] error: 	z2m: Publish 'set' 'state' to 'Kitchen Pendant' failed: 'Error: No datapoint defined for 'state''
[2025-03-22 15:06:32] error: 	z2m: Publish 'set' 'brightness' to 'Kitchen Pendant' failed: 'Error: No datapoint defined for 'brightness''
```

Koenkk, 2025-03-22:
Updated https://gist.github.com/Koenkk/c4bf7723166640c38b13291a874b9b9b, please try again.

NoahColly, 2025-03-22:
Hey hey, that worked perfect!
I was able to increase/decrease the brightness fully in both directions.

It's obviously just on/off now that doesn't work.

Thank you so much for putting the time and effort in to help get this working!

```
[2025-03-23 09:12:32] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Pendant', payload '{"brightness":0,"linkquality":168,"state":"OFF"}'
[2025-03-23 09:12:34] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Pendant', payload '{"brightness":254,"linkquality":168,"state":"OFF"}'
```

Koenkk, 2025-03-23:
I updated https://gist.github.com/Koenkk/c4bf7723166640c38b13291a874b9b9b, can you check if on/off works now?

NoahColly, 2025-03-23:
You're a wizard!
Works perfectly!

Koenkk, 2025-03-24:
Added some more features, can you check if it works: https://gist.github.com/Koenkk/c4bf7723166640c38b13291a874b9b9b ?

NoahColly, 2025-03-24:
Hey hey, 
I have given this a shot and below are my findings

**Working:**
- State 
- Brightness
- Voltage
- Current
- Power
- Child lock - Will disable touch controls on physical device when turned on
- Backlight mode - Will turn backlight on/off

**Not Working:**
- Min brightness - I set this to 60 (as this is around when my pendant will turn off with this dimmer), but I was still able to set the brightness lower
- Max brightness - As above but opposite

IMO, the Min/Max brightness was a little clunky even in the Tuya app when I tried.
With the help of ChatGPT I wrote the below to manually adjust the min value...would you be able to make something like this that let you configure the value within HA? This was making what was 22%, now 0% , if that makes sense.

```
[142, "brightness", {
                from: (value) => {
                    // Convert from device brightness (0-1000) to Zigbee brightness (0-254)
                    return Math.round((value - 220) * (254 / (1000 - 220)));
                },
                to: (value) => {
                    // Convert from Zigbee brightness (0-254) to device brightness (0-1000)
                    return Math.round((value * (1000 - 220) / 254) + 220);
                }
            }],
```

The only other thing I can think of that was configurable in the app was the backlight colour. These units use an RGB backlight and within the Tuya app we could change the colour (this was set for the whole panel) and then adjust the brightness for when the light was on/off.
I assume the colour would be set with DP 107 but not sure about the button brightness...

Thanks again for all your work in getting this supported, I really appreciate it :)

Koenkk, 2025-03-25:
I'm not sure exactly what the min/max brightness does, the code you posted limits brightness value that you can send to the device but does not set the actual min/max brightness on the device. I propose to ignore this feature for now.

Regarding the color, the value I see is quite strange: `BmQA5gPo`, not sure how to convert that to a color.

I've added initial support for the device.

Could you also submit a picture for the docs? This can be done by:

1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `MODEL.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

NoahColly, 2025-03-25:
Perfect thank you!

Re colour and min/max, all good.
The min/max may only be an issue that I singularly face with my connected device so not an issue.

I have now done the image and should all be submitted: https://github.com/Koenkk/zigbee2mqtt.io/pull/3634
Let me know if there are any issues with that :)
Again, thank you so much!!

closed 2025-06-01 by github-actions[bot]