Koenkk/zigbee2mqtt #26532: [New device support]: Tuya Smart Zigbee/Wifi Smoke Detector

meekm, 2025-02-27:
### Link

https://www.aliexpress.com/item/1005006862684253.html?spm=a2g0o.order_list.order_list_main.5.5aa779d2qZQJ7I&gatewayAdapt=glo2nld

### Database entry

0xa4c13841fa448443

### Zigbee2MQTT version

1.30.2

### Comments

I tried to add Tuya Smart Zigbee Smoke detector, but it gives an error message: 
Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_rccxox8p'

[log.txt](https://github.com/user-attachments/files/19017776/log.txt)

### External definition

```shell
I tried to generate an External definition, but it did not give any result.
```

### What does/doesn't work with the external definition?

no response

AlexanderLeporiday, 2025-02-28:
It appears this device is already support in a newer version of Zigbee2MQTT.
Support was added to dev branch around Oct 6, 2023, [v15.89.0](https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v15.89.0) of the converter.

Update Zigbee2MQTT to get better device support.

meekm, 2025-03-02:
Thanks a lot, I pulled the Zigbee2MQTT latest version V2.2.1. The Tuya Smart Zigbee Smoke Detector is recognized and works perfectly.
Only all Zigbee devices in Home Assistant didn't work anymore. By upgrading to the newest HA version, this problem also solved.

closed 2025-03-02 by meekm