Koenkk/zigbee2mqtt #26539: Smartwings Blinds State inverted.

DigitalCharon, 2025-02-28:
### What happened?

I have some zigbee blinds reporting as Smartwings WM25L-Z. They report as closed when open and open when closed. There is a device specific setting to invert cover but it doesn't do what the name implies. It instead inverts the reported position so a position of 99% becomes a position of 1% but the open/close state of the blinds remains the opposite of the actual state.

### What did you expect to happen?

I expect the blinds to read as closed when they are closed and open when they are open. I expect the setting labelled "Invert cover" to invert the open/close state, not the % posistion of the blinds. 

### How to reproduce it (minimal and precise)

1) buy, install, and pair smartwings blinds to zigbee2mqtt
2) close the blinds. Note that they read as open
3) open the blinds. Note that they read as closed.
4) look in device specific settings. Set Invert cover to true.
5) Note that blinds still read as open when closed and closed when open. 

### Zigbee2MQTT version

2.1.1-dev commit: 102857a7fec60e3f65887d92b6fe7a2d81add110

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

docker container

dolce08, 2025-04-20:
I've the same issue with a new smart wings blind. The directional controls are correct but it reports 'open' as 'closed' and vice versa. I've tried the following external which I found on another issue but I get an invalid syntax error. Any help is appreciated.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const utils = require('zigbee-herdsman-converters/lib/utils');

backwards_cover_state = {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = {'open': 'downClose', 'close': 'upOpen', 'stop': 'stop', 'on': 'downClose', 'off': 'upOpen'};
            value = value.toLowerCase();
            utils.validateValue(value, Object.keys(lookup));
            await entity.command('closuresWindowCovering', lookup[value], {}, utils.getOptions(meta.mapped, entity));
        },
    };

module.exports = [
    {
        zigbeeModel: ['WM25/L-Z'],
        model: 'WM25L-Z',
        vendor: 'Smartwings',
        description: 'Roller shade',
        fromZigbee: [fz.cover_position_tilt, fz.battery],
        toZigbee: [backwards_cover_state, tz.cover_position_tilt],
        meta: {battery: {dontDividePercentage: true}, coverInverted: `true},`
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.currentPositionLiftPercentage(endpoint);
        },
        exposes: [e.cover_position(), e.battery()],
    }
];
```

Koenkk, 2025-04-21:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/smartwings.ts)?

dolce08, 2025-04-22:
I've made a PR [here](https://github.com/dolce08/zigbee-herdsman-converters/blob/master/src/devices/smartwings.ts)  and looked through the previous changes to the smart wings converter which were made earlier this year. Using the new .mjs extension as per the new document but the file still gets changed to 'invalid' when I restart Z2M and nothing in logs. Here's the updated code. There must be a syntax error I'm missing.
```
import * as fz from "../converters/fromZigbee";
import * as tz from "../converters/toZigbee";
import * as exposes from "../lib/exposes";
import * as reporting from "../lib/reporting";
import {assertString, getFromLookup, getOptions} from "../lib/utils";
import type {DefinitionWithExtend, Tz, Zh} from "../lib/types";

const tzLocal = {
    backwards_cover_state: {
        key: ['state'],
        convertSet: async (entity: Zh.Endpoint, key: string, value: number | string, meta: Tz.Meta) => {
            const lookup = {open: 'downClose', close: 'upOpen', stop: 'stop', on: 'downClose', off: 'upOpen'};
            assertString(value, key);
            value = value.toLowerCase();
            await entity.command('closuresWindowCovering', getFromLookup(value, lookup), {}, getOptions(meta.mapped, entity));
        },
    },
};

const e = exposes.presets;

export const definitions: DefinitionWithExtend[] = [
    {
        zigbeeModel: ["WM25/L-Z"],
        model: "WM25L-Z",
        vendor: "Smartwings",
        description: "Roller shade",
        fromZigbee: [fz.cover_position_tilt, fz.battery],
        toZigbee: [tzLocal.backwards_cover_state, tz.cover_position_tilt],
        meta: {battery: {dontDividePercentage: true}, coverInverted: true},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ["genPowerCfg", "closuresWindowCovering"]);
            device.powerSource = "Mains (single phase)";
            device.save();
        },
        exposes: [e.cover_position(), e.battery()],
    },
];
```

dolce08, 2025-04-23:
In the log I found the error:

z2m: Error: Cannot find module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/dist/converters/fromZigbee.js.js' imported from /opt/zigbee2mqtt/dist/external_converters/>
    at finalizeResolution (node:internal/modules/esm/resolve:283:11)
    at moduleResolve (node:internal/modules/esm/resolve:952:10)
    at defaultResolve (node:internal/modules/esm/resolve:1188:11)
    at ModuleLoader.defaultResolve (node:internal/modules/esm/loader:642:12)
    at ModuleLoader.#cachedDefaultResolve (node:internal/modules/esm/loader:591:25)
    at ModuleLoader.resolve (node:internal/modules/esm/loader:574:38)
    at ModuleLoader.getModuleJobForImport (node:internal/modules/esm/loader:236:38)
    at ModuleJob._link (node:internal/modules/esm/module_job:130:49)

dolce08, 2025-04-24:
The external converter [here](https://github.com/dolce08/zigbee-herdsman-converters/blob/master/src/devices/smartwings.ts) in .mjs works for me. Both the state and directional controls are correct. @DigitalCharon does this work for you?

@Koenkk As there appears to be changes in the new firmware which mean the current converter doesn't work for new devices, can you think of a way to handle this long term?

Koenkk, 2025-04-26:
We can make a separate definition by adding a `fingerprint` with `softwareBuildId` https://github.com/Koenkk/zigbee-herdsman-converters/blob/5dbeb018fdf5744bc91ffad2393cd9a290737dcf/src/devices/sunricher.ts#L856

itssomthing, 2025-08-29:
Still an open issue for me as well

LordSushiPhoenix, 2025-10-29:
Still seems to be an issue

Koenkk, 2025-10-30:
Could you provide your data/database.db entry of this device?

Koenkk, 2025-11-18:
Alternatively you can add the ext converters through the frontend:

[image]

DigitalCharon, 2025-12-24:
I added the external converter linked above through the frontend, restarted, and see Roller Shades CUSTOM under the model. 

With the external converter and the motors in the default direction:
 - Open commands close the blinds. Close commands open the blinds (
 - The state reports correctly. 
 - Home Assistant greys out up/down buttons based on state. Since the controls are reversed, closed blinds can't be opened with the close button. 
 - 100% shows as closed in the home assistant UI as it should.

If I reverse the motor direction as described above, 
- the remote works in reverse. up arrow lowers the blinds, down arrow raises them.
- the % position is correct in the zigbee2mqtt UI
- the state is still reporting closed when opened and opened when closed
- open/close commands through the zigbee2mqtt ui work
- home assistant greys out the up/down buttons according to state. Since the state is reversed, this results in the up button being greyed out when the blinds are closed and the down button being greyed out when the blinds are raised. Automation commands work though. 
-  the home assistant UI element shows 100% as open but with the external converter fixing the %, that's incorrect. So home assistant shows the inverse of the position. 

If I remove the external converter, with the motors reversed:
- the remote's still reversed (just noting again because I would prefer a solution that allows the remote to work correctly)
- open/close commands work correctly through the zigbee2mqtt ui
- the state reports correctly
- position in the zigbee2mqtt ui is inverted from the physical blinds (0 is closed, 100 is open)
- home assistant controls mostly work correctly. Open/Close works and the correct button is available according to the blind's state. The only quirk is that while the UI element shows the blinds down, it reads that is 0% instead of 100%. 

I had to reload the mqtt integration in home assistant for these issues to become apparent in some cases. There were times when testing that I thought it worked correctly before doing that. 

So yeah, the external converter fixes some aspects of the problem but creates more issues.

LordSushiPhoenix, 2025-12-29:
Is that not the normal behaviour for cover percentages? With this converter and the zigbee directions reversed mine also report 100% as fully open and 0% as fully closed. I have bluetooth switchbot curtain 3s that also report 100% as fully open and 0% as fully closed.

ThrashinVictim, 2025-12-29:
Quick Question, if I change the ext converter will it affect all smartwing shades? I have 6 and only 1 has an issue.

chasebolt, 2026-01-01:
Running into this issue as well with my setup. Thought setting the invert would fix it, but found out it doesnt invert the state.

guillochon, 2026-01-22:
> The external converter [here](https://github.com/dolce08/zigbee-herdsman-converters/blob/master/src/devices/smartwings.ts) in .mjs works for me. Both the state and directional controls are correct. [...]

This worked for me, I added the external converter in the dev console in the Z2M webui as @Koenkk had suggested. Needed to restart Z2M after the change, but now the open/close states are correct.

Great-Chart, 2026-08-27:
I suspect that this will get resolved through the revision for cover open / closed inversion in the imminent release
https://github.com/Koenkk/zigbee-herdsman-converters/pull/12707

LordSushiPhoenix, 2026-08-27:
It looks like that should fix it, I'll test it once available.

LordSushiPhoenix, 2026-09-01:
It looks like https://github.com/Koenkk/zigbee-herdsman-converters/pull/12707 has fixed the behaviour of my shades when enabling the invert cover setting.

LordSushiPhoenix, 2026-09-04:
As of 2.14.1 I had to invert the control direction using the smartwings remote (press and hold up+down for 5 sec, then press the P button behind the remote cover) but with that and invert cover set to `true` my shades are working.

drizztguen77, 2026-09-05:
I also had to do the invert with the remote (press and hold up+down for 5 sec, then press the P button behind the remote cover) on mine on version 2.14.1-1 and it works fine.

LordSushiPhoenix, 2026-10-04:
2.14.2 changed the behaviour of the open/close commands and I had to revert the control direction. https://github.com/Koenkk/zigbee-herdsman-converters/commit/e9437ee9efd9ef16bb017810003d723ef135059a

drizztguen77, 2026-10-04:
> 2.14.2 changed the behaviour of the open/close commands and I had to revert the control direction. [Koenkk/zigbee-herdsman-converters@e9437ee](https://github.com/Koenkk/zigbee-herdsman-converters/commit/e9437ee9efd9ef16bb017810003d723ef135059a)

I really wish you people would stop this. It broke all my shades automations again so now I get to spend a day fixing everything again after the last change where I had to fix them the other way. Can we change terminology or define somewhere what "open" and "closed" means exactly because nobody seems to be able to agree.

drizztguen77, 2026-10-04:
This will likely break things again for many people but I found with mine I was just able use the remote and change it back to the way it was before and I didn't have to change all my automations again. Here is what you need to do if your blinds break again. You need to do this for each blind you have.

Using the remote, press and hold up+down buttons for 5 sec, then press the P button on the back under the remote cover