Koenkk/zigbee2mqtt #26541: Raspberry Pi 3 model B hangs immediately after z2m starts (Sonoff ZBDongle-E)

defanator, 2025-03-01:
### What happened?

Zigbee2MQTT start immediately causes Raspberry Pi 3 to hang with the following messages in system log (kernel 6.12.16-v8+):

```
Mar 01 11:02:55 rpi3-mon kernel: WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 6
```

or these (kernel 6.6.74+rpt-rpi-v8):

```
Mar 01 10:24:37 rpi3-mon kernel: ------------[ cut here ]------------
Mar 01 10:24:37 rpi3-mon kernel: Firmware transaction timeout
Mar 01 10:24:37 rpi3-mon kernel: WARNING: CPU: 2 PID: 1048 at drivers/firmware/raspberrypi.c:69 rpi_firmware_property_list+0x264/0x288
```

Dongle used - SONOFF Zigbee 3.0 USB Dongle Plus V2:
```
# lsusb
Bus 001 Device 004: ID 1a86:55d4 QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2
```

### What did you expect to happen?

z2m is able to communicate with the adapter or at least not causing the whole system hang.

### How to reproduce it (minimal and precise)

I was using container image with zigbee2mqtt 1.39.0 and 1.42.0. Previously, 1.39.0 worked just fine for a several months on the same system.

### Zigbee2MQTT version

1.39.0, 1.42.0

### Adapter firmware version

7.4.3 [GA]

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Setup

Plain, container on 64-bit Debian 12 "bookworm" running on Raspberry Pi 3 model B

### Debug log

No chance to collect any logs from z2m - everything hangs after the initial message:
```
Mar 01 10:24:21 rpi3-mon z2m[1508]: Starting Zigbee2MQTT without watchdog.
```

Kernel logs - 6.12.16-v8+:
```
Mar 01 11:02:40 rpi3-mon z2m[1583]: Starting Zigbee2MQTT without watchdog.
Mar 01 11:02:55 rpi3-mon kernel: 
Mar 01 11:02:55 rpi3-mon kernel: WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 6
Mar 01 11:02:55 rpi3-mon kernel: ------------[ cut here ]------------
Mar 01 11:02:55 rpi3-mon kernel: Firmware transaction 0x00030046 timeout
```

Kernel logs - 6.6.74+rpt-rpi-v8:
```
Mar 01 10:24:21 rpi3-mon z2m[1508]: Starting Zigbee2MQTT without watchdog.
Mar 01 10:24:37 rpi3-mon kernel: ------------[ cut here ]------------
Mar 01 10:24:37 rpi3-mon kernel: Firmware transaction timeout
Mar 01 10:24:37 rpi3-mon kernel: WARNING: CPU: 2 PID: 1048 at drivers/firmware/raspberrypi.c:69 rpi_firmware_property_list+0x264/0x288
Mar 01 10:24:37 rpi3-mon kernel: Modules linked in: xt_addrtype xt_nat xt_tcpudp xt_mark xt_conntrack xt_comment nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables veth bridge stp llc nfnetlink overlay vc4 brcmfmac_wcc snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper binfmt_misc drm_kms_helper brcmfmac snd_soc_core snd_compress brcmutil snd_pcm_dmaengine hci_uart cfg80211 btbcm bcm2835_codec(C) bluetooth bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem cdc_acm bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc raspberrypi_hwmon videobuf2_memops videobuf2_v4l2 ecdh_generic ecc videodev rfkill libaes snd_bcm2835(C) videobuf2_common mc snd_pcm snd_timer snd vc_sm_cma(C) raspberrypi_gpiomem uio_pdrv_genirq uio drm fuse drm_panel_orientation_quirks dm_mod backlight ip_tables x_tables ipv6 i2c_bcm2835
Mar 01 10:24:37 rpi3-mon kernel: CPU: 2 PID: 1048 Comm: kworker/2:1 Tainted: G         C         6.6.74+rpt-rpi-v8 #1  Debian 1:6.6.74-1+rpt1
Mar 01 10:24:37 rpi3-mon kernel: Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
Mar 01 10:24:37 rpi3-mon kernel: Workqueue: events get_values_poll [raspberrypi_hwmon]
Mar 01 10:24:37 rpi3-mon kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Mar 01 10:24:37 rpi3-mon kernel: pc : rpi_firmware_property_list+0x264/0x288
Mar 01 10:24:37 rpi3-mon kernel: lr : rpi_firmware_property_list+0x264/0x288
Mar 01 10:24:37 rpi3-mon kernel: sp : ffffffc08122bca0
Mar 01 10:24:37 rpi3-mon kernel: x29: ffffffc08122bca0 x28: 0000000000000000 x27: ffffff8001fc1c40
Mar 01 10:24:37 rpi3-mon kernel: x26: 0000000000000004 x25: ffffffc080498008 x24: 0000000000001000
Mar 01 10:24:37 rpi3-mon kernel: x23: ffffff800979c5c0 x22: ffffffd2fe7ec2e0 x21: 0000000000000010
Mar 01 10:24:37 rpi3-mon kernel: x20: ffffff8001fc1c00 x19: ffffffc080498000 x18: 0000000000000000
Mar 01 10:24:37 rpi3-mon kernel: x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
Mar 01 10:24:37 rpi3-mon kernel: x14: 0000000000000000 x13: 74756f656d697420 x12: 6e6f69746361736e
Mar 01 10:24:37 rpi3-mon kernel: x11: fffffffffffe0000 x10: ffffffd2fe6c64b0 x9 : ffffffd2fd10bdd4
Mar 01 10:24:37 rpi3-mon kernel: x8 : 00000000ffffefff x7 : ffffffd2fe6c3a38 x6 : 00000000000001c5
Mar 01 10:24:37 rpi3-mon kernel: x5 : ffffff803b1a5d88 x4 : 0000000000000000 x3 : 0000000000000027
Mar 01 10:24:37 rpi3-mon kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff8003941ec0
Mar 01 10:24:37 rpi3-mon kernel: Call trace:
[..]
```

Might be relevant (?) - https://github.com/raspberrypi/firmware/issues/1804.

defanator, 2025-03-02:
> Did you try unplugging the adapter, and starting Z2M to see if the error remains the same (if it goes far enough to tell you the adapter is unplugged)?

Unfortunately I do not have physical access to the device at the moment as it is a remote location. There is a wifi socket through which I can do power resets in case if the hardware watchdog did not catch an issue.

> Did you try an external powered USB hub?

I did not. I manage a few identical setups with RPi 3 model B, z2m, and identical Sonoff dongles (ZBDongle-E) across different geo locations, and it turned out that if there are no other USB devices plugged in, RPi 3 with decent power supply is able to provide enough power to the dongle for stable operation.

I'm currently trying to schedule a replacement of a power supply or the whole setup to bring newer Pi to the problematic location - thinking of RPi 4 or even 5, as 1Gb RAM already started to be an issue for the environment I'm running on those Pi's.

But, while this is still under consideration, I tried to flash latest firmware (https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl) directly from that RPi (https://github.com/defanator/zigbee-tools/tree/main/sonoff-zbdongle-firmware-upgrading), and, to my surprise, z2m was able to start. As of now, it's running for more than 11 hours without a single issue.

Still monitoring things closely.

If there is anything I could grab from a working system to try to figure out why it was struggling with the 7.4.3 fw, I'm eager to try and share the findings with the community.

(I'm not excluding a possibility of hardware issue, but as I said, I have quite a limited way to confirm or deny it at this very moment.)

Nerivec, 2025-03-02:
Could be flashing the firmware was enough to bring back the adapter into a better state. Continue to monitor closely to see if it starts re-appearing after a while.
I don't know that flasher you used, so, make sure it has the proper logic too. And make sure it properly closes the serial after and exits cleanly, else Z2M will definitely fail to start.
https://www.zigbee2mqtt.io/guide/adapters/emberznet.html#firmware-flashing

Note: newest firmware are now in `Release` section for most builders (which should be 8.0.2, most common ones also in above link).

defanator, 2025-03-02:
I was using `minicom` to upload firmware directly after switching the dongle to bootloader mode (https://github.com/defanator/zigbee-tools/tree/main/sonoff-zbdongle-firmware-upgrading).

I saw a few warning around 8.0.x firmwares, so decided to stick with 7.4.x for now (in fact, 7.4.3 is running just fine for many months in many of my identical setups).

Is there a link to 8.0.2? I used to grab binaries from https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee, but there is no 8.x yet.

Nerivec, 2025-03-02:
Links & details are on the page I linked (there are a few links to flashers too):
https://github.com/darkxst/silabs-firmware-builder/releases
https://github.com/Nerivec/silabs-firmware-builder/releases

7.4.5 or 8.0.2 should be pretty similar. But 8.0.2 have a few fixes from user feedback that previous won't have.

amanteaux, 2025-04-16:
I have a similar issue, although I can't access the kernel logs: they seem to not be written fast enough before the freeze.
I tried many things:
- Update the dongle to the latest version: `7.4.5.0`
- Update the Raspbian OS to the latest version: `6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux`
- Update zigbee2mqtt to the 2.2.1 tag

If I unplug the dongle before the boot and then I plug it after the boot, then Z2M starts correctly. However, if the dongle is plugged at boot time, even if I wait a couple of minutes before starting Z2M, then the freeze occurs immediately.

I suspected an issue with the driver cdc_acm, so I blocked it and then manually started it using `sudo modprobe cdc_acm` to make the dongle available at `/dev/ttyACM0`, but starting Z2M lead to the freeze.

Finally I used picocom to see if I could reproduce the issue without Z2M with `sudo picocom /dev/ttyACM0 --baud 115200 --flow hw --echo`, but everything works fine with that. Although when I launch this command, after running Z2M works perfectly :O

Here are the last logs I see when I run Z2M and the system freezes:
```
:/opt/zigbee2mqtt $ pnpm start

> zigbee2mqtt@2.2.1 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.

[2025-04-16 22:00:55] info:     z2m: Logging to console, file (filename: log.log)
[2025-04-16 22:00:56] info:     z2m: Starting Zigbee2MQTT version 2.2.1 (commit #a14feef8)
[2025-04-16 22:00:56] info:     z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-16 22:00:56] info:     zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20231122134335","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231122134335-if00","vendorId":"1a86","productId":"55d4"} => ember: 4
[2025-04-16 22:00:56] info:     zh:ember: Using default stack config.
[2025-04-16 22:00:56] info:     zh:ember: ======== Ember Adapter Starting ========
[2025-04-16 22:00:56] info:     zh:ember:ezsp: ======== EZSP starting ========
[2025-04-16 22:00:56] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-04-16 22:00:56] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
```

Do you have any idea how to make Z2M starts correctly on its own after boot without freezing the system?

Nerivec, 2025-04-16:
Looks like something might be blocking the port during boot? Something that somehow gets knocked out when using the picocom cmd? 👀
Can you check the system logs when you use that command, see if anything is happening with the hardware (might be forcing the usb to reboot or something...)?

amanteaux, 2025-04-17:
Thank you @Nerivec for your help!

When I run this command (and actually when I quit it using ctrl+a, ctrl+x), I can see this in the logs:
```
WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 1

WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 2

WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 5

WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 4
```

We lead to the related issue https://github.com/raspberrypi/firmware/issues/1804

amanteaux, 2025-04-17:
At least now I can have a working setup by adding this line in `/etc/systemd/system/zigbee2mqtt.service`: `ExecStartPre=picocom /dev/ttyACM0 --baud 115200 --flow hw -qrx 1000`

defanator, 2025-04-17:
TWIMC, I was able to run a few onsite tests with a couple of different RPi3 model Bs. Here's what I caught several times during those tests - every time the trigger for this panic was starting zigbee2mqtt:

```
Mar 29 15:54:12 rpi3-mon kernel: ------------[ cut here ]------------
Mar 29 15:54:12 rpi3-mon kernel: Firmware transaction 0x00030046 timeout
Mar 29 15:54:12 rpi3-mon kernel: WARNING: CPU: 2 PID: 30 at drivers/firmware/raspberrypi.c:132 rpi_firmware_property_list+0x220/0x288
Mar 29 15:54:12 rpi3-mon kernel: Modules linked in: xt_addrtype xt_nat xt_tcpudp xt_mark xt_conntrack xt_comment nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables veth bridge stp llc nfnetlink overlay vc4 binfmt_misc brcmfmac_wcc snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper drm_kms_helper brcmfmac snd_soc_core snd_compress snd_pcm_dmaengine cdc_acm brcmutil raspberrypi_hwmon cfg80211 hci_uart btbcm bcm2835_codec(C) bcm2835_v4l2(C) bluetooth bcm2835_isp(C) bcm2835_mmal_vchiq(C) vc_sm_cma(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videodev ecdh_generic ecc rfkill i2c_bcm2835 videobuf2_common libaes snd_bcm2835(C) mc snd_pcm snd_timer snd raspberrypi_gpiomem uio_pdrv_genirq uio drm fuse drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6
Mar 29 15:54:12 rpi3-mon kernel: CPU: 2 UID: 0 PID: 30 Comm: kworker/2:0 Tainted: G     C   6.12.16-v8+ #1859
Mar 29 15:54:12 rpi3-mon kernel: Tainted: [C]=CRAP
Mar 29 15:54:12 rpi3-mon kernel: Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
Mar 29 15:54:12 rpi3-mon kernel: Workqueue: events get_values_poll [raspberrypi_hwmon]
Mar 29 15:54:12 rpi3-mon kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=—-)
Mar 29 15:54:12 rpi3-mon kernel: pc: rpi_firmware_property_list+0x220/0x288
Mar 29 15:54:12 rpi3-mon kernel: lr: rpi_firmware_property_list+0x220/0x288
Mar 29 15:54:12 rpi3-mon kernel: sp: ffffffc080123ca0
Mar 29 15:54:12 rpi3-mon kernel: x29: ffffffc080123ca0 x28: ffffffd3987669c0 x27: ffffffa682b1e5c0
Mar 29 15:54:12 rpi3-mon kernel: x26: 00000000ffffff92 x25: ffffffc08069f008 x24: 0000000000001000
Mar 29 15:54:12 rpi3-mon kernel: x23: ffffffa686870840 x22: ffffffd39890f078 x21: 0000000000000010
Mar 29 15:54:12 rpi3-mon kernel: x20: ffffffa682b1e580 x19: ffffffc08069f000 x18: ffffffffffffffff
Mar 29 15:54:12 rpi3-mon kernel: x17: 0000000000000000 x16: 0000000000000000 x15: ffffffd3989a8410
Mar 29 15:54:12 rpi3-mon kernel: x14: 0000000000000008 x13: ffffffd3989a8402 x12: 61736e6172742065
Mar 29 15:54:12 rpi3-mon kernel: x11: fffffffffffe0000 x10: ffffffd3987e7dc8 x9 : ffffffd39711bf28
Mar 29 15:54:12 rpi3-mon kernel: x8 : 00000000ffffefff x7 : ffffffd3987e5380 x6 : 0000000000002a48
Mar 29 15:54:12 rpi3-mon kernel: x5 : ffffffa6bb1a93c8 x4 : 0000000000000000 x3 : 0000000000000027
Mar 29 15:54:12 rpi3-mon kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffffa68231c200
Mar 29 15:54:12 rpi3-mon kernel: Call trace:
Mar 29 15:54:12 rpi3-mon kernel:  rpi firmware_property_list+0x220/0x288
Mar 29 15:54:12 rpi3-mon kernel:  rpi_firmware_property+0x78/0x118
Mar 29 15:54:12 rp13-mon kernel:  get_values_poll+0x54/0x138 [raspberrypi_hwmon]
Mar 29 15:54:12 rpi3-mon kernel:  process_one_work+0x15c/0x3c0
Mar 29 15:54:12 rpi3-mon kernel:  worker_thread+0x2e4/0x3f0
Mar 29 15:54:12 rpi3-mon kernel:  kthread+0x120/0x130
Mar 29 15:54:12 rpi3-mon kernel:  ret_from_fork+0x10/0x20
Mar 29 15:54:12 rpi3-mon kernel: ---[ end trace 0000000000000000 ]---
Mar 29 15:54:12 rpi3-mon kernel: hwmon hwmon1: Failed to get throttled (-110)
```

(see also - https://github.com/raspberrypi/firmware/issues/1804#issuecomment-2811678049)

Nerivec, 2025-04-17:
Just to be sure, that's always on OS version 11 or above, right? `cat /etc/issue.net`
Previous ones are having weird troubles with nodejs.

amanteaux, 2025-04-17:
On my side yes, the Raspbian OS is fully up to date (as of yesterday):
```
$ cat /etc/issue.net
Debian GNU/Linux 12
```

Nerivec, 2025-04-17:
Since you're using an EmberZNet adapter, can you try running the `stack` command of https://github.com/Nerivec/ember-zli/wiki just to confirm the problem is also triggered in the same way (re pre-boot vs post-boot plug)?

defanator, 2025-04-17:
> Just to be sure, that's always on OS version 11 or above, right?

Yes. In my case it was latest Debian 12 with both 6.6.74+rpt-rpi-v8 and 6.12.16-v8+ kernels.

amanteaux, 2025-04-18:
I confirm that the problem is also triggered the same way when I run `ember-zli stack` here are the logs that I can see before the crash:
```
$ ember-zli stack

[2025-04-18 08:39:10.491] info:         cli: Data folder: /home/zigbee/ember-zli.
✔ Path: /dev/ttyACM0, Baudrate: 115200, RTS/CTS: false. Use this config? Yes
[2025-04-18 08:39:11.033] info:         zh:ember:ezsp: ======== EZSP starting ========
[2025-04-18 08:39:11.037] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-04-18 08:39:11.042] info:         zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
```

Nerivec, 2025-04-18:
It appears to be as soon as something is sent to the serial (hitting the USB).

Can you try this in your Z2M install folder, see if it makes any difference?
```bash
rm -rf `find ./node_modules/.pnpm/ -wholename "*/@serialport/bindings-cpp/prebuilds" -type d`
pnpm rebuild @serialport/bindings-cpp
```
Make sure to keep the logs and post them after (if it doesn't work 😅).

PS: you can revert this with: `rm -rf ./node_modules && pnpm i --frozen-lockfile`

amanteaux, 2025-04-18:
Yes I can try this!

For information, I tried a fix that worked well from https://github.com/raspberrypi/firmware/issues/1804 : to put `dtoverlay=dwc2` in the config.txt file. So I am wondering if the bug is actually in the default `dwc_otg` USB driver

amanteaux, 2025-04-18:
It didn't work, but I have more logs:
```
/opt/zigbee2mqtt $ pnpm start

> zigbee2mqtt@2.2.1 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-04-18 21:09:55] debug:    z2m: Removing old log directory '/opt/zigbee2mqtt/data/log/2025-04-17.07-01-02'
[2025-04-18 21:09:55] info:     z2m: Logging to console, file (filename: log.log)
[2025-04-18 21:09:56] debug:    z2m: Loaded state from file /opt/zigbee2mqtt/data/state.json
[2025-04-18 21:09:56] info:     z2m: Starting Zigbee2MQTT version 2.2.1 (commit #a14feef8)
[2025-04-18 21:09:56] info:     z2m: Starting zigbee-herdsman (3.4.11)
[2025-04-18 21:09:56] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":51916,\"extendedPanID\":[24,69,197,255,71,100,47,124],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/opt/zigbee2mqtt/data/database.db\",\"databaseBackupPath\":\"/opt/zigbee2mqtt/data/database.db.backup\",\"backupPath\":\"/opt/zigbee2mqtt/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2025-04-18 21:09:56] debug:    zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20231122134335","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231122134335-if00","vendorId":"1a86","productId":"55d4"}]
[2025-04-18 21:09:56] info:     zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20231122134335","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20231122134335-if00","vendorId":"1a86","productId":"55d4"} => ember: 4
[2025-04-18 21:09:57] info:     zh:ember: Using default stack config.
[2025-04-18 21:09:57] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":51916,"extendedPanID":[24,69,197,255,71,100,47,124],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"ember"},"adapter":{"disableLED":false},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}'
[2025-04-18 21:09:57] info:     zh:ember: ======== Ember Adapter Starting ========
[2025-04-18 21:09:57] info:     zh:ember:ezsp: ======== EZSP starting ========
[2025-04-18 21:09:57] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-04-18 21:09:57] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-04-18 21:09:57] debug:    zh:ember:uart:ash: Opening serial port with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
```

Nerivec, 2025-04-18:
If it's a problem with the USB driver, I don't think there's much that Z2M will be able to do unfortunately...

amanteaux, 2025-04-18:
Yes for sure! I'll keep this thread updated if it's confirmed that the root issue is the USB driver.

Thank you for your help!

closed 2025-08-13 by github-actions[bot]