Koenkk/zigbee2mqtt #26548: [Wrong device]: Paulmann 500.46 detected as Paulmann 50064

jule-ha, 2025-03-01:
### Link

https://www.zigbee2mqtt.io/devices/500.46.html

### Model

500.46

### Description

Zigbee white controller 24V

### Vendor

Paulmann

### Picture (link)

https://content.paulmann.com/media/28/af/42/1618883046/50046ip01.jpg?ts=1740711235&

### Database entry

{"id":6,"type":"Router","ieeeAddr":"0x00158d00087ac4da","nwkAddr":23311,"manufId":4632,"manufName":"Paulmann Licht GmbH","pow:1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x7cc6b6fffefb3b7b","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x7cc6b6fffefb3b7b","endpointID":1},{"cluster":8,erSource":"DC Source","modelId":"CCT","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0:"0x7cc6b6fffefb3b7b","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":1,"stackVersion":106,",4,3,6,8,5,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"CCT","manufacturerName":"Paulmann49},"lastSeen":1740837428017,"checkinInterval":43200} Licht GmbH","powerSource":4,"zclVersion":3,"appVersion":19,"stackVersion":1,"hwVersion":16,"dateCode":"20190515","swBuildId"rce":"Mains (single phase)","modelId":"TRADFRI bulb E27 WS globe 1055lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64636],"outCl:"1400-0001"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":37f Sweden","powerSource":1,"zclVersion":8,"appVersion":3,"stackVersion":113,"hwVersion":1,"dateCode":"20240716","swBuildId":"3.0.23"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outCluredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3sterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":19,"stackVersion":1,"hwVersion":16{"configured":332242049},"lastSeen":1740852895812},"dateCode":"20190515","swBuildId":"1400-0001","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1740856102135}

### Notes

Hi, i have a 24V Controller Paulmann 500.46 and this one is wrong detected as Paulmann 50064 LED-Light. I have made a new pairing, interview of the device and a new configuration, but nothing helps. Always the wrong Device is detected. 
The Problem is, that the maxcolortemp is only 370 in the wrong device and the right device can until 500. 
I have addition tried  to make an external_converter with the info from the development-consol for generated_external_definition. 
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['CCT'],
    model: 'CCT',
    vendor: 'Paulmann Licht GmbH',
    description: 'Automatically generated definition',
    extend: [m.light({"colorTemp":{"range":[153,370]}})],
    meta: {},
};

module.exports = definition;

I have change the range for colorTemp to 500 and in the Log i was correctly read, but i don't change anything...

jule-ha, 2025-03-01:
Hi unfortunately my database-copy was wrong here the right on:
{"id":6,"type":"Router","ieeeAddr":"0x00158d00087ac4da","nwkAddr":23311,"manufId":4632,"manufName":"Paulmann Licht GmbH","powerSource":"DC Source","modelId":"CCT","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,4,3,6,8,5,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"CCT","manufacturerName":"Paulmann Licht GmbH","powerSource":4,"zclVersion":3,"appVersion":19,"stackVersion":1,"hwVersion":16,"dateCode":"20190515","swBuildId":"1400-0001"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":19,"stackVersion":1,"hwVersion":16,"dateCode":"20190515","swBuildId":"1400-0001","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1740859701488}#

Uritec, 2025-04-27:
Same problem with my device model 500.46.

closed 2025-07-05 by github-actions[bot]