Koenkk/zigbee2mqtt #26553: Invalid supported_color_modes for lights

LouisMT, 2025-03-01:
### What happened?

I'm getting the following error in Home Assistant 2025.3b2:

```
voluptuous.error.Error: Invalid supported_color_modes ['brightness', 'color_temp', 'xy']
```

This causes all the lights to be unavailable in Home Assistant.

---

I'm not entirely sure, but I think according to the [docs](https://developers.home-assistant.io/docs/core/entity/light/) and [code](https://github.com/home-assistant/core/blob/dev/homeassistant/components/light/__init__.py#L125), the light should have either `brightness` or `color_temp` set, not both:

> ColorMode.BRIGHTNESS | The light can be dimmed. This mode must be the only supported mode if supported by the light.

### What did you expect to happen?

My lights to be available after updating to zigbee2mqtt 2.1.2.

### How to reproduce it (minimal and precise)

Update to zigbee2mqtt 2.1.2, with lights that support multiple color modes.

### Zigbee2MQTT version

2.1.2

### Adapter firmware version

20230507

### Adapter

zzh!

### Setup

Docker containers

LouisMT, 2025-03-01:
I think it's caused by #26545.

hmmbob, 2025-03-01:
> I think it's caused by [#26545](https://github.com/Koenkk/zigbee2mqtt/pull/26545).

It is. And you are correct, those modes are exclusive

BaldKingy, 2025-03-01:
Sharp analysis. Just upgraded and none of my color bulbs work in Home Assistant.

There used to be instructions how to downgrade to a specific version, I can't find it anymore. Any hints? (It's Saturday, normaly I backup the install directory, today I didn't :) )

MerekW, 2025-03-01:
Almost a complete contrast, I lost dimming functionality on some Zigbee bulbs/groups, but only in HA (mqtt dimmers are still there)?  It's as if my non-colored Zigbee bulbs lost their brightness feature as far as HA is concerned!

Chartreusito, 2025-03-01:
Thanks for pointing that out. I just replaced a bulb and was going crazy trying to troubleshoot this one. At least my backup binded remotes still works. Then i realized a good part of my lights were affected.

fnordpojk, 2025-03-02:
Yep, all my color/color temperature adjustable Trådfri lights went unavailable with these log errors with 2.1.2. Rolled back to 2.1.1 and everything works fine again. I'm on HA 2025.2.5 but same same.

MerekW, 2025-03-02:
@fnordpojk - Are you running the Z2M Edge HA add-on?

fnordpojk, 2025-03-02:
> [@fnordpojk](https://github.com/fnordpojk) - Are you running the Z2M Edge HA add-on?

No, just the normal one.

lordbah, 2025-03-02:
Seeing this in HA 2025.2.5 and z2m 2.1.2-1.
Too bad there isn't anything on the Exposes tab to just not expose X/Y and color temp, since I don't use them.

Koenkk, 2025-03-02:
Fixed in 2.1.3!

closed 2025-03-02 by Koenkk