Koenkk/zigbee2mqtt #26555: Gledopto GL-C-002P stopped transmitting light entity to HA

mitrokun, 2025-03-02:
### What happened?

After update 2.1.2-1 Gledopto GL-C-002P (5in1) stopped transmitting light entity to HA.
There are two devices, one configured as GL-C-007P and one configured as GL-C-006P. Tried deleting and rejoining - no result. 
In Z2M everything works as it did before the update. 
In HA the device still has all entities (update, identify, effect selector...), except for the light itself.
There is nothing unusual in the mqtt topic, quite standard data.
```
info 2025-03-02 04:30:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138badb767171', payload '{"brightness":29,"effect":null,"identify":null,"last_seen":"2025-03-02T04:30:24+03:00","linkquality":32,"power_on_behavior":"off","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null}}'
info 2025-03-02 04:30:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138badb767171', payload '{"brightness":29,"effect":null,"identify":null,"last_seen":"2025-03-02T04:30:24+03:00","linkquality":32,"power_on_behavior":"off","state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null}}'
```
I also use a regular GL-C-009P - it is not affected by the problem.

### Zigbee2MQTT version

2.1.2-1

### Adapter firmware version

20240710

### Adapter

SONOFF ZB Dongle Plus-P

### Setup

add-on

mluckau, 2025-03-02:
I have the same problem with the Nous P3Z lamp.

rapha1rocha, 2025-03-02:
same here.

device: Zigbee 3.0 18W led light bulb E27 RGBCW
model: TS0505B_1_1 (Tuya)
adapter: Dongle Sonoff-e

It is possible to change the entity "state"  using the Zigbee2MQTT web interface. 
However, the entity cannot be used in Home Assistant (HA) because MQTT lacks the state entity necessary to turn the lamp on or off.

ChxGuillaume, 2025-03-02:
> I have the same problem with the Nous P3Z lamp.
> 
> Edit: I have just noticed that it affects all my light entities.

Yup, 2.1.2 broke compatibility with all my IKEA lights too...
Ngl, those past updates have been quite a nightmare with compatibility changes and stuff not working anymore ^^

genehand, 2025-03-02:
Agreed, I was ok with 2.0 because it was announced well in advanced and very obviously marked as a breaking change. I'm back up on 2.1.1 now.  From the main PR 8471, it seems like this warrants more than a patch release:

>This being a deep change in the architecture, some in-depth testing was required.

Koenkk, 2025-03-02:
Fixed in 2.1.3!

closed 2025-03-02 by Koenkk