Koenkk/zigbee2mqtt #26556: Stuck at startup on Starting zigbee-herdsman

Nastaliss, 2025-03-02:
### What happened?

Hi there, 
I am using zigbee2mqtt with docker compose and after a host restart the z2m container is no longer able to start, without erroring. 
Another issue is that the container does not respond to SIGTERM / SIGKILL signals from docker when shutting down while hanging.

### What did you expect to happen?

I expect the z2m container to start using the coordinator.
And to stop when stopped by docker daemon

### Zigbee2MQTT version

1.22.1 an 1.22.2

### Adapter firmware version

Firmware: 7.4.4-0 (EZSP v13)

### Adapter

Sonoff E dongle using ember driver

### Setup

Docker (compose) on a raspberry pi 3b (HA is running on another server)

### Debug log

Startup logs :
```
 ✔ Container zigbee2mqtt  Created                                                                                                                   3.6s 
 ✔ Container mqtt5        Created                                                                                                                   0.0s 
Attaching to mqtt5, zigbee2mqtt
zigbee2mqtt  | Using '/app/data' as data directory
mqtt5        | 1740878224: mosquitto version 2.0.20 starting
mqtt5        | 1740878224: Config loaded from /mosquitto/config/mosquitto.conf.
mqtt5        | 1740878224: Warning: File /mosquitto/config/pwfile has world readable permissions. Future versions will refuse to load this file.
mqtt5        | To fix this, use `chmod 0700 /mosquitto/config/pwfile`.
mqtt5        | 1740878224: Opening ipv4 listen socket on port 1883.
mqtt5        | 1740878224: Opening ipv6 listen socket on port 1883.
mqtt5        | 1740878224: Opening websockets listen socket on port 9001.
mqtt5        | 1740878224: mosquitto version 2.0.20 running
mqtt5        | 1740878228: New connection from 192.168.1.17:36285 on port 1883.
mqtt5        | 1740878228: New client connected from 192.168.1.17:36285 as 5xGBOk7Oy8ojzghoIW2lhL (p2, c1, k60, u'nasta').
zigbee2mqtt  | Starting Zigbee2MQTT without watchdog.
zigbee2mqtt  | [2025-03-02 02:17:17] info:      z2m: Logging to console, file (filename: log.log)
zigbee2mqtt  | [2025-03-02 02:17:17] debug:     z2m: Loaded state from file /app/data/state.json
zigbee2mqtt  | [2025-03-02 02:17:17] info:      z2m: Starting Zigbee2MQTT version 2.1.2 (commit #d2d7ebeed9bffb9723a7a9fefee8ea6deb27e000)
zigbee2mqtt  | [2025-03-02 02:17:17] debug:     z2m: sd-notify loaded
zigbee2mqtt  | [2025-03-02 02:17:17] info:      z2m: Starting zigbee-herdsman (3.2.7)
zigbee2mqtt  | [2025-03-02 02:17:17] debug:     z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":27413,\"extendedPanID\":[102,184,169,159,220,25,57,43],\"channelList\":[20],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
zigbee2mqtt  | [2025-03-02 02:22:17] debug:     z2m: Saving state to file /app/data/state.json
```

Shutdown logs (on the host): 
```
$ docker kill zigbee2mqtt 
Error response from daemon: cannot kill container: zigbee2mqtt: tried to kill container, but did not receive an exit event
```

Shutdown logs (inside the container) : 
```
[...]
[2025-03-02 02:17:17] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":27413,\"extendedPanID\":[102,184,169,159,220,25,57,43],\"channelList\":[20],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/app/data/database.db\",\"databaseBackupPath\":\"/app/data/database.db.backup\",\"backupPath\":\"/app/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2025-03-02 02:22:17] debug: 	z2m: Saving state to file /app/data/state.json
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0xe406bffffee7d5e9): default expiration timeout set to 0
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0x0ceff6fffe69ff72): default expiration timeout set to 3600000
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0xa46dd4fffe412d73): default expiration timeout set to 0
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0xf0d1b8000015c54a): default expiration timeout set to 0
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0x4c5bb3fffea61c8c): default expiration timeout set to 0
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0xd44867fffeaa0489): default expiration timeout set to 0
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0x286847fffe2e053c): default expiration timeout set to 0
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0x0ceff6fffe5b1798): default expiration timeout set to 3600000
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0x0ceff6fffe9027f7): default expiration timeout set to 3600000
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0x44e2f8fffe1a9969): default expiration timeout set to 3600000
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0x44e2f8fffe045295): default expiration timeout set to 3600000
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0x00158d0006203cbe): default expiration timeout set to 0
[2025-03-02 02:25:48] debug: 	zh:controller:device: Request Queue (0xa4c1384113baa9e0): default expiration timeout set to 0
[2025-03-02 02:25:49] debug: 	zh:controller:device: Request Queue (0x286847fffe2e0871): default expiration timeout set to 0
[2025-03-02 02:25:49] debug: 	zh:controller:device: Request Queue (0x0ceff6fffe59787f): default expiration timeout set to 3600000
[2025-03-02 02:25:49] debug: 	zh:controller:device: Request Queue (0xa46dd4fffe413155): default expiration timeout set to 0
[2025-03-02 02:25:49] debug: 	zh:controller:device: Request Queue (0x94ec32fffe296af9): default expiration timeout set to 0
[2025-03-02 02:25:49] debug: 	zh:controller:device: Request Queue (0x0ceff6fffeee24af): default expiration timeout set to 3600000
[2025-03-02 02:25:49] debug: 	zh:controller:device: Request Queue (0xa4c138ad5dd627ec): default expiration timeout set to 0
[2025-03-02 02:25:49] debug: 	zh:controller:device: Request Queue (0x0ceff6fffe8fd593): default expiration timeout set to 3600000
```
I have tried opening the permissions on the serial device with no success, I also tried probing the coordinator from the host : 
```
$ python3 Elelabs_EzspFwUtility.py probe -p /dev/ttyUSB0
2025/03/02 00:02:32 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2025/03/02 00:02:32 Elelabs_EzspFwUtility:   Firmware: 7.4.4-0
2025/03/02 00:02:32 Elelabs_EzspFwUtility:   EZSP v13
```
Which makes me think there is no connection issue between the server and the coordinator.

Nerivec, 2025-03-02:
The logs don't really make sense. Entire chunks are missing. Timings are off too.
Can you post the compose file and any other _specific_ configuration you might have?

Nastaliss, 2025-03-02:
Here is the docker-compose.yaml I use : 
```yaml
services:
  # mqtt5 eclipse-mosquitto
  mqtt5:
    image: eclipse-mosquitto
    container_name: mqtt5
    user: 1000:1000
    ports:
      - "1883:1883" #default mqtt port
      - "9001:9001" #default mqtt port for websockets
    volumes:
      - ./config:/mosquitto/config:rw
      - ./data:/mosquitto/data:rw
      - ./log:/mosquitto/log:rw
    restart: unless-stopped

  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:2.1.1
    restart: unless-stopped
    volumes:
        - ./z2data:/app/data
        - /run/udev:/run/udev:ro
    ports:
        # Frontend port
        - 8080:8080
    environment:
        - TZ=Europe/Paris
    devices:
      - /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_d804b0512b14ef11b99a6db8bf9df066-if00-port0:/dev/ttyUSB0
    group_add:
    - dialout
    user: 1000:1000
```
I tried running the container without the rootless configuration with no success
and the configuration.yaml :
```yaml
version: 4
homeassistant:
  enabled: true
frontend:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://mqtt5:1883
  user: [redacted]
  password: [redacted]
serial:
  port: /dev/ttyUSB0
  adapter: ember
availability:
  active:
    timeout: 20
  passive:
    timeout: 240
  enabled: true
advanced:
  channel: 20
  network_key:
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
    - [redacted]
  pan_id: 27413
  ext_pan_id:
    - 102
    - 184
    - 169
    - 159
    - 220
    - 25
    - 57
    - 43
  last_seen: ISO_8601
  log_level: debug
devices: { redacted } 
```
I redacted mqtt creds, network key and the specific devices overrides .

Nastaliss, 2025-03-05:
Update on this. 
I copied all the files on my PC + plugged the usb adapter and it started instantly without any issue. 
I checked for resource starvation on the initial server (a raspberry pi 3b) and it looked fine. 
Any suggestion on what could be the culprit here ?

Nerivec, 2025-03-05:
Can you check for updates on OS, docker?
`uname -a`
`uname -m`
`cat /etc/issue.net`
`docker --version`

Nastaliss, 2025-03-05:
I got :
```
Linux ubuntu 6.11.0-1008-raspi #8-Ubuntu SMP PREEMPT_DYNAMIC Tue Jan 21 12:05:23 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux
aarch64
Ubuntu 24.10
Docker version 27.5.1, build 9f9e405
```

Nerivec, 2025-03-05:
Can you check:
- `- dialout` indentation is off in compose, I doubt docker would mind, but just in case...
- if you can get more info from `docker logs CONTAINER_ID` from a few container startups (maybe one after a full reboot of the pi?)

Nastaliss, 2025-03-06:
Indentation is fine yaml-wise in the docker-compose (I tried anyway, it did not change anything).
I tried restarting many times and it did not change the behavior shared previously. Starting, then hanging at `Using zigbee-herdsman with setting`, but still saving state periodically, event without having completely started.
I'm away from home for some days, but my next plan of action will be to swap the microsd card, and then to try with raspberry pi OS. But I'm still puzzled to why this is happening as it worked fine on the same hardware for some time now.

njordan77, 2025-04-01:
is there any update?

Im on 2.2.0 and also stuck with a yaml config that worked with 2.1.3, but now i only get that far.....re-installed from scratch, same. even with the standard yaml config it runs into this stuck issue.

```
pi@SmartPi:/opt/zigbee2mqtt $ pnpm start

> zigbee2mqtt@2.2.0 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
[2025-04-01 23:30:54] info:     z2m: Logging to console, file (filename: log.log)
[2025-04-01 23:30:54] info:     z2m: Starting Zigbee2MQTT version 2.2.0 (commit #c5c07e7d)
[2025-04-01 23:30:54] info:     z2m: Starting zigbee-herdsman (3.4.11)
```

njordan77, 2025-04-02:
pi@SmartPi:/opt/zigbee2mqtt $ uname -a
Linux SmartPi 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
pi@SmartPi:/opt/zigbee2mqtt $ name -m
-bash: name: command not found
pi@SmartPi:/opt/zigbee2mqtt $ uname -m
aarch64
pi@SmartPi:/opt/zigbee2mqtt $ cat /etc/issue.net
Debian GNU/Linux 12

Nerivec, 2025-04-02:
The two versions affected by this are both aarch64 from January 2x (granted, kernel is different...).
Any chance you have an upgrade available?

Nerivec, 2025-04-02:
An OS upgrade (SmartPi).
The most likely scenario for the hang at that moment is the loading of the serialport library, but the rebuild should take care of that since it's then definitely "built for the machine it's run on"... so I'm a bit out of ideas... 😅
Could you try running Z2M with the adapter unplugged, see if you get an error, or if it still hangs like before?

njordan77, 2025-04-02:
ok, let me check. but small question, can this really be the issue if 2.1.3 worked fine and 2.2.0 is a huge problem case?

Nerivec, 2025-04-02:
Did you have any other updates (non-Z2M-related) on your machine between 2.1.3 and 2.2.0? It doesn't take much sometimes to create an incompatibility, and the Pi is often having some...
We have to find a way to pinpoint where the problem is exactly, which is really hard when it just hangs like this. Hopefully the startup with unplugged adapter will indicate more.

njordan77, 2025-04-02:
@Nerivec i'm on the road and can provide unplug feedback later. what i definitely can do is to do an OS update. you do believe that this upgrade will most probably be the key to success? beside finding the real issue without update?

I regularily do package updates, so definitely something has changed between the two versions. but i also have rebooted several times before going to 2.2.0 and never did block the startup procedure.....but as you said, something minor can have major impact

Nerivec, 2025-04-02:
The behavior is abnormal at a stage that does indicate some kind of problem between nodejs and the OS. But without pinpointing the exact step that does cause the hang, it's hard to say what the fix will be.
Since you have no new kernel upgrade, I guess we're left with finding that exact cause.
Here's something else you can try:
- Reboot the entire PI (if you haven't already)
- Remove the contents of the Z2M `data` folder, everything (make sure to backup anything you might want to backup first), this will trigger the new onboarding feature in 2.2.0, which could help us figuring out if serialport is indeed the issue (the onboarding will change the entry point, which, if serialport is the problem, should also change where it hangs).
  - https://www.zigbee2mqtt.io/guide/getting-started/#onboarding

njordan77, 2025-04-02:
i already tried onboarding, but it forcces to be on 8080. that is taken by other services.
Is there a way to redirect onbloarding to another port in any kind of config?

Nerivec, 2025-04-02:
Yes, see the section in the docs `Environment variables available to customize the onboarding process` in the link above.
`Z2M_ONBOARD_URL=http://0.0.0.0:12345 pnpm start`

njordan77, 2025-04-02:
without the stick its the same behaviour as before. can you tell me the way to roll back to 2.1.3 to see if this now is as bad for the previous release as for 2.2.0....

Nerivec, 2025-04-02:
You must have kept the `configuration.yaml` otherwise it would not try to start like this, but would start onboarding instead (onboarding starts before anything from Z2M actually starts, it detects the presence of the `configuration.yaml` file to determine if it should run or not).

njordan77, 2025-04-03:
@Nerivec 
oh sorry, i removed it but with the first unsuccessful start at a blocked port it already created a blank yaml file. removed it - now the log looks ok, but the strange thing is that i cannot open the url:port in browser. when i open openhab at 8080 it works fine....
Any idea...

i've tried standard 0.0.0.0:9090 and 192.168.1.11:9090....both with no success to see the onboarding site.
```

pi@SmartPi:/opt/zigbee2mqtt $ Z2M_ONBOARD_URL=http://192.168.1.11:9090 pnpm start

> zigbee2mqtt@2.2.0 start /opt/zigbee2mqtt
> node index.js

Starting Zigbee2MQTT without watchdog.
Onboarding page is available at http://192.168.1.11:9090/
```

Nerivec, 2025-04-03:
> i've tried standard 0.0.0.0:9090 and 192.168.1.11:9090....both with no success to see the onboarding site.

I'm not sure, this is very dependent on your setup. Are you maybe not entering the proper URL on the browser? You may have to enter for e.g. `http://192.168.1.11:9090/` in the browser, while using `http://0.0.0.0:9090` for the onboarding. It should be the same as you used to reach frontend (assuming you use the same port for onboarding).

> Can you please be so kind and let me know how to build the project with a certail version like 2.1.3 again, would like to see if this would work again or same as 2.2.0

You can clone the zigbee2mqtt repository with a [specific tag](https://github.com/Koenkk/zigbee2mqtt/tags):
```bash
git clone --depth 1 --branch 2.1.3 https://github.com/Koenkk/zigbee2mqtt
```

njordan77, 2025-04-03:
@Nerivec was not expecting this as 2.1.3 worked til the day i updated to 2.2.0. But now 2.1.3 is stuck as well, guess you assumed this to happen.....

is there any update upgrade whatever i could do to fix this soon, my house relies a lot on my zigbee installation.....little pain in the ass if all systems now are stuck :-)

Nerivec, 2025-04-03:
Without much details on the _why_ this is happening, the only "quick" options I can see are pretty radical...
- Try another OS on the PI
- Try another machine
- Try a container/VM setup (bit backwards, I know, but maybe it can bypass whatever issue the bare install has...)

PS: I noticed [HAOS was having issues with the PI5](https://github.com/home-assistant/operating-system/releases/tag/15.1) recently, you might want to investigate that, in case it somehow relates to this issue as well... I don't know much about the whole PI _eco-system_, so won't be able to help much there...

njordan77, 2025-04-03:
@Nerivec i now tried once more from home. i can see the log about the port opened for the frontend to do the onboarding, i did it in several ways, also to start the browser, but nothing works, after a minute or so there comes a timeout but nothing showing in any kind of browser from firefox to edge, and chrome.......

Is that possible, i thought when i see this log message a webserver has started for sure.....

jimmyeao, 2025-04-05:
Having the same issue
```
Apr 05 18:44:08 zigbeepi npm[4692]: Starting Zigbee2MQTT without watchdog.
Apr 05 18:44:11 zigbeepi npm[4692]: TypeError: Module "file:///opt/zigbee2mqtt/package.json" needs an import assertion of type "json"
Apr 05 18:44:11 zigbeepi npm[4692]:     at new NodeError (node:internal/errors:406:5)
Apr 05 18:44:11 zigbeepi npm[4692]:     at validateAssertions (node:internal/modules/esm/assert:94:15)
Apr 05 18:44:11 zigbeepi npm[4692]:     at defaultLoad (node:internal/modules/esm/load:122:3)
Apr 05 18:44:11 zigbeepi npm[4692]:     at ModuleLoader.load (node:internal/modules/esm/loader:396:13)
Apr 05 18:44:11 zigbeepi npm[4692]:     at ModuleLoader.moduleProvider (node:internal/modules/esm/loader:278:56)
Apr 05 18:44:11 zigbeepi npm[4692]:     at new ModuleJob (node:internal/modules/esm/module_job:65:26)
Apr 05 18:44:11 zigbeepi npm[4692]:     at ModuleLoader.#createModuleJob (node:internal/modules/esm/loader:290:17)
Apr 05 18:44:11 zigbeepi npm[4692]:     at ModuleLoader.getJobFromResolveResult (node:internal/modules/esm/loader:248:34)
Apr 05 18:44:11 zigbeepi npm[4692]:     at ModuleLoader.getModuleJob (node:internal/modules/esm/loader:229:17)
Apr 05 18:44:11 zigbeepi npm[4692]:     at ModuleLoader.import (node:internal/modules/esm/loader:315:23)
Apr 05 18:44:11 zigbeepi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Apr 05 18:44:11 zigbeepi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 05 18:44:11 zigbeepi systemd[1]: zigbee2mqtt.service: Consumed 4.708s CPU time.
Apr 05 18:44:11 zigbeepi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 22.
Apr 05 18:44:11 zigbeepi systemd[1]: Stopped zigbee2mqtt.
Apr 05 18:44:11 zigbeepi systemd[1]: zigbee2mqtt.service: Consumed 4.708s CPU time.
Apr 05 18:44:11 zigbeepi systemd[1]: Started zigbee2mqtt.
Apr 05 18:44:12 zigbeepi npm[4725]: > zigbee2mqtt@2.2.1 start
Apr 05 18:44:12 zigbeepi npm[4725]: > node index.js

```
Running Node V20.8.0

Nerivec, 2025-04-05:
@jimmyeao this is not the same issue as this. As mentioned in the [release notes](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.2.1), try updating your nodejs version to latest 20.19.x (or better yet, to latest v22).
Instructions here: https://github.com/Koenkk/zigbee2mqtt/issues/26937#issuecomment-2773534427

_There are some incompatibilities with older versions of v18 and v20 showing up due to recent code changes ([20.8.0 is from 2023](https://versionlog.com/nodejs/))._

jimmyeao, 2025-04-05:
> [@jimmyeao](https://github.com/jimmyeao) this is not the same issue as this. As mentioned in the [release notes](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.2.1), try updating your nodejs version to latest 20.19.x (or better yet, to latest v22). Instructions here: [#26937 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/26937#issuecomment-2773534427)
> 
> _There are some incompatibilities with older versions of v18 and v20 showing up due to recent code changes ([20.8.0 is from 2023](https://versionlog.com/nodejs/))._

I have updated to 22 - the issue persists and it loops trying to start. Have reverted back to the 1.9.x release which works

Nerivec, 2025-04-05:
What's the error you are getting when it crashes before the retry?

jimmyeao, 2025-04-06:
Sorry I meant the 2.1.3 release is what I have reverted to. The only logs I had from the new version are above.

f4l0, 2025-04-07:
Same issue here. 
It works with a clean data directory and a fresh installation, but once I copy over my old data directory, it crashes...

Nerivec, 2025-04-07:
@f4l0 could you share your data (`database.db`, `configuration.yaml`)? You can find me on Discord, or create a private Github repo, if you prefer to share it privately.

closed 2025-06-15 by github-actions[bot]