Koenkk/zigbee2mqtt #26557: 2.1.2 - makes the hue lighbulb NOT not report to MQTT, not be available.

tomtom13, 2025-03-02:
### What happened?

As in subject after updating to 2.1.2, ALL hue lightbulbs (with exception of filament simulating ones) shows as unavailable in MQTT. 

[image]

For the time being, I'm reverting to old backup. 

### Zigbee2MQTT version

2.1.2

### Adapter firmware version

20240710

### Adapter

UZG-01

### Setup

addon to HASS CORE. Everything running as a VM. 

### Debug log

Nothing int the logs.

trenchcoatjedi, 2025-03-02:
EXACT same issue with Philips Hue white and color ambiance LightStrip outdoor 2m (9290022890)

I was in the process of opening an issue when I saw this.

tomtom13, 2025-03-02:
Yeah, my whole house is non operational more or less. Good that I make those updates at night, but c'mon I would like to get some sleep rather than keep reverting to old backups.

tomtom13, 2025-03-02:
I can report that revert to VM backup from 4 hours ago WORK without a glitch - all lights are acting perfectly well. Only thing that shows up is an update to Z2M to 2.1.2 :) who knows, maybe I'll click it for s&g.

trenchcoatjedi, 2025-03-02:
Reverted to Z2M 2.1.1-1 through HA built in backup/restore function. Did not restore the entire VM...just Z2M and the issue was solved. MQTT was showing the light control as unavailable/"no longer provided by this entity" in HA under Z2M 2.1.2 with a similar screenshot to the one from @tomtom13 in the OP. My screenshot was taken after removing the device from HA entirely and re-adding it.

jelmervdm, 2025-03-02:
Same issue with an Innr AE 270 T. Device shows in Z2MQTT, but is showing offline in HA. Reverted back to 2.1.1, which resolved the issue.

ireun, 2025-03-02:
Same here, IKEA lighbulbs all affected, though ikea smart plug works

kaisamng, 2025-03-02:
Not running as an addon; running in my own self-managed containers; same issue in Hue and Innr lights. How does an update this borked get pushed?

Here's the error that HA logs for every light when on 2.1.2. 

```
2025-03-01 23:14:35.036 ERROR (Dummy-1) [homeassistant.helpers.dispatcher] Exception in _async_setup_entity_entry_from_discovery when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}, {'topic': 'zigbee2mqtt/Basement Ceiling Light 1/availability', 'value_template': '{{ value_json.state }}'}], 'availability_mode': 'all', 'brightness': True, 'brightness_scale': 254, 'command_topic': 'zigbee2mqtt/<Light Name>/set', 'device': {'hw_version': 1, 'identifiers': ['zigbee2mqtt_<id>'], 'manufacturer': 'Philips', 'model': 'Hue white ambiance BR30 flood light', 'model_id': '464800', 'sw_version': '1.122.2', 'via_device': 'zigbee2mqtt_bridge_<bridge_id>, 'name': 'Basement Ceiling Light 1'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'candle', 'finish_effect', 'stop_effect', 'stop_hue_effect'], 'max_mireds': 500, 'min_mireds': 150, 'object_id': 'basement_ceiling_light_1', 'origin': {'support_url': 'https://www.zigbee2mqtt.io', 'name': 'Zigbee2MQTT', 'sw_version': '2.1.2'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/Basement Ceiling Light 1', 'supported_color_modes': ['brightness', 'color_temp'], 'unique_id': '<id>_light_zigbee2mqtt', 'name': None},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/entity.py", line 260, in _async_setup_entity_entry_from_discovery
    config: DiscoveryInfoType = discovery_schema(discovery_payload)
                                ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 259, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 382, in _exec
    v = func(v)
  File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
           ~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 779, in validate_callable
    return schema(data)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 43, in validate_mqtt_light_discovery
    config: ConfigType = schemas[config_value[CONF_SCHEMA]](config_value)
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 259, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 382, in _exec
    v = func(v)
  File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
           ~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 779, in validate_callable
    return schema(data)
  File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
           ~~~~~~~~~~~~~~^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 549, in validate_dict
    return base_validate(path, data.items(), out)
  File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 330, in validate_mapping
    cval = cvalue(key_path, value)
  File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 256, in _run
    return self._exec(self._compiled, value, path)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 384, in _exec
    v = func(path, v)
  File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 779, in validate_callable
    return schema(data)
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 137, in valid_supported_color_modes
    raise vol.Error(f"Invalid supported_color_modes {sorted(color_modes)}")
voluptuous.error.Error: Invalid supported_color_modes ['brightness', 'color_temp']
```

rentonmark, 2025-03-02:
Same issue with Philips 9290022944 Hue white ambiance E14 and others

tibmeister, 2025-03-02:
Updated to 2.1.2, anything with a light control lost the light control entity stating the entity was no longer available.  Could control through the Z2M dashboard, but HA lost the entities for the light switches all together.  Reverting to 2.1.1 restored all lights and controls.

gouthamravee, 2025-03-02:
This is affecting all of my ikea and osram lights too. 

The error I'm getting is 

```
Exception in _async_setup_entity_entry_from_discovery when dispatching 'mqtt_discovery_new_light_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}, {'topic': 'zigbee2mqtt/Outside Front Lights/availability', 'value_template': '{{ value_json.state }}'}], 'availability_mode': 'all', 'brightness': True, 'brightness_scale': 254, 'command_topic': 'zigbee2mqtt/Outside Front Lights/set', 'device': {'identifiers': ['zigbee2mqtt_1221051039810110150109113116116_8'], 'manufacturer': 'Zigbee2MQTT', 'model': 'Group', 'sw_version': 'Zigbee2MQTT 2.1.2', 'via_device': 'zigbee2mqtt_bridge_0x00124b00258c9560', 'name': 'Outside Front Lights'}, 'effect': True, 'effect_list': ['blink', 'breathe', 'okay', 'channel_change', 'finish_effect', 'stop_effect', 'colorloop', 'stop_colorloop'], 'max_mireds': 454, 'min_mireds': 250, 'object_id': 'outside_front_lights', 'origin': {'name': 'Zigbee2MQTT', 'sw_version': '2.1.2', 'support_url': 'https://www.zigbee2mqtt.io'}, 'schema': 'json', 'state_topic': 'zigbee2mqtt/Outside Front Lights', 'supported_color_modes': ['brightness', 'xy', 'color_temp'], 'unique_id': '8_light_zigbee2mqtt', 'name': None},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/entity.py", line 260, in _async_setup_entity_entry_from_discovery config: DiscoveryInfoType = discovery_schema(discovery_payload) ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 259, in __call__ return self._exec((Schema(val) for val in self.validators), v) ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 382, in _exec v = func(v) File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 205, in __call__ return self._compiled([], data) ~~~~~~~~~~~~~~^^^^^^^^^^ File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 779, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/light/__init__.py", line 43, in validate_mqtt_light_discovery config: ConfigType = schemas[config_value[CONF_SCHEMA]](config_value) ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^ File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 259, in __call__ return self._exec((Schema(val) for val in self.validators), v) ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 382, in _exec v = func(v) File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 205, in __call__ return self._compiled([], data) ~~~~~~~~~~~~~~^^^^^^^^^^ File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 779, in validate_callable return schema(data) File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 205, in __call__ return self._compiled([], data) ~~~~~~~~~~~~~~^^^^^^^^^^ File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 549, in validate_dict return base_validate(path, data.items(), out) File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 330, in validate_mapping cval = cvalue(key_path, value) File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 256, in _run return self._exec(self._compiled, value, path) ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.13/site-packages/voluptuous/validators.py", line 384, in _exec v = func(path, v) File "/usr/local/lib/python3.13/site-packages/voluptuous/schema_builder.py", line 779, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 137, in valid_supported_color_modes raise vol.Error(f"Invalid supported_color_modes {sorted(color_modes)}") voluptuous.error.Error: Invalid supported_color_modes ['brightness', 'color_temp', 'xy']
``` 

Seems like an old issue has come back - https://github.com/esphome/issues/issues/2358

kkellner, 2025-03-02:
Is there a way to revert to the old plugin version (docker image) via UI or command-line (other the storing backup).   The `ha addon update` command does not seem to have a version option.

eeclee, 2025-03-02:
same problem, restored the previous version and everything is back to normal.

tomtom13, 2025-03-02:
@Koenkk - it might be worth looking into this one, since "quite a few" are affected with 2.1.2 straight after the upgrade.

mrplow, 2025-03-02:
> Is there a way to revert to the old addon version (docker image) via UI or command-line (other then restoring backup). The `ha addon update` command does not seem to have a version option.

image: koenkk/zigbee2mqtt:2.1.1

gouthamravee, 2025-03-02:
For anyone dealing with this and didn't have a backup, here's how to roll back. 

1. Make sure you have access to the `addons` folder in HA. I did this by installing the samba add-on and setting up SMB access to HA. 
2. Download the source code zip archive for the hassio-zigbee2mqtt add-on version 2.1.1 from the releases page - https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/releases
3. Extract the zip archive somewhere
4. Connect to the SMB share for HA and go to the add-ons folder/share
5. Go to the folder you extracted in step 3, copy the contents into the add-ons SMB share. The contents should include folders like `common`, `zigbee2mqtt`, `zigbee2mqtt-edge`, etc. Copy these folders and all the files in the parent folder. 
6. Restart Home Assistant
7. Once restarted, go to the add-ons page. You might have to do a `ctrl + shift + r`. 
8. You should see a new row of addons under "local addons"
9. Go to the already running 2.1.2 Zigbee2Mqtt addon and copy everything from the 'config' section. 
10. Return to the local Zigbee2Mqtt addon and paste everything into the config section.
11. Stop the 2.1.2 Zigbee2Mqtt 
12. Start the local Zigbee2Mqtt addon
13. Everything should work again. 

I got the idea for this from - https://www.youtube.com/watch?v=yxbZAv0oYcE&t=73s

hmmbob, 2025-03-02:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/26553

PtiBed0, 2025-03-02:
Hello,

I had the same problem this morning after updating to version 2.1.2
Restored the previous version and everything is OK

Thank

Ropuh, 2025-03-02:
Hi,
The same here:
IKEA LED1537R6/LED1739R5 entity (in HA) gone
but
IKEA LED1537R6/LED1739R5 exists and works well.

Both works well in zigbee2mqtt frontend all the time.

ippo112, 2025-03-02:
same issue with Müller Licht 404006/404008/404004  (spot colour light) after upgrade to to version 2.1.2

liamlain, 2025-03-02:
[image]

yes,updated to 2.1.2 today is not work ,the light entity not available anymore.

ukmgranger, 2025-03-02:
Seeing the same issue here. I'm mainly on IKEA bulbs and many or not available in HA.  It also looks like some groups (eg light.kitchen - a group of about 20 devices) are unavailable too.

hmmbob, 2025-03-02:
### Do not respond with "yes, my lights are broken too"

The issue is clear and the root cause is also clear. We'll just need to wait for a fix or rollback to the previous version.

If you have this issue too, give a thumbs up 👍 in the first post to indicate this and if you'd like to keep updated on progress, hit the "subscribe" button (all the way at the bottom on mobile or right-hand side of the screen on PC).

teymur1988, 2025-03-02:
THe issue I submitted #26563 was marked as duplicate if this, even though I don't have any IKEA or Philips Hue lights. I restored to 2.1.1 from addon backups and still have this issue? any ideas?

Koenkk, 2025-03-02:
Fixed in 2.1.3!

closed 2025-03-02 by Koenkk

teymur1988, 2025-03-02:
> Fixed in 2.1.3!

```
[12:00:01] INFO: Preparing to start...
[12:00:02] INFO: Socat not enabled
[12:00:02] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-02 12:00:03] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-02 12:00:03] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
[2025-03-02 12:00:03] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-02 12:00:03] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8ee983a24db6ed119fbfdc0ea8669f5d-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-03-02 12:00:03] info: 	zh:zstack:znp: Serialport opened
[2025-03-02 12:00:03] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-02 12:00:03] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-02 12:00:03] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-02 12:00:03] error: 	z2m: Exiting...
[2025-03-02 12:00:03] error: 	z2m: Error: --> 'SREQ: ZDO - extAddGroup - {"endpoint":242,"groupid":2948,"namelen":0,"groupname":[]}' failed with status '(0xb2: APS_TABLE_FULL)' (expected '(0x00: SUCCESS)')
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:258:27
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZnpAdapterManager.addToGroup (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:491:13)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:102:9)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
```

Noir1992, 2025-03-02:
After installing the 2.1.3 my lights are working again. Thanks a lot for the fast fix, great work!

fnordson, 2025-03-02:
> Same here, IKEA lighbulbs all affected, though ikea smart plug works

My TRADFRIs are all not affected by this.  
Everything works fine here.

trenchcoatjedi, 2025-03-02:
Confirm issue fixed in 2.1.3 Thank you @Koenkk