Koenkk/zigbee2mqtt #26558: 2.1.2-1 Error after update

DChe47, 2025-03-02:
### What happened?

Hello all! After update from 2.1.2 to 2.1.2-1 my device TS0202 _TZ3210_0aqbrnts have error:

`z2m: Failed to configure 'Датчик у входа', attempt 2 (Error: ZCL command 0x8cf681fffebf9c5e/3 msIlluminanceMeasurement.configReport([{"minimumReportInterval":10,"maximumReportInterval":3600,"reportableChange":5,"attribute":"measuredValue"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 34995 - 3 - 52 - 1024 - 7 after 10000ms) at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:663:47) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:695:28) at /app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20 at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16) at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:949:28) at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9))`

Despite these errors, the device transmits data correctly.
Please help me fix these errors.
Thank you.

PS: You can also correct the manufacturer's name of this device from EFK to EKF. It must be a typo.

### Zigbee2MQTT version

2.1.2-1

### Adapter firmware version

20220302

### Adapter

zStack3x0

### Setup

Addon of Home assistant Supervised, Debian 12, Micro PC x86-64

### Debug log

[07:32:16] INFO: Preparing to start...
[07:32:17] INFO: Socat not enabled
[07:32:17] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-02 07:36:18] error: 	z2m: Failed to configure 'Датчик у входа', attempt 1 (Error: ZCL command 0x8cf681fffebf9c5e/3 msIlluminanceMeasurement.configReport([{"minimumReportInterval":10,"maximumReportInterval":3600,"reportableChange":5,"attribute":"measuredValue"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 34995 - 3 - 26 - 1024 - 7 after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:663:47)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:695:28)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:949:28)
    at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9))
[2025-03-02 07:38:37] error: 	z2m: Publish 'set' 'state' to 'Розетка Н' failed: 'Error: ZCL command 0xa4c1383d71485bc0/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 51925 - 1 - 36 - 6 - 11 after 10000ms)'
[2025-03-02 07:38:37] error: 	z2m: Publish 'set' 'state' to 'Розетка К' failed: 'Error: ZCL command 0xa4c138c885cce020/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 14062 - 1 - 38 - 6 - 11 after 10000ms)'
[2025-03-02 07:41:19] error: 	z2m: Failed to configure 'Датчик у входа', attempt 2 (Error: ZCL command 0x8cf681fffebf9c5e/3 msIlluminanceMeasurement.configReport([{"minimumReportInterval":10,"maximumReportInterval":3600,"reportableChange":5,"attribute":"measuredValue"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 34995 - 3 - 52 - 1024 - 7 after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:663:47)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:695:28)
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:949:28)
    at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9))
[2025-03-02 07:43:37] error: 	z2m: Publish 'set' 'state' to 'Розетка К' failed: 'Error: ZCL command 0xa4c138c885cce020/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 14062 - 1 - 65 - 6 - 11 after 10000ms)'
[2025-03-02 07:46:20] error: 	z2m: Failed to configure 'Датчик у входа', attempt 3 (Error: ZCL command 0x8cf681fffebf9c5e/3 msIlluminanceMeasurement.configReport([{"minimumReportInterval":10,"maximumReportInterval":3600,"reportableChange":5,"attribute":"measuredValue"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 34995 - 3 - 79 - 1024 - 7 after 10000ms)
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/waitress.ts:67:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:663:47)
[... cut ...]
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:949:28)
    at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9))

tomtom13, 2025-03-02:
@Koenkk - might be worth looking into this one, since "quite a few" are affected with 2.1.2

MaJerle, 2025-03-02:
https://github.com/Koenkk/zigbee2mqtt/issues/26557

DChe47, 2025-03-02:
After upgrading to 2.1.3, this problem remained. :(

DChe47, 2025-03-02:
I have restarted z2m twice and now there are no such errors. I have closed the problem. Thank you.

Nerivec, 2025-03-05:
If configure fails for a battery powered device:
- press the button on the device every 1-2 seconds until configure is done.
- trigger "reconfigure" in frontend

closed 2025-03-07 by DChe47