Koenkk/zigbee2mqtt #26559: Some device features are unavailable (only 1 of 2 endpoints exposed)

s9596362, 2025-03-02:
### What happened?

I have a Tuya-compatible device (a 1 gang ac wall outlet with 2 USB charging ports and power monitoring), which when used with the tuya smartlife app, I'm able to turn off the outlet and the USB charging ports on/off separately. Despite currently being supported in z2m, I can only turn on/off the outlet. There's no on/off state shown for the USB charging ports & therefore, I'm unable to turn on/off the USB charging ports. Z2M currently identifies the device as TS011F_plug_1. Going through zigbee-herdsman-converters/blob/master/src/devices/tuya.ts on github, I saw that there are existing TS001F devices that are similar to the one (eg: TS011F_2_gang_power) I have but unfortunately Z2M failed to recognize my device as such.

Here's the general external definition generated by z2m of my device for your reference:

```
const definition = {
    zigbeeModel: ['TS011F'],
    model: 'TS011F',
    vendor: '_TZ3000_gazjngjl',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2}}), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};
```

### What did you expect to happen?

Seeing two on/off states to allow turning on/off the outlet and USB charging ports respectively.

### How to reproduce it (minimal and precise)

1) Pair the device to Z2M.
2) See the Exposes tab for the device on the frontend.

### Zigbee2MQTT version

2.1.2

### Adapter firmware version

20221226

### Adapter

No-brand/Generic Model DS-1530ZN-2USB

### Setup

Add-on on Home Assistant OS on Beelink EQi12

### Debug log

I don't think this is necessary since there's no errors in when using the device.

bobo-cher, 2025-03-05:
There is a trick to force z2m to use other converter. Stop z2m and edit it’s database.db in a text editor (HA has a plug-in "File editor" for that). Set manufacturerName of your device to "_TZ3000_bep7ccew". This id I've got by searching https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts for "TS011F_2_gang_power".

Koenkk, 2025-03-06:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-03-06 by Koenkk

Kamishiku, 2025-04-04:
I am experiencing this issue with _TZ3000_mup3wbfp it is a 2 gang switch from Melery it detects as a TS011F_plug_1 instead.