Koenkk/zigbee2mqtt #26561: Phillips Hue Brightness Not Showing in Home Assistant

Spankythemusical, 2025-03-02:
### What happened?

I have three different Hue bulbs, 9290023351, 9290022268, and 9290022266A and all of them appear in Home Assistant as expected.  For some reason, the 9290023351 and 9290022268 bulbs do not expose the brightness attribute in Home Assistant.  Both show "light (state, brightness), power_on_behavior, effect" as exposures on the device pages.

[image]
[image]
[image]

### What did you expect to happen?

Ability to control brightness to all Phillips Hue bulbs via HA

### Zigbee2MQTT version

 1.39.1 commit: 403d3c0

### Adapter firmware version

20220219

### Adapter

Sonoff Adapter 3.0

### Setup

RPi, running only Z2M

MaJerle, 2025-03-02:
[Duplicate of #26557](https://github.com/Koenkk/zigbee2mqtt/issues/26557)

Spankythemusical, 2025-03-02:
The lights are available.  I can turn them on/off and set the effects.  There is just no option for brightness on those two models. I have 2 of all three types.

MaJerle, 2025-03-02:
I deleted the previous comment as I think it may not be a duplicate.

Spankythemusical, 2025-03-02:
No worries!

Koenkk, 2025-03-02:
Should be fixed in 2.1.3

Spankythemusical, 2025-03-02:
Awesome!  Thank you!

Koenkk, 2025-03-02:
@Spankythemusical can you confirm it works now?

Spankythemusical, 2025-03-02:
Just removed and re-added the 9290023351 bulbs and now I see the Brightness attributes in HA.  Perfect!

closed 2025-03-02 by Koenkk