Koenkk/zigbee2mqtt #26562: Devices unavailable in HA after Z2M update

fhuebner2, 2025-03-02:
### What happened?

I can still control my devices via Zigbee2Mqtt GUI but not via Home Assistant anymore after update to 2.1.2-1

### Zigbee2MQTT version

2.1.2-1

### Adapter firmware version

0x26720700

### Adapter

Conbee2

### Setup

Add-On on Home Assistant on Raspi4

MaJerle, 2025-03-02:
[Duplicate of #26557](https://github.com/Koenkk/zigbee2mqtt/issues/26557)

Koenkk, 2025-03-02:
Fixed in 2.1.3!

closed 2025-03-02 by Koenkk

fhuebner2, 2025-03-02:
Thanks, when is release?

Koenkk, 2025-03-02:
@fhuebner2 few mins from now

boesing, 2025-03-16:
Hm, I do still have the same issue on 2.1.3 with zzh! on `20220219`.
Should I create a dedicated issue for that or can we have that handled in this one?