Koenkk/zigbee2mqtt #26563: Z2M Fails to start after updating to 2.1.2

teymur1988, 2025-03-02:
### What happened?

Hi 

Could you please help. I updated Z2M from 2.1.1 to 2.1.2 and now it fails to start. I'm using z2m as HA addon, with sonoff usb dongle P as adapter. Debug logs are attached.

### What did you expect to happen?

I expected that it would start normally.

### How to reproduce it (minimal and precise)

Just update from 2.1.1 to 2.1.2 using 

### Zigbee2MQTT version

2.1.2

### Adapter firmware version

can't check as addon doesn't start now.

### Adapter

sonoff dongle P

### Setup

Addon on HA which is run in a VM.

### Debug log

```
[11:03:02] INFO: Preparing to start...
[11:03:02] INFO: Socat not enabled
[11:03:02] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-02 11:03:03] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-02 11:03:03] info: 	z2m: Starting Zigbee2MQTT version 2.1.2 (commit #unknown)
[2025-03-02 11:03:03] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-02 11:03:03] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8ee983a24db6ed119fbfdc0ea8669f5d-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-03-02 11:03:03] info: 	zh:zstack:znp: Serialport opened
[2025-03-02 11:03:04] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-02 11:03:04] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-02 11:03:04] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-02 11:03:04] error: 	z2m: Exiting...
[2025-03-02 11:03:04] error: 	z2m: Error: --> 'SREQ: ZDO - extAddGroup - {"endpoint":242,"groupid":2948,"namelen":0,"groupname":[]}' failed with status '(0xb2: APS_TABLE_FULL)' (expected '(0x00: SUCCESS)')
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:258:27
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZnpAdapterManager.addToGroup (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:491:13)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:102:9)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[11:04:06] INFO: Preparing to start...
[11:04:06] INFO: Socat not enabled
[11:04:06] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-02 11:04:07] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-02 11:04:08] info: 	z2m: Starting Zigbee2MQTT version 2.1.2 (commit #unknown)
[2025-03-02 11:04:08] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-02 11:04:08] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8ee983a24db6ed119fbfdc0ea8669f5d-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-03-02 11:04:08] info: 	zh:zstack:znp: Serialport opened
[2025-03-02 11:04:08] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-02 11:04:08] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-02 11:04:08] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-02 11:04:08] error: 	z2m: Exiting...
[2025-03-02 11:04:08] error: 	z2m: Error: --> 'SREQ: ZDO - extAddGroup - {"endpoint":242,"groupid":2948,"namelen":0,"groupname":[]}' failed with status '(0xb2: APS_TABLE_FULL)' (expected '(0x00: SUCCESS)')
[... 197 lines cut ...]
[2025-03-02 11:19:30] debug: 	zh:zstack:unpi:parser: --> parsed 19 - 3 - 5 - 74 - [1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0] - 61
[2025-03-02 11:19:30] debug: 	zh:zstack:znp: <-- SRSP: ZDO - extFindGroup
[2025-03-02 11:19:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-02 11:19:30] debug: 	zh:zstack:znp: --> SREQ: ZDO - extAddGroup - {"endpoint":242,"groupid":2948,"namelen":0,"groupname":[]}
[2025-03-02 11:19:30] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,75,242,132,11,0,23]
[2025-03-02 11:19:30] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,75,178,157]
[2025-03-02 11:19:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,75,178,157]
[2025-03-02 11:19:30] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 75 - [178] - 157
[2025-03-02 11:19:30] debug: 	zh:zstack:znp: <-- SRSP: ZDO - extAddGroup
[2025-03-02 11:19:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-02 11:19:30] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-02 11:19:30] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-02 11:19:30] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-02 11:19:30] error: 	z2m: Exiting...
[2025-03-02 11:19:31] error: 	z2m: Error: --> 'SREQ: ZDO - extAddGroup - {"endpoint":242,"groupid":2948,"namelen":0,"groupname":[]}' failed with status '(0xb2: APS_TABLE_FULL)' (expected '(0x00: SUCCESS)')
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:258:27
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZnpAdapterManager.addToGroup (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:491:13)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:102:9)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)``
```

teymur1988, 2025-03-02:
I went back to 2.1.1 and I keep getting the same error:

```
Starting Zigbee2MQTT without watchdog.
[2025-03-02 11:34:51] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-02 11:34:51] info: 	z2m: Starting Zigbee2MQTT version 2.1.1 (commit #unknown)
[2025-03-02 11:34:51] info: 	z2m: Starting zigbee-herdsman (3.2.5)
[2025-03-02 11:34:51] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8ee983a24db6ed119fbfdc0ea8669f5d-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-03-02 11:34:51] info: 	zh:zstack:znp: Serialport opened
[2025-03-02 11:34:52] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-02 11:34:52] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-02 11:34:52] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-02 11:34:52] error: 	z2m: Exiting...
[2025-03-02 11:34:52] error: 	z2m: Error: --> 'SREQ: ZDO - extAddGroup - {"endpoint":242,"groupid":2948,"namelen":0,"groupname":[]}' failed with status '(0xb2: APS_TABLE_FULL)' (expected '(0x00: SUCCESS)')
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:258:27
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZnpAdapterManager.addToGroup (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:491:13)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:102:9)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
```

teymur1988, 2025-03-02:
Updating to 2.1.3 didn't help:

```
[12:00:01] INFO: Preparing to start...
[12:00:02] INFO: Socat not enabled
[12:00:02] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-02 12:00:03] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-02 12:00:03] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
[2025-03-02 12:00:03] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-02 12:00:03] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_8ee983a24db6ed119fbfdc0ea8669f5d-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-03-02 12:00:03] info: 	zh:zstack:znp: Serialport opened
[2025-03-02 12:00:03] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-02 12:00:03] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-02 12:00:03] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-02 12:00:03] error: 	z2m: Exiting...
[2025-03-02 12:00:03] error: 	z2m: Error: --> 'SREQ: ZDO - extAddGroup - {"endpoint":242,"groupid":2948,"namelen":0,"groupname":[]}' failed with status '(0xb2: APS_TABLE_FULL)' (expected '(0x00: SUCCESS)')
    at /app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:258:27
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZnpAdapterManager.addToGroup (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:491:13)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:102:9)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
```

Koenkk, 2025-03-02:
Could you try reflashing the latest adapter firmware on your stick?

teymur1988, 2025-03-02:
Has there been a new version release for sonoff dongle P recently?

teymur1988, 2025-03-02:
> Could you try reflashing the latest adapter firmware on your stick?

I just reflashed my adapter. It started working. Still didn't understand what the issue was. Thanks anyways!

oopiicaa, 2025-03-02:
Close the issue then

Koenkk, 2025-03-02:
@teymur1988 I'm also not sure what caused it, let me know if it occurs again.

closed 2025-03-02 by Koenkk